emsesp/EMS-ESP32 #884: AM200 heatsource/ahs1/bufconfig enum missing a possible value / enum wrong

tglynx, 2023-01-03:
Hi, i believe the enum for the bufconfig value (heatsource/ahs1/bufconfig) seems to be wrong.

I am using ems-esp v3.5.0-dev.15 from @MichaelDvP 's repo...

from src/devices/heatsource.cpp
`has_update(telegram, bufConfig_, 20);     // pos 20: Konfig WW-Speicher Monovalent(01), Bivalent(02) (buffer)`

When i configure the AM200 settings from my RC310 to 'undefined', ems-esp displays 'monovalent',
when i configure 'monovalent' it displays 'bivalent'

However, the values from the comment in the source seem to be fine.
My system is natively configured to 'monovalent' at the moment
Pos 20 is '01'
`002+13:20:56.019 N 5: [emsesp] Heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054C), data: 00 00 00 00 01 00 3C 46 00 78 
00 78 01 01 05 3C 02 00 5A 00 01 32 00`

This is currently displayed as 'bivalent' in the ems-esp web interface

I believe the enum currently is: 
00 = monovalent, 01 = bivalent
and it should be
00 = undefined, 01 = monovalent, 02 = bivalent

From the english manual
[image]

So maybe the correct values (in english) would even be 'undefined', 'mono', 'dual' / up to you guys... 

Let me know anytime if further input is needed...

tglynx, 2023-01-04:
Hi Michael,

from the RC310 Screen
[image]

from 'watch 60' switching through -> undefined 0x00, mono 0x01, dual 0x02 -> back to mono 0x01
```
004+00:19:08.371 N 0: [emsesp] Thermostat(0x10) -W-> Heatsource(0x60), AmSettings(0x054C), data: 00 (offset 20) <--
004+00:19:09.669 N 1: [emsesp] Heatsource(0x60) -B-> All(0x00), AmSettings(0x054C), data: 00 (offset 20) <--
004+00:19:16.500 N 2: [emsesp] Thermostat(0x10) -R-> Heatsource(0x60), ?(0xF9), length: 0x11
004+00:19:16.562 N 3: [emsesp] Heatsource(0x60) -W-> Thermostat(0x10), ?(0xF9), data: FF 04 4C 14 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 00
004+00:19:19.771 N 4: [emsesp] Thermostat(0x10) -W-> Heatsource(0x60), AmSettings(0x054C), data: 01 (offset 20)  <--
004+00:19:21.019 N 5: [emsesp] Heatsource(0x60) -B-> All(0x00), AmSettings(0x054C), data: 01 (offset 20)  <--
004+00:19:25.226 N 6: [emsesp] Thermostat(0x10) -R-> Heatsource(0x60), ?(0xF9), length: 0x11
004+00:19:25.287 N 7: [emsesp] Heatsource(0x60) -W-> Thermostat(0x10), ?(0xF9), data: FF 04 4C 14 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 01
004+00:19:27.971 N 8: [emsesp] Thermostat(0x10) -W-> Heatsource(0x60), AmSettings(0x054C), data: 02 (offset 20)  <--
004+00:19:30.469 N 9: [emsesp] Heatsource(0x60) -B-> All(0x00), AmSettings(0x054C), data: 02 (offset 20) <--
004+00:19:30.699 N 10: [emsesp] Heatsource(0x60) -B-> All(0x00), AmTemperatures(0x054D), data: 80 00 80 00 80 00 02 C2 80 00 02 4B 02 4B 02 18 00 00 80 00
004+00:19:32.435 N 11: [emsesp] Heatsource(0x60) -B-> All(0x00), AmStatus(0x054E), data: 32 21 00 04 32 32
004+00:19:32.680 N 12: [emsesp] Heatsource(0x60) -B-> All(0x00), AmExtra(0x0550), data: 00 06 0A FF 0A FF FF 1D 01 32 9C 00 00 00 03 00 00 03 00 1D 1B 58 02 62 3C
004+00:19:33.151 N 13: [emsesp] Thermostat(0x10) -R-> Heatsource(0x60), ?(0xF9), length: 0x11
004+00:19:33.213 N 14: [emsesp] Heatsource(0x60) -W-> Thermostat(0x10), ?(0xF9), data: FF 04 4C 14 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 02
004+00:19:35.000 N 15: [emsesp] Heatsource(0x60) -B-> All(0x00), AmExtra(0x0550), data: 3C 06 FF (offset 25)
004+00:19:35.896 N 16: [emsesp] Thermostat(0x10) -W-> Heatsource(0x60), AmSettings(0x054C), data: 01 (offset 20)  <--
004+00:19:36.794 N 17: [emsesp] Heatsource(0x60) -B-> All(0x00), AmSettings(0x054C), data: 01 (offset 20)  <--
004+00:19:59.717 N 18: [emsesp] Me(0x0B) -R-> Heatsource(0x60), AmSettings(0x054C), length: 0x19
004+00:19:59.795 N 19: [emsesp] Heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054C), data: 00 00 00 00 01 00 3C 46 00 78 00 78 01 01 05 3C 02 00 5A 00 01 32 00
```

Hope this helps ...

tglynx, 2023-01-11:
I have this running in the current @MichaelDvP dev build and it is working fine! (00=off, 01=mono, 02=dual)

proddy, 2023-01-12:
we can close it after it's merged

proddy, 2023-01-22:
in dev-16

closed 2023-01-22 by proddy