emsesp/EMS-ESP32 #888: Template variable warning: 'dict object' has no attribute 'temp'

eelcohn, 2023-01-05:
I'm using EMS-ESP v3.4.4 with MQTT auto-discovery in Home Assistant. When a Dallas sensor reports 85,0 °C, the value is not available in the MQTT payload, and Home Assistant reports a warning:
```
Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 11:28:38 (1114 occurrences)
Last logged: 20:34:17

Template variable warning: 'dict object' has no attribute 'temp' when rendering '{{value_json['12-3456-789A-BCDE'].temp}}'
```
Ideally the `temp` attribute should be set to the value of the Dallas sensor (`85,0 °C`), or it could even be set to `null` to report to Home Assistant that no valid temperature is available.

proddy, 2023-01-10:
There was a whole thread about this in https://github.com/emsesp/EMS-ESP32/issues/466

85 is indeed an invalid reading, typically from cheaper DS18B20 clones. I thought in EMS-ESP we filter them out and don't send the MQTT payload but I couldn't find it in the code (@MichaelDvP do you remember?)

MichaelDvP, 2023-01-11:
@proddy  You have already referenced #466. As harvey637 wrote, 85°C is a sign of underpower. But it is in the valid range, so we dont filter it. The guy in #466 have used parasite sensors, but i think not enabled parasite powering. (instead lower pullups).

I guess this case is similar, the sensor reports one/some 85 values and then gives error. After 5 errors in row the sensor is removed. But without any information about the system it's only a guess.

proddy, 2023-01-11:
thanks @MichaelDvP . Your brain is like an elephant. Can you take this one?

MichaelDvP, 2023-01-11:
Ok, as far as i see we only need a 
```
            if (!has_value && sensor.ha_registered) {
                remove_ha_topic(sensor.id());
                sensor.ha_registered = false;
```
to remove the broken sensor from HA. (as for device values).

proddy, 2023-01-22:
in dev-16