emsesp/EMS-ESP32 #891: Blacklist / Remove Entities

HansRemmerswaal, 2023-01-06:
You have done a great job in adding many entities to the EMS-ESP. In my case I see 125 for the Boiler and 53 for the Thermostat. But due to all these entities the memory of my BBQKEES module is getting full and my system started to restart due to this, see #859. I have disabled as much as possible all the functionality to free up memory and now it's running stable again for a few days. Soon you will add a few new entities and my module will start restarting again. 

Part of these entities are useless for my configuration, I have full electric so all the ones related to gas are irrelevant. 
Part of these entities will never change or are less relevant, e.g. I don't need the current date/time of my thermostat.
Luckily you have thought about that and added the option to hide these.  Unfortunatly hiding these will not reduce the memory.

**Proposed solution**
In the ideal world we would have in the settings the option to indicate the kind of configuration you have (hybrid / full electric / ...) and automatically all the irrelevant entities are excluded and not part of the memory. But I doubt that is a feasible solution as someone needs to categorize all the entities of all the devices.

A more workable solution could be a kind of black list. From the customization tab you can blacklist an entity which will remove it from the list of entities. A new blacklist tab will show all the entities that are blacklisted, from there it can be deleted from the blacklist. During the receiving of the telegrams the blacklist is checked and blacklisted entities are ignored. I think it's acceptable that the user needs to restart the EMS-ESP after blacklisting entities to free up the already occupied memory.

My expectation is that I can eliminate a part of the entities and with that free up memory. That allows me to enable some disabled functionality again, and will make it possible for you to add new entities again without worrying about the available memory.

proddy, 2023-01-06:
I've created a specific GH issue to look into optimizing the memory (#869) as it's not clear to me what the thresholds and lower limits are for Heap Memory and Max Alloc block before it causes EMS-ESP to run out of RAM and crash. I'm writing some code to simulate complex setups (like yours) which loads multiple device and entities to see where the memory goes. I have a hunch the problem is in the Max Alloc which is the largest amount of continuously aligned memory blocks which can be used to store a buffer. Often this becomes fragmented and gets smaller and smaller. We use large buffers for MQTT and the Web. If we solve that I think we should be fine.

Having all the entities loaded is important to be able to enable/disable them from the customization screen. I don't think they take too much internal memory as the long string names are stored in Flash memory. If the results of the memory investigation gets us nowhere we can implement the black-list as you suggested. 

The are other alternatives. One is to switch to using the ESP32-S2 which has an extra 2MB PSRAM and costs 2 EUR. There is still some coding to be done to support using this extra RAM but we're close.

HansRemmerswaal, 2023-01-06:
@proddy wrote in #859:
> Your free mem and max alloc is very low (mine is 133/75, yours 83/39).
> It's 100+ KB free heap seem normal for 3 devices (boiler and 2 thermostats) with a total of180 entities. My system has 78 so hard to compare.

Out of that I got the impression that there is a relationship between the number of entities and the free mem / max alloc.
So if you say that there is not really such relation, then what explains the big difference in free mem / max alloc?

HansRemmerswaal, 2023-01-07:
@MichaelDvP 

**Summary:**
- Between EMS-ESP-3_5_0-dev_14-ESP32 and EMS-ESP-3_5_0-dev_15-ESP32 I **lost 10 KB** of Free
- emsesp_customizations.json also contains entities which are not flagged in customization
- 185 entities results in 99 / 39 (Free / Max Alloc)
- 104 entities results in 113 / 46 (Free / Max Alloc)
- It seems to work but I don't think that normally I would disable 81 entities, so it's an option that helps to reduce heap usage but it in normal cases it might not reduce enough

**Question:**
Why did I lost 10 KB Free Heap between dev_14 and dev_15

**Test results:**
Downloaded EMS-ESP-3_5_0-dev_15-ESP32 from
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

**Before installing your EMS-ESP-3_5_0-dev_15-ESP32**
Free: 109
Max Alloc: 39
Entities Boiler: 125
Entities Thermostat RC300: 53
Entities Thermostat RC100H: 4

**After installing**
Free: 99
Max Alloc: 39
Entities Boiler: 129
Entities Thermostat RC300: 52
Entities Thermostat RC100H: 4

**Marked all of RC300 as exclude from MQTT and hide from dashboard**
Exported emsesp_customizations.json
Replace "0 with "8
Upload emsesp_customizations.json

**After restart**
Free: 113
Max Alloc: 46
Entities Boiler: 100   <--- This also reduced, see text at the end
Entities Thermostat RC300: 0
Entities Thermostat RC100H: 4

**Replaced "8 by "0 and loaded emsesp_customizations.json again**
Free: 97
Max Alloc: 39
Entities Boiler: 129 <--- back to normal
Entities Thermostat RC300: 0
Entities Thermostat RC100H: 4

** Set all to visible again for RC300 **
Free: 99
Max Alloc: 39
Entities Boiler: 129
Entities Thermostat RC300: 52
Entities Thermostat RC100H: 4

Downloaded EMS-ESP-3_5_0-dev_15-ESP32.bin from
https://github.com/emsesp/EMS-ESP32/releases/tag/latest

**After restart**
Free: 99
Max Alloc: 39
Entities Boiler: 129
Entities Thermostat RC300: 52
Entities Thermostat RC100H: 4

**Double checked customization, removed 2 favorites**
Exported emsesp_customizations.json
Expected it to be "empty" but it does contain a number of entities, so that might explain why 29 of the boiler disappeared.
[emsesp_customizations.txt](https://github.com/emsesp/EMS-ESP32/files/10365652/emsesp_customizations.txt)

MichaelDvP, 2023-01-07:
Do you count the entities from dashboard (registered with value) or from customization (registered)? 
From datastructure the register of a sensor consumes ~40bytes, a setting (with command) ~60bytes. But it seems to be more in real life, as you gain 14k by removing 81 entities. 
> Why did I lost 10 KB Free Heap between dev_14 and dev_15

I don't know, there are a lot of changes: library updates (arduinojson), new device-types (heatsource), a lot of new settings, etc. 
> Expected it to be "empty" but it does contain a number of entities

These are entities where you have set custom min/max values.

HansRemmerswaal, 2023-01-07:
> Do you count the entities from dashboard (registered with value) or from customization **(registered)?**

From dashboard.
[image]

> These are entities where you have set custom min/max values.

I was able to reproduce this issue after a factory reset. Raised #895 for this.

What are we going to do with this issue? Close it or keep it as an enhancement for reducing memory?

proddy, 2023-01-07:
Keep it until we know what to do with the memory. The heap can be a problem but still 50KB is a lot left. My bet is still on the max allocation size is the root cause.

MichaelDvP, 2023-01-07:
I've tested to read the removed entities from customization file and show with trash-can.
Also possible to set/reset the 0x80 flag, needs reboot to activate.
[image]

HansRemmerswaal, 2023-01-08:
That looks promising, I tried your latest build but that one doesn't contain this option yet.
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Note: in your screenshot, below the options, the "hide from dashboard" disappeared.

MichaelDvP, 2023-01-10:
It was a first attempt to show the manual removed shortnames. For setting in web and storing it needs some more code.
Maybe something for v.3.6.

HansRemmerswaal, 2023-01-11:
Sure no problem. 

For now this change helps to free up some of the heap, but if #869 is working then maybe this whole change is not required anymore.

MichaelDvP, 2023-01-13:
@HansRemmerswaal I've found the way to edit the blacklist in web customizations. 
Please try the blacklist build from here https://github.com/MichaelDvP/EMS-ESP32/releases
@proddy Changes here: https://github.com/MichaelDvP/EMS-ESP32/commit/23d40ee49d9242671e0341d880aad06492ed6772
~ 1k flash usage, mostly the new icons.

MichaelDvP, 2023-01-13:
> When I blacklist all the entities of one device then the device doesn't show up anymore in the customization

Good point, should work now.

proddy, 2023-01-13:
while you boys are playing in the Customization screen, ever get this problem? https://github.com/emsesp/EMS-ESP32/issues/911

MichaelDvP, 2023-01-14:
Only a few times after the customization was introduced. First i was confused by the save-button in setting dialog. For me i renamed it to "update"  and each time i see the update-button i remember to click save. (But this only works for me).

HansRemmerswaal, 2023-01-14:
Installed the lastest version and it's working nice! Thanks a lot.
I guess we can closes this issue?

proddy, 2023-01-14:
> Installed the lastest version and it's working nice! Thanks a lot. I guess we can closes this issue?

When it's merged into the dev branch we can close it, otherwise we'll loose track of the changes

proddy, 2023-01-22:
tests have been added to 3.5.0-dev-16

MichaelDvP, 2023-01-24:
@proddy, should we add the feature to set/reset in webui ti v3.5? It's ~1k flash, most for the extra webui-icon.
I have only a branch with this feature and first parts of TR language, but i can make make a own PR for this.

HansRemmerswaal, 2023-01-24:
I suggested this improvement to prevent the automatic restart each time of the EMS-ESP due to not enough free heap. Since I disabled all standard options, including the fixed ethernet connection, my device is not restarting anymore. 

The solutions works and frees up enough memory for users who can not disable all options. But once #869 has freed up enough heap then I think this #891 might become obsolete.

proddy, 2023-01-24:
> @proddy, should we add the feature to set/reset in webui ti v3.5? It's ~1k flash, most for the extra webui-icon. I have only a branch with this feature and first parts of TR language, but i can make make a own PR for this.

sure, if it fits let's add it.

closed 2023-02-04 by proddy