emsesp/EMS-ESP32 #892: 9 new entities 0x49C and 0x49D

HansRemmerswaal, 2023-01-06:
I have found the following 9 entities which, as far as I can see, do not exist yet in EMS-ESP. Could you add them? 
The model I have is Nefit Eviline A/W Split 7.0 E-S.

1) Switch on heating cable
2) VC0 valve
3) Start primary heat. pump
4) Primary heat pump output
5) 3-way valve
6) Add. elec. heater step 1
7) Add. elec. heater step 2
8) Add. elec. heater step 3
9) Valve/pump cooling on

**1) Switch on heating cable**

On
000+00:02:51.434 N 0: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 2)
000+00:02:52.082 N 1: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 01 00 00 00 00 00 00 00 00 00

Off
000+00:02:59.872 N 2: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 2)
000+00:03:00.178 N 3: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**2) VC0 valve**

On
000+00:11:52.979 N 36: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 3)
000+00:11:53.101 N 37: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 01 00 00 00 00 00 00 00 00

Off
000+00:11:57.304 N 38: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 3)
000+00:11:58.300 N 39: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**3) Start primary heat. pump**

On
000+00:25:47.613 N 16: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 4)
000+00:25:48.073 N 17: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 01 00 00 00 00 00 00 00

Off
000+00:25:55.776 N 18: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 4)
000+00:25:56.748 N 19: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**4) Primary heat pump output*

10%
000+00:23:51.488 N 10: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 0A (offset 5)
000+00:23:52.471 N 11: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 0A 00 00 00 00 00 00

20%
000+00:24:03.988 N 12: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 14 (offset 5)
000+00:24:04.135 N 13: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 14 00 00 00 00 00 00

0%
000+00:24:17.287 N 14: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 5)
000+00:24:18.246 N 15: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**5) 3-way valve**

DHW
000+00:16:11.445 N 40: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 6)
000+00:16:12.278 N 41: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 01 00 00 00 00 00

Heating
000+00:16:15.607 N 42: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 6)
000+00:16:16.691 N 43: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**6) Add. elec. heater step 1**

000+00:03:14.885 N 4: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 7)
000+00:03:15.032 N 5: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 01 00 00 00 00
000+00:03:20.197 N 6: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 7)
000+00:03:21.268 N 7: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**Add. elec. heater step 2**

On
000+00:04:07.435 N 12: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 8)
000+00:04:08.544 N 13: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 01 00 00 00

Off
000+00:04:15.235 N 14: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 8)
000+00:04:16.907 N 15: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**Add. elec. heater step 3**

On
000+00:04:39.336 N 20: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 01 (offset 9)
000+00:04:39.895 N 21: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 01 00 00

Off
000+00:04:47.286 N 22: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049D), data: 00 (offset 9)
000+00:04:48.895 N 23: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00

**9) Valve/pump cooling on**

On
000+00:32:03.030 N 57: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049C), data: 01 (offset 3)
000+00:32:03.071 N 58: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049C), data: 00 00 00 01

Off
000+00:32:07.456 N 59: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x049C), data: 00 (offset 3)
000+00:32:08.156 N 60: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049C), data: 00 00 00 00

[image]
[image]

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10361807/emsesp_info.txt)

HansRemmerswaal, 2023-01-08:
The **good news** is that I see the values changing when I change the settings from the diagnostic->Function test menu on the HMC310. 

The **bad new** is that I do not see the values changing (yet) during normal operation. E.g. I started the thermal disinfection hoping to see the El.heater step 1, 2 and 3 to change, but no luck yet.

Note: the value of the 3-way valve on the menu is actually DHW / heating instead of On / Off.

[image]

proddy, 2023-01-10:
Make them read only or also writeable? Also can you confirm if you `read 49C` and `49D` that the settings are still in that telegram (to make sure it's not just a control telegram for sending commands).

proddy, 2023-01-22:
Michael added this to dev-16

closed 2023-01-22 by proddy