emsesp/EMS-ESP32 #894: Connecting to WiFi network is consuming 38 / 12 from Heap

HansRemmerswaal, 2023-01-07:
**Question**
Is it normal that Connecting to the WiFi network is taking
- 38 KB from the Free Heap
- 12 KB from the Max Alloc Heap

**After factory reset, and EMS-ESP not connected EMS bus**
Heap (Free/Max Alloc) = 168 / 95

**Disable Analog Sensors and mDNS Service, enable NTP and set fixed IP**
Heap (Free/Max Alloc) = 177 / 107

**Setup WiFi**
Heap (Free/Max Alloc) = **139 / 95**

[emsesp_info (1).txt](https://github.com/emsesp/EMS-ESP32/files/10366157/emsesp_info.1.txt)

MichaelDvP, 2023-01-07:
Yes each network connection needs some heap. You have a E32, if you only connect wifi and not eth, go to settings boardprofile, switch to custom and disable the phy. If using ethernet, clear the network ssid to disable wfi.
You can also disable OTA to save memory, set AP to never, and (in my build) set weblog level to off (clearing the buffer).

HansRemmerswaal, 2023-01-07:
Heap (Free / Max Alloc)
100 KB / 39 KB

**disabled OTA**
Heap (Free / Max Alloc)
102 KB / 39 KB

**disabled ethernet**
Heap (Free / Max Alloc)
118 KB / 45 KB

Great!
Maybe something to add to the **It may be memory related* in
https://emsesp.github.io/docs/Troubleshooting/#ems-esp-sometimes-crashes-and-restarts

proddy, 2023-01-07:
> Maybe something to add to the It may be memory related in
https://emsesp.github.io/docs/Troubleshooting/#ems-esp-sometimes-crashes-and-restarts

done!

closed 2023-02-19 by proddy