emsesp/EMS-ESP32 #902: "Invalid value" errors in HomeAssistant log

pswid, 2023-01-10:
I'm testing the latest v3.5.0-dev.15 with Buderus WSW196i heat pump . I noticed that Home Assisntant shows such errors in log:

[image]

Of course, I can solve the problems by customisation of these entites, but I think that at default settings  of EMS-ESP (no customisation), such errors should not happen. What do you think?

It seems that `boiler_on_off_hyst_heat` and `boiler_on_off_hyst_cool` are not used in my hp because they return "0". Maybe in cases there value (of any entity) is outside valid range and equal to 0, simply assume that it's not being used and ignore/disable such entity?

Furthermore, `boiler_aux_heater_max_temperature` doesn't show correct data. In my system it shows 210, so it can't be a temperature (K or °C). BTW are we sure about the range? Values 0-10°C are too low to be apropriete for "max temperature" of the heater.

MichaelDvP, 2023-01-10:
The value should be scaled to 2.1, it is saved as 21 and wrongly multplicated instead of divided. I'll fix.

pswid, 2023-01-11:
Hmm, 2.1 is still not a value that can be considered the correct max(!) temperature (because it is almost absolute zero temperature!), even if we assume that the unit is °C not K. Maybe it's just 21°C? Although even this value seems too small

MichaelDvP, 2023-01-12:
> Hmm, 2.1 is still not a value that can be considered the correct max(!) temperature (because it is almost absolute zero temperature!)

LOL, Bosch uses kelvin unit in newer systems for relative temperatures (temperaturdifferences). This is a clever idea, because some silly homeautomations calculate °C->°F always as absolute value with offset.

See #878, it's range 0-10.0 with 0.1 steps.

pswid, 2023-01-12:
Silly me. You were right, value 2.1K is correct. I found this setting on my heat pump menu. However, the parameter name is very misleading. It is not a specific temperature, but a temperature difference. Parameter is visible in the "Max. add. heater temp." submenu and it's name is "Maximum limit". Therefore, the name of the entity (also the short one) should be changed to contain information that it is about "max limit". 
There is also a second parameter in this menu called "Limit start", so you could add this as well. It seems it is being send at offset 15 in the `HpAdditionalHeater` telegram (in my case it's value is 2.0K):

`boiler(0x08) -W-> Me(0x0B), HPAdditionalHeater(0x0491), date: 01 00 00 00 00 01 64 00 00 04 01 2C 00 0A 15 14 00 1D 00 00 1E 0A 1E 05 05`

[image]

According to the manual, the range of the both parameters is 3-15K. "Maximum limit" can also be set "off" (=0?).

BTW, meaning of the `auxHeatMode` (values: eco/comfort) parameter is "Aux heater mode **after blocking from the power company**". Would you adjust the names (long and short) to be more accurate?

MichaelDvP, 2023-01-12:
Changing names is a very annoying work. One time i had searched all the translations from manuals, and then user reports that the cotroller have different names. Also the limits are reported 0-10.0, now 3-15. We have the custom limit function, i will not change it anymore.
For the names please **define** short and long names,  I'll change it this time, but never again. Andthe shortnames for mqtt should be **short**, there is limited space in the json.

pswid, 2023-01-13:
I agree that entity naming is debatable. However, we should always try to use such names (especially the long ones) from which the user will be able to deduce what a particular parameter do, don't you think? Although, unfortunately, many names are "bad" even in the official manuals.

As for the ranges. I checked on my WSW196i and the possible values for auxMaxLimit are indeed 0-10 (so manual is wrong) but auxLimitStart is 0.1-10 (without "0"). Will you correct the range for auxLimitStart?

BTW, I noticed that some fractional values cannot be set using the ems-esp dasboard! E.g. 1.9, 2.3, 3.3, 3.8, etc. (trying to set 2.3 we will get 2.2, etc.). The problem appears only on the ems-esp dashboard. Changes in the other way (e.g. from HomeAssistant) work OK. In the log you can see that dasboard is sending wrong values to EMS bus.

MichaelDvP, 2023-01-14:
It seems bosch uses different names for the same parameter on every brand, and each version of manual and software. There is a never ending line of complains, that is does not fit the manual/controllerdisplay. That's why we have added the custom names and custom min/max settings. The min/max is stored as interger, no chance to set it to 0.1.
The fraction in webpage is known, but it unknown how to fix it.

closed 2023-01-22 by proddy