emsesp/EMS-ESP32 #903: entity without name

pswid, 2023-01-10:
I'm testing the latest v3.5.0-dev.15 with Buderus WSW196i heat pump. 

On the customisation page I see the entity without (a proper) name which has a value. Because of this, it is not visible on the dashboard.

[image]

the same in MQTT:

```
"wwdisinfectday": "tu",
  "wwdisinfecttime": 105,
  "hc1": {
    "seltemp": 21,
    "": "selTemp",
    "mode": "manual",
    "modetype": "comfort",
    "ecotemp": 15,
```

Shouldn't it be named a `climate` (`hc1/climate` in my case)?

MichaelDvP, 2023-01-10:
The name was "HA climate config creation" as shortname with no fullname, but was removed in https://github.com/emsesp/EMS-ESP32/commit/147be12583b41d031081dbdbee63b3e09dc6d84d. 
Don't know why. Also the link now does not show the climate info, it expands wrongly to hc1/values.

proddy, 2023-01-10:
I removed it because it broke HA (see the start of mqtt.cpp) - but should be looked it. The climate is a special entity, a bit of hack and for these type of changes we really need to add comments in the code so we don't forget how they work. And write tests. I'll add it back and figure a way to stop HA from exploding

MichaelDvP, 2023-01-10:
How can it break HA? It's in all v3.4 releases, is not published to mqtt, does not create sensor config, only used to create the special climate config.

proddy, 2023-01-10:
that's the thing, due to other changes in 3.5 it was creating HA discovery configs

proddy, 2023-01-18:
I've added it back to my tec branch, but it's an easy change that we'll add to 3.5.0 before it's released.

proddy, 2023-01-22:
in dev-16. In 3.6.0 it will be renamed to something different

closed 2023-01-22 by proddy