emsesp/EMS-ESP32 #904: [Enhancement] Workaround for better Domoticz MQTT intergration?

Tigerstrength, 2023-01-10:
Hi,

I am using this software with great pleasure to control my Nefit Enviline 9kw split heatpump. 

Currently I am running v3.5.0b12. 

I use a Mosquitto MQTT broaker and Domoticz 2022.2 as backend. Al runs smooth, but recently I noticed missing some sensors which Home Assistant users did have and I had not. Digging into it I found out that Domoticz does not support the number type mqtt messages. That is why Domoticz users are missing out a lot of usefull sensors.

From the domoticz files I see this is supported for MQTT autodiscovery (MQTTAutoDiscover.cpp):

`(pSensor->component_type == "sensor")
                || (pSensor->component_type == "binary_sensor")
                || (pSensor->component_type == "switch")
                || (pSensor->component_type == "light")
                || (pSensor->component_type == "lock")
                || (pSensor->component_type == "select")
                || (pSensor->component_type == "cover")
                || (pSensor->component_type == "climate")`

Is it possibible to add an option for sending numbers in an other format to MQTT? Most of the numbers can also be send as a select, climate or even text state.

Hoping for a solution. Curently I have a Python script running to handle the missing data, but that is far from ideal.

proddy, 2023-01-10:
One solution is to go into the ESM-ESP Customization page and select these missing entities (they will be numerical and writeable) and mark them as read-only. This will force the Discovery topics to make them sensors and not numbers. I haven't tried it, but in theory it should work. If it's a huge problem we can add a workaround into EMS-ESP but I'd rather not. You could also log an enhancement/bug against Domotic that they're not supporting the full Discovery protocol :-)

Tigerstrength, 2023-01-10:
Thanks @proddy for your solution. This might work, but i disables the option of writing new settings to these entities. 
 
As an example: I would like Domoticz to control the min.flow temperature to write a softstart service (preventing overshoot)

Or am I wrong are the entities still writeable by MQTT?

proddy, 2023-01-10:
oh yeah, you're right. if the write action is disabled then both MQTT and API will not work. If more people have the same issue we'd need to change EMS-ESP to choose which flavor of MQTT Discovery is being used (Domoticz, HA) and use that flag to determine whether to use 'number' or 'sensor'. It's more code so I'd rather not just yet. There are only 2 devs on the project and little time at the moment.

Tigerstrength, 2023-01-10:
I see it he dilemma here. 

Hopefully you still find time to implement this, as I know there are still a lot of Domoticz users around. Not as many as with Home Assistant I guess, but as the enhancement seems do-able with not too much code it is worth the effort imho.

Thanks so far! 👍

bbqkees, 2023-01-12:
Auto Discovery was created for HA, and is now more or less becoming a de facto standard.
As it was created for HA, it is not always a good match for other systems like Domoticz.
Some parts of Discovery are 'easy' to implement, others don't fit properly on the structure a certain system implements.
So there is this tension between a good implementation of Discovery, and what can actually be matched in a home automation system. 
Maybe it's better to open an issue on the Domoticz repository to see if they can have another look at this.

Tigerstrength, 2023-01-15:
I understand the tension it creates, but I also would like to refer to the fact that de EMS-ESP software is listed as compatible with both HA and Domoticz. So please keep that in mind when considering wether to make a modification/enhancement to patch this is issue or not. 

I will also start a topic on the Domoticz GitHub, it is valid to say they a also part of this issue. Also for the future support of mqqt discovery with number topics is a must for Domoticz. 

Please note that I absolutely love the work which is done hear. So hands downs to you all 👍

Tigerstrength, 2023-01-16:
Also made a Domotiicz issue: https://github.com/domoticz/domoticz/issues/5550

proddy, 2023-02-19:
in 3.6.0

closed 2023-02-19 by proddy

Tigerstrength, 2023-03-06:
Please note that on Domoticz side things are moving aswel. Maybe @proddy can guide Gizomocus with the trouble of implementing this?

gizmocuz, 2023-03-08:
> Please note that on Domoticz side things are moving aswel. Maybe @proddy can guide Gizmocuz with the trouble of 
implementing this? [https://github.com/[domoticz/domoticz/issues/5550](https://github.com/domoticz/domoticz/issues/5550)]

There's no trouble I think? Anyway it is not possible to configure the number values.