emsesp/EMS-ESP32 #906: [Enhancement] Flue gas temperature reading for alternative heat sources (AM200)

Snypes88, 2023-01-11:
Hello everyone, 

as per @tglynx suggestion from the following thread:

https://github.com/emsesp/EMS-ESP32/issues/857

I would like to create a feature request / enhancement to the alternative heat sources feature in EMS ESP in which the flue gas temperatures which can be logged by the Buderus AM200 module and a PT1000 are also picked up within the EMS ESP dashboard and interface. Doing so, could enable those of us who have a water jacket wood furnace attached to the home heating system to log said exhaust flue gas temperatures and use said read out to notify the user that temperatures have fallen below a given threshold value for example.... HA could then send a notification to load up wood in stove for example. This info could also be used to regulate the air flow to the combustion chamber to attain more complete combustion with an automatic flow damper, for example 😁 All ideas that can be more easily realized if this data is picked up by the already amazing EMS ESP project! 

Here as a picture from the RC310 controller showing how this info is read

[image]

MichaelDvP, 2023-01-11:
Make a full log of more than a minute and not the sensor value shown in the display. If possible change the sensor value by heating, not the new value while logging. 
For the 21.5°V we are searching for a `00 D7` somewhere in a telegram. If we found it and it changes in the second step, we have the position.

Snypes88, 2023-01-11:
So, you're saying it probably makes more sense to try and identify the information when all other values are held constant? I.e. the fore place is off and I heat the sensor up with a heat gun or my hands for example?

Can you recommend what telnet command I should use? Is it the watch 60 you mentioned in the other thread? I'm a major newb when it comes to debugging 😅

tglynx, 2023-01-11:
yes 'watch 60' will do ... it will log all messages coming from and going to the 0x60 device which is the AM200 module.

Snypes88, 2023-01-11:
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x60
000+02:43:14.827 N 0: [emsesp] Me(0x0B) -R-> heatsource(0x60), AmSettings(0x054C                                                          ), length: 0x19
000+02:43:14.910 N 1: [emsesp] heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054C                                                          ), data: 01 01 01 01 01 00 3C 46 00 78 00 78 00 00 05 1E 01 00 5A 00 01 23 00 64                                                           23
000+02:43:15.427 N 2: [emsesp] Me(0x0B) -R-> heatsource(0x60), AmSettings(0x054C                                                          ), length: 0x19 (offset 25)
000+02:43:15.450 N 3: [emsesp] heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054C                                                          ), data: 0F 05 (offset 25)
000+02:43:36.681 N 4: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 01 1E (offset 4)
000+02:43:38.056 N 5: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 01 86 (offset 4)
000+02:43:41.206 N 6: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 01 F4 (offset 4)
000+02:43:43.006 N 7: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 02 69 (offset 4)
000+02:43:45.906 N 8: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 02 E5 (offset 4)
000+02:43:46.604 N 9: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0                                                          x054D), data: 80 00 80 00 02 E5 02 96 02 58 02 3B 80 00 02 4D 00 00 80 00 01
000+02:43:46.802 N 10: [emsesp] heatsource(0x60) -B-> All(0x00), AmStatus(0x054E                                                          ), data: 23 01 00 06 34 00
000+02:43:47.816 N 11: [emsesp] heatsource(0x60) -B-> All(0x00), AmExtra(0x0550)                                                          , data: 00 06 0A FF FF 0A 0D 1B 01 23 00 35 35 01 03 01 00 03 00 1B 1B 58 02 62                                                           1E
000+02:43:48.010 N 12: [emsesp] heatsource(0x60) -B-> All(0x00), AmExtra(0x0550)                                                          , data: 1E 06 01 FF 00 (offset 25)
000+02:43:48.706 N 13: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 03 5E (offset 4)
000+02:43:49.906 N 14: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 03 D3 (offset 4)
000+02:43:51.481 N 15: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 52 (offset 4)
000+02:43:54.006 N 16: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 AA (offset 4)
000+02:43:56.931 N 17: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 D7 (offset 4)
000+02:43:59.756 N 18: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 E8 (offset 4)
000+02:44:00.982 N 19: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 05 10 (offset 4)
000+02:44:04.556 N 20: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 05 53 (offset 4)
000+02:44:05.781 N 21: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 05 A1 (offset 4)
000+02:44:07.731 N 22: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 05 DF (offset 4)
000+02:44:10.881 N 23: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 06 0A (offset 4)
000+02:44:13.431 N 24: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 06 20 (offset 4)
000+02:44:16.152 N 25: [emsesp] Me(0x0B) -R-> heatsource(0x60), AmSettings(0x054                                                          C), length: 0x19
000+02:44:16.235 N 26: [emsesp] heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054                                                          C), data: 01 01 01 01 01 00 3C 46 00 78 00 78 00 00 05 1E 01 00 5A 00 01 23 00 6                                                          4 23
000+02:44:16.727 N 27: [emsesp] Me(0x0B) -R-> heatsource(0x60), AmSettings(0x054                                                          C), length: 0x19 (offset 25)
000+02:44:16.751 N 28: [emsesp] heatsource(0x60) -W-> Me(0x0B), AmSettings(0x054                                                          C), data: 0F 05 (offset 25)
[... cut ...]
000+02:44:55.041 N 47: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 76 (offset 4)
000+02:44:57.432 N 48: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 57 (offset 4)
000+02:44:59.357 N 49: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(                                                          0x054D), data: 04 3A (offset 4)

Snypes88, 2023-01-11:
this is the output I get from putty... hope this is an OK way to display the info?

The RC310 is showing the following also... 
[image]

tglynx, 2023-01-11:
i am pretty sure it is offset 4/5 then ... i can confirm that i have no values in the AmTemperature message on my end (i do not have that sensor)

tglynx, 2023-01-11:
according to the manual the connector is called TF1 ... (as a note for the message comment in the source)

Snypes88, 2023-01-11:
Exactly!! BTE, where did you connect the temperature probe to to get the middle buffer temperature read out? TB2?

tglynx, 2023-01-11:
sure ... TB1 is top, TB2 is center, TB3 is bottom

Snypes88, 2023-01-12:
and TB4 seems to be System Flow Temperature with TR2 being System Return Temperature 😇 Both also very helpful to have as they will dictate whether it makes sense to block the primary heating device... the AM200 seems to automatically block as soon as the buffer begins loading and not necessarily based on the buffer temperature being above the set DHW temperature (which in my case is the highest heat load at 53° C)

tglynx, 2023-01-12:
sure, TB4 and TR2 are implemented as well... 
i am already running the latest dev release of @MichaelDvP and the fluegastemp is implemented already.
However, i can not see values as i do not have a sensor connected to TF1...

Snypes88, 2023-01-12:
OMG!! Where can I find the latest dev release to test? It is not showing up within my EMS ESP Dashboard update tab....

MichaelDvP, 2023-01-12:
> OMG!! Where can I find the latest dev release to test? 

The latest dev is always where the dashboard points to: https://github.com/emsesp/EMS-ESP32/releases
For testing my new changes before adding to the official dev there is a testbuild on my repo. But often changed until all works, then it is merged to dev. See: https://github.com/MichaelDvP/EMS-ESP32/releases

Snypes88, 2023-01-13:
@MichaelDvP amazing news! I can confirm that your update works like a charm!! Flue gas temperature is not included in the latest blacklist release, but in the one before that it is 🙌

[image]

MichaelDvP, 2023-01-13:
> Flue gas temperature is not included in the latest blacklist release, but in the one before that it is

It's also in the blacklist-build. Sometimes it takes a minute to collect all entities.

Snypes88, 2023-01-13:
Yes, I can confirm it is also available in the latest blacklist-build! Thank you for the lightning fast implementation 🙏 EMS-ESP Rocks!

proddy, 2023-01-22:
in dev-16

closed 2023-01-22 by proddy