emsesp/EMS-ESP32 #907: Add Turkish language

proddy, 2023-01-12:
Dreixin on Discord kindly added translations for Turkish that is attached in the file below and needs to be tested and merged.

I suggest we sort out the v3.5.0 memory issues first and then add the additional language.

Thanks Dreixin!!

[emsesp32_tr_mod.zip](https://github.com/emsesp/EMS-ESP32/files/10404127/emsesp32_tr_mod.zip)

dreixin, 2023-01-13:
if you could show me the path to the entities I can translate them as well.

MichaelDvP, 2023-01-13:
The original file is here: https://raw.githubusercontent.com/emsesp/EMS-ESP32/dev/src/locale_translations.h
But there are some new, untranslated entities from pswid and me. Here is a merged file (renamed to txt for upload): 
[locale_translations.h.txt](https://github.com/emsesp/EMS-ESP32/files/10409893/locale_translations.h.txt)

I've merged your webtranslations in a dev build, but, as expected, need to add some characters to the font (Scedilla, etc.)
firmware to test, check if something looks odd: 
[dev_15_tr.zip](https://github.com/emsesp/EMS-ESP32/files/10409860/dev_15_tr.zip)

MichaelDvP, 2023-01-13:
@proddy This is now with turkish web translations, 5 more characters in font, and all new entities including the pr from pswid. Flash space in partition left:
[image]

proddy, 2023-01-13:
that was my fear, which is why we should include this in the next 3.6 build and think about grouping and bundling languages into separate builds.

dreixin, 2023-01-15:
[locale_translations.h.txt](https://github.com/emsesp/EMS-ESP32/files/10420217/locale_translations.h.txt)
translated to Turkish until entities for now.
language names were missing so I took the liberty to add sweedish, polish, norwegian and turkish; if I did something wrong please do tell.
if you could supply it within an excel file I would be much faster.
will continue working on it, hope it helps.

MichaelDvP, 2023-01-15:
> if you could supply it within an excel file I would be much faster.

[locale_translations.xlsx](https://github.com/emsesp/EMS-ESP32/files/10420277/locale_translations.xlsx)

proddy, 2023-02-24:
@MichaelDvP I think there's flash room now. Shall I add to the tech-upgrade branch. I hadn't looked at the translations, is it ok to go? You mentioned the additional of two new characters to the font.

MichaelDvP, 2023-02-24:
The entiy translation are not completed. But for the actual tr-translations i have tech-branch compatible files. Also the adapted font-files. Should i attach the files here, or make a PR?

proddy, 2023-03-01:
@dreixin this is now in the 3.6.0 dev build but missing some translations (which are marked with `// need translation` in the code).

closed 2023-04-07 by proddy