emsesp/EMS-ESP32 #910: Make the mqtt icon for ha discovery customizable

tp1de, 2023-01-13:
Every ems-esp restart will re-eset changes done within HA in respect to icon changes. Please make the icon changeable within ems-esp customization ... next to changing the entity name.

proddy, 2023-01-13:
as a workaround, you could add a line to the `customization.yaml` file (https://www.home-assistant.io/docs/configuration/customizing-devices/)

tp1de, 2023-01-13:
> as a workaround, you could add a line to the `customization.yaml` file

Thanks @proddy that' what I have done.

tp1de, 2023-01-13:
I tested with my km200 workflow in HA using node-red and mqtt discovery. Manual icon changes are kept persistent when re-sending mqtt config.

You said in discord:
>when EMS-ESP restarts it removes all the discovery config topics and rebuilds them, and some of them contain the 'ic' which is the icon, so any custom settings in HA are overwritten.

Why do you need to remove all discovery config topics on ems-esp restart? 
Is this linked to the "Set Clean Session" flag in mqtt settings? (without config topics will not be removed?)

proddy, 2023-01-13:
> Why do you need to remove all discovery config topics on ems-esp restart?
Is this linked to the "Set Clean Session" flag in mqtt settings? (without config topics will not be removed?)

If there is a Discovery Config topic in MQTT that points to another topic and data element and that doesn't exist, HA will vomit hundreds of warnings in its log because it can't find and render the value. This is because the data topic (e.g. boiler_data, thermostat_data) is dynamically built-up and only shows the values that exist in EMS-ESP. The same logic applies when a value is removed or becomes 'dormant', we want to exclude it from the payloads and also delete the Discovery topic to avoid warnings.

Clean Session is something different and internal to MQTT. See http://www.steves-internet-guide.com/mqtt-clean-sessions-example/