emsesp/EMS-ESP32 #911: Warn user in WebUI of unsaved changes

proddy, 2023-01-13:
Feature to prompt the user of unsaved changes before navigating off a page.

Many times I've gone into the Customization page, make some changes and then forget to click the 'SAVE' button, so have to go back and do it all over again. Maybe it's just me :)

proddy, 2023-01-14:
@MichaelDvP said in https://github.com/emsesp/EMS-ESP32/issues/891#issuecomment-1382695977

> Only a few times after the customization was introduced. First i was confused by the save-button in setting dialog. For me i renamed it to "update" and each time i see the update-button i remember to click save. (But this only works for me).

which got me thinking. I checked how for example my Unifi network Web interface does this. If you make any changes then an UPDATE and CANCEL button appear like:

[image]

this is a nice approach.

MichaelDvP, 2023-01-15:
Or just a messagebox on top or bottom  of page if something i changed. 
BTW: the edit-dialog cancel button only resets name/min/max, but not masks that are changed. I've tried to change using `deviceEntities.m`instead of `de.m` , but seems i did not understand the algorithm of masking correctly. Please check.

proddy, 2023-01-15:
> BTW: the edit-dialog cancel button only resets name/min/max, but not masks that are changed. I've tried to change using deviceEntities.minstead of de.m , but seems i did not understand the algorithm of masking correctly. Please check.

I'll take a look

proddy, 2023-01-19:
> I'll take a look

it was too complicated to adjust so I just took it out. It was always a 'nice to have' to show the mask icons on the edit screen but just duplicating the functionality. Well, that's my excuse anyway ;-)

proddy, 2023-01-19:
In the tec build I've
- renamed all SAVE to UPDATE in the Dialog boxes, and 'APPLY UPDATES (x)' in the Form screens, using the UniFi webUI as an reference (they're a billion $ company so must be doing something right)
- I have prototype code for navigating off the form when the form is Dirty, but I need to first refactor the routing to use `createBrowserRouter` https://reactrouter.com/en/main/routers/picking-a-router which came in router-dom 6.4

proddy, 2023-01-19:
I noticed in the Customization page there's a bug when it uses the old name, so need to fix that first

MichaelDvP, 2023-02-15:
Works good and looks nice, but after saving the changes the old-status is not updated, i have to leave the page and reload it to set  this setting back.

MichaelDvP, 2023-02-15:
Already found, insert here:
https://github.com/emsesp/EMS-ESP32/blob/b218bf1d2e0e8130c8822632c7e148e18895f1e8/interface/src/utils/useRest.ts#L56
`setOrigData(response.data);`
and it works as expected.

proddy, 2023-02-15:
great, I'll add and test this weekend when I'm back home. I didn't see this behaviour when running standalone (`yarn run standalone`)

proddy, 2023-02-18:
> Works good and looks nice, but after saving the changes the old-status is not updated, i have to leave the page and reload it to set this setting back.

I couldn't reproduce this? If I make a change, apply it, the form is updated. Can you give an example for me to try out?

MichaelDvP, 2023-02-18:
Go to settings, change a setting, let's say activate `log hexadecimal`
- the save button appears with one change,
- click save, the button disappears, so far all ok
- without leaving the page try to set back: deactivate `log hexadecimal`, the button does not appear
  (if you activate `log hex`, e.g. the saved setting, the button appears wrongly)

proddy, 2023-02-18:
bingo! got it, thanks @MichaelDvP

closed 2023-02-19 by proddy