emsesp/EMS-ESP32 #912: Rebooting Home Assistant Supervisor with MQTT Broker leads to permanent disconnect

zillion42, 2023-01-14:
Rebooting HA Supervisor with internal MQTT Broker leads to disconnect, obviously, the problem is, there is no reconnect after MQTT Broker becomes available again.
Testet with  EMS-ESP32 3.5.0b4 and HA Core 2023.1.2.

proddy, 2023-01-14:
I'll try and reproduce next week when I'm back home. Did you try the latest dev build? Are you using IPV6?

zillion42, 2023-01-14:
no IP6, no, I didnt try v3.5.0-dev.15, should I ?

proddy, 2023-01-14:
yes, we made lots of changes and I lost count. perhaps there is an MQTT restart fix in there

MichaelDvP, 2023-01-15:
I think it's something like in #756 and some other issues. Mostly it's not reproduceable, but seems to happen in some environments.

proddy, 2023-01-15:
I'll try to reproduce it by adding the MQTT service to HA and using this as a new LXC on my proxmox. And then shutting the whole LCX down and restarting it a few minutes later to see if EMS-ESP reconnects. If I can't reproduce I'll close this bug as an edge case.

zillion42, 2023-01-17:
I updated to  v3.5.0-dev.15 yesterday night and rebooted again, it all worked, and reconnected again.  EMS-ESP32 3.5.0b4 does not.

```
Jan 16 22:10:45 ems-esp mqtt 000+00:04:56.146 W 14: MQTT disconnected: TCP
Jan 16 22:12:20 ems-esp mqtt 000+00:06:31.251 I 15: MQTT connected
Jan 16 22:33:51 ems-esp telegram 000+00:28:02.879 W 16: Incomplete Rx: 10 0B FF 00 01 A5 00 B6 22 2C 4B 00 2C 23 00 1B 03 03 01 00 1B 03
```

proddy, 2023-01-17:
ok great, if it happens again we can re-open it.

closed 2023-01-17 by proddy