emsesp/EMS-ESP32 #913: Buderus GB125/RC300/310 - missing data

mp3junkie, 2023-01-15:
Hello,
the data points for the switch-on and switch-off times of the heating and hot water programs are not displayed.
What am I doing wrong? The programs for HK1 are missing.
[image]
Thanks for help.
Andy

MichaelDvP, 2023-01-16:
> What am I doing wrong?

Nothing. Program switchtimes for RC300 are not implemented. There were a ot of discussions about that (and holiday modes), but no useful result. It's to much data for mqtt:thermostat_data and web-dashboard. For RC35 i've implemented a small/simple way to set/read a single point at a time, but this results only in a lot of complains. Users want to have  a comfortable editor with all switchtimes. That's a lot of work, how to publish is unclear, and Bosch uses different schemes for RC35, RC300 and Junkers, so it's 3 times the effort.

mp3junkie, 2023-01-16:
Thank you for your answer.

mp3junkie, 2023-01-16:
Would it work with a KM200?

MichaelDvP, 2023-01-16:
> Would it work with a KM200?

I don't have one, @tp1de: Does KM200 support editing the program-switchtimes?

proddy, 2023-01-16:
> Nothing. Program switchtimes for RC300 are not implemented. There were a ot of discussions about that (and holiday modes), but no useful result. It's to much data for mqtt:thermostat_data and [...]

That answer deserves a spot on the FAQ page in the Wiki. I'll add it.

tp1de, 2023-01-16:
> Does KM200 support editing the program-switchtimes?

Yes it does. Like in my ioBroker Adapter km200 returns a JSON-structure for each switchtime element:

[{"dayOfWeek":"Mo","setpoint":"comfort2","time":300},{"dayOfWeek":"Mo","setpoint":"eco","time":1260},{"dayOfWeek":"Tu","setpoint":"comfort2","time":300},{"dayOfWeek":"Tu","setpoint":"eco","time":1260},{"dayOfWeek":"We","setpoint":"comfort2","time":300},{"dayOfWeek":"We","setpoint":"eco","time":1260},{"dayOfWeek":"Th","setpoint":"comfort2","time":300},{"dayOfWeek":"Th","setpoint":"eco","time":1260},{"dayOfWeek":"Fr","setpoint":"comfort2","time":300},{"dayOfWeek":"Fr","setpoint":"eco","time":1260},{"dayOfWeek":"Sa","setpoint":"comfort2","time":300},{"dayOfWeek":"Sa","setpoint":"eco","time":1260},{"dayOfWeek":"Su","setpoint":"comfort2","time":300},{"dayOfWeek":"Su","setpoint":"eco","time":1260}]

tp1de, 2023-01-16:
Nevertheless once getting the data there is not an easy way to maintain the switchtimes by frontend..
Since I am moving from ioBroker to HA I started to use the custom scheduler card from HACS.

It works with the climate elements for the hc's which come from mqtt discovery. Just changing from automatic mode (with switchtimes) to manual mode with the new switchtimes defined.  Negative impact is that in manual mode you loose some other automations. This has to be considered.

mp3junkie, 2023-01-16:
@proddy : I have problems with Discord. My account was deactivated immediately after registration. Another attempt with a different email address also failed and was deactivated. What can I do? I can not participate in the forum.

closed 2023-02-19 by proddy