emsesp/EMS-ESP32 #915: MQTT auto-discovery/publish broken for analog sensors attached to GPIOs lower than 10

gwilford, 2023-01-16:
### Description
Attaching an analog sensor to GPIO 4 results in "unknown" values in HA

### Steps to reproduce
Configure GPIO less than 10 (e.g. 4) as ADC in UI (no need to connect anything to the pin)
Check value of entity in HA - it will be "unknown"

### Expected behaviour
The value in HA should be 0 (or the measured value)

### The issue

Data from MQTT Explorer.
The MQTT auto-discovery config:
```
{
  "stat_t": "ems-esp/analogsensor_data",
  "val_tpl": "{{value_json['4'].value}}",
  "object_id": "analog_sensor_Garage ldr",
  "name": "Garage ldr",
  "uniq_id": "analogsensor_4",
  "unit_of_meas": "mV",
  "dev": { 
    "ids": ["ems-esp"]
  }
}
```

The MQTT publish:
```
{
  "04": {
    "name": "Garage ldr",
    "value": 60
  }
}
```

The config refers to **4** whereas the publish is for **04**

This is due to use of the following function in helpers.cpp for the publish but not for the config
```

// for decimals 0 to 99, printed as a 2 char string
char * Helpers::smallitoa(char * result, const uint8_t value) {
    result[0] = ((value / 10) == 0) ? '0' : (value / 10) + '0';
    result[1] = (value % 10) + '0';
    result[2] = '\0';
    return result;
}
```

Formatting the publish on line 375 of analogsensor.cpp:

`JsonObject dataSensor = doc.createNestedObject(Helpers::smallitoa(s, sensor.gpio()));`

Formatting the config on line 403:

`snprintf(str, sizeof(str), "{{value_json['%d'].value}}", sensor.gpio());`

Using smallitoa() to format the config or removing it from the publish should fix this.

Firmware v3.4.4

MichaelDvP, 2023-01-16:
Thanks for reporting. 
I think changing all to "04" looks better in formatting, but breaks HA by changing also the `uniq_id`. So changing the publish to "4" is better. Do you agree?

proddy, 2023-01-16:
agree. Thanks for the detailed breakdown @gwilford

gwilford, 2023-01-16:
Using "04" does look better.

It would break existing uniq_id but only for configs using low GPIOs which I don't think are working anyway. However, changing the publish from "04" to "4" might break anyone who had manually configured an MQTT subscribe in HA...

gwilford, 2023-01-16:
OK, just tested and can't seem to add a manual HA mqtt sensor with "0" prefix. This fails to parse:

```
mqtt:
  sensor:
    - name: "EMS-ESP GPIO4"
      state_topic: "ems-esp/analogsensor_data"
      value_template: "{{ value_json.04.value }}"
```

with `Invalid config for [mqtt]: invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'integer') for dictionary value @ data['mqtt']['sensor'][0]['value_template']. Got '{{ value_json.04.value }}'. (See ?, line ?).`

whereas this parses just fine:

```
mqtt:
  sensor:
    - name: "EMS-ESP GPIO4"
      state_topic: "ems-esp/analogsensor_data"
      value_template: "{{ value_json.4.value }}"
```

In summary, I don't think changing the publish to "4" would break anyone who had tried to manually configure within HA and it may well be safer.

MichaelDvP, 2023-01-16:
I think this should be `{{ value_json.['04'].value }}` to handle correctly. Or in code:
`snprintf(str, sizeof(str), "{{value_json['%02d'].value}}", sensor.gpio());`

gwilford, 2023-01-16:
Sorry, spoke too soon. This does currently work if manually configured as below:

```
mqtt:
  sensor:
    - name: "EMS-ESP GPIO4"
      state_topic: "ems-esp/analogsensor_data"
      value_template: "{{ value_json['04'].value }}"
```

Either fix works for me though.

proddy, 2023-01-22:
in dev-16

commit a38f4978 (2023-01-22): analogsensor publish wth single gpio number fix #915
closed 2023-01-22 by proddy