emsesp/EMS-ESP32 #917: Support for IVT GEO608C

dominikstupka, 2023-01-17:
Hi, i have heat pump IVT GEO 608C. EMS ESP board can communicate with heat pump i can see some values, temperatures, power etc. but i think that information coming from pump isnt complete. I can't control and also see set heating temperature, device mode like vacation/normal. Biggest issue is that i can't control device. If i change some value like dhw comfort mode nothing happens. EMS Write is ok without failure but mode doesn't change.

My machine consist this modules:

**Main board**
[image]
**WiFi module K30RF**
[image]
**Control panel UI800**
[image]
[image]
**Wireless thermostat TR120RF**
[image]

**Exported files from EMS-ESP DashBoard**
[device_entities (1).csv](https://github.com/emsesp/EMS-ESP32/files/10439404/device_entities.1.csv)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/10439283/log.txt)
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10439282/emsesp_info.txt)

Thanks for Help!

MichaelDvP, 2023-01-18:
Please try this build: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
and post another info.txt and log.

BTW: You have shower timer activated, this works only on flow type dhw-systems, but i think heatpumps always have a  dhw-buffer.

dominikstupka, 2023-01-18:
Now it seems perfect. I can control the thermostat remotely.

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10449137/emsesp_info.txt)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/10449138/log.txt)

MichaelDvP, 2023-01-19:
The mapping of integrated devices (0x48, 0x50) needs a fix, but both have no entities, so it's a cosmetic fix. I'll update my build.

Funny, once again bosch do not know what bosch do, the thermostat floods the bus with version requests on wifi-module, but can not map the answer.

proddy, 2023-01-22:
in dev-16

closed 2023-01-22 by proddy