emsesp/EMS-ESP32 #918: DHW flow rate in UBAMonitorWWPlus(0xE9) telegram

gwilford, 2023-01-19:
The controller display on my Greenstar 8000i can show DHW flow rate via the service menus (L.1 b2) in l/min so I went hunting for it in the WW telegrams as existing EMS-ESP flow rate entities don't map on this boiler.

I found DHW flow rate  at offset 11 in 0xE9. Needs dividing by 10. e.g.

0x36 = 54,  5.4 l/min
0x2F = 47, 4.7 l/min
0x31 = 49, 4.9 l/min

Partial log trimmed for brevity:

```
In this position-------------------------------------------------||
(0x08) -> (0x00), (0xE9), data: 3A 01 CB 00 00 80 00 01 CB 46 00 00 00 02 00 61 AD 00 43 F7 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 01 F2 00 00 80 00 01 F2 46 00 36 11 22 00 61 AD 00 43 F7 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 01 FD 00 00 80 00 01 FD 46 00 2F 11 22 00 61 AE 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 05 00 00 80 00 02 05 46 00 31 11 22 00 61 AE 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 15 00 00 80 00 02 15 46 00 36 11 22 00 61 AE 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 1D 00 00 80 00 02 1D 46 00 36 11 22 00 61 AE 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 27 00 00 80 00 02 27 46 00 36 11 22 00 61 AF 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3B 00 00 80 00 02 3B 46 00 37 11 22 00 61 AF 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 40 00 00 80 00 02 40 46 00 35 11 22 00 61 AF 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 43 00 00 80 00 02 43 46 00 35 11 22 00 61 AF 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 43 00 00 80 00 02 43 46 00 36 11 22 00 61 AF 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 40 00 00 80 00 02 40 46 00 36 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 45 00 00 80 00 02 45 46 00 36 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 44 00 00 80 00 02 44 46 00 35 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 46 00 00 80 00 02 46 46 00 36 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 48 00 00 80 00 02 48 46 00 35 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 46 00 00 80 00 02 46 46 00 35 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 47 00 00 80 00 02 47 46 00 36 11 22 00 61 B0 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 44 00 00 80 00 02 44 46 00 34 11 22 00 61 B1 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 46 00 00 80 00 02 46 46 00 35 11 22 00 61 B1 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 49 00 00 80 00 02 49 46 00 36 11 22 00 61 B1 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 48 00 00 80 00 02 48 46 00 37 11 22 00 61 B1 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 44 00 00 80 00 02 44 46 00 38 11 22 00 61 B1 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 43 00 00 80 00 02 43 46 00 38 11 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 44 00 00 80 00 02 44 46 00 38 11 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 48 00 00 80 00 02 48 46 00 36 11 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 4A 00 00 80 00 02 4A 46 00 00 20 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3D 00 00 80 00 02 3D 46 00 00 20 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 33 00 00 80 00 02 33 46 00 00 20 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 2D 00 00 80 00 02 2D 46 00 00 20 22 00 61 B2 00 43 F8 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 2B 00 00 80 00 02 2B 46 00 38 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 0A 00 00 80 00 02 0A 46 00 35 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 04 00 00 80 00 02 04 46 00 34 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 04 00 00 80 00 02 04 46 00 35 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 0A 00 00 80 00 02 0A 46 00 36 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 0E 00 00 80 00 02 0E 46 00 38 11 22 00 61 B3 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 1E 00 00 80 00 02 1E 46 00 38 11 22 00 61 B4 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 2A 00 00 80 00 02 2A 46 00 37 11 22 00 61 B4 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 30 00 00 80 00 02 30 46 00 38 11 22 00 61 B4 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 37 00 00 80 00 02 37 46 00 36 11 22 00 61 B4 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3B 00 00 80 00 02 3B 46 00 38 11 22 00 61 B4 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3B 00 00 80 00 02 3B 46 00 3B 11 22 00 61 B5 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3C 00 00 80 00 02 3C 46 00 39 11 22 00 61 B5 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 3C 00 00 80 00 02 3C 46 00 3A 11 22 00 61 B5 00 43 F9 00 80 00 3A 01 00
(0x08) -> (0x00), (0xE9), data: 3A 02 37 00 00 80 00 02 37 46 00 00 20 22 00 61 B5 00 43 F9 00 80 00 3A 01 00
```

Please can you add this as a new entity?

proddy, 2023-01-21:
Sure can!

MichaelDvP, 2023-01-21:
I think we can add it to v3.5, the `wwcurflow` is already registered and read for ems1.0 devices, also ems+ telegram 0x9E is read, so it's a single read and takes no extra memory.

proddy, 2023-01-21:
> I think we can add it to v3.5, the `wwcurflow` is already registered and read for ems1.0 devices, also ems+ telegram 0x9E is read, so it's a single read and takes no extra memory.

if it's easy lets add it to 3.5.0.

closed 2023-01-22 by proddy

proddy, 2023-01-22:
in dev-16