emsesp/EMS-ESP32 #920: Dashboard showing parameters as 'read only'

guardianbs, 2023-01-20:
I don't know if this an error or a misunderstanding on my part.

My Greenstar Confort 2 Thermostat (recognized as CRF200S) has a very limited interface with the boiler and only has one command to change values, UBASetPoints(0x1A)

This changes the parameters 'burner set power' (reported as sensor.boiler_burner_set_power to HA) and 'set flow temperature' (reported as sensor.boiler_set_flow_temperature).

Both of these parameters are shown as 'read-only' on the EMS-ESP dashboard and hence generate sensors for HA. Is this correct as these values are obviously being written by the Thermostat ?

The Settings / Customization menu has the ability to set 'Write-able' parameters as 'read only' but not the reverse (could be nice).

[image]

{
  "System Status": {
    "version": "3.4.2b4",
    "uptime": "069+06:51:51.817",
    "freemem": 139,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/London"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 1993589,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 2,
    "API fails": 1
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 9127673,
    "bus reads (tx)": 399103,
    "bus writes (tx)": 2,
    "bus incomplete telegrams": 1172,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "12.10",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "CRF200S",
      "device id": "0x18",
      "product id": 216,
      "version": "01.03",
      "entities": 7,
      "handlers received": "0x06",
      "handlers fetched": "0x02A5",
      "handlers pending": "0xA3 0xA2 0x12 0x02A6 0x02A7 0x02A8"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "12.10",
      "entities": 0
    }
  ]
}

proddy, 2023-01-21:
The entities 'burner set power' and 'set flow temperature' are read only telegrams in the code, for everyone. 

The entities for writing the values are called 'selected flow temperature' and 'burner selected max power'.

I know the names are a bit misleading.

I haven't checked the code but I think you should be seeing these two. Which version are you on?

guardianbs, 2023-01-21:
Thank you for the swift response.

I'm using version 3.4.2b4.

It's all a little confusing ! From what I can observe the following is the sequence of events:

When the Thermostat generates, for example, UBASetPoints(0x1A), data: 55 40 (flow 85C, power 64%) or 'burner selected max power' is changed from the Dashboard

 - 'burner set power' is updated immediately
 - 'burner selected max power' is updated to the following value based on the conditions
   - 'burner set power' if 'burner set power' is not greater than 'burner max power'
   - 'burner max power' if 'burner set power' is greater than 'burner max power'
   - Zero if 'burner set power' is less than the minimum at which the Boiler can modulate
   
The 'burner current power' changes (sometimes slowly) to match 'burner selected max power' unless constrained by 'current flow temperature' exceeding 'selected flow temperature'.

There seems to be some Boiler logic that determines how quickly 'burner selected max power' follows 'burner set power' and 'burner current power' follows 'burner selected max power'.

On the basis that the UBASetPoints command updates 'burner set power' and Boiler logic updates 'burner selected max power' and 'burner current power' it would seem more logical for 'burner set power' to be the update field to generate the UBASetPoints command rather than 'burner selected max power'.

I'm guessing that it could be the same logic for 'set flow temperature' and 'selected flow temperature' but I cannot observe these as 'selected flow temperature' is set by the physical knob on the Boiler and the 'set flow temperature' sent with the UBASetPoints command does not change it.

On the assumption that my observations are correct (big assumption) I guess it doesn't matter which of the two fields are used to change the Burner Power if (big if) one understands what's going on.

guardianbs, 2023-01-27:
All done I guess . . .

proddy, 2023-01-27:
think so. should it be a FAQ article?

guardianbs, 2023-01-28:
Hmmm . . .

closed 2023-02-19 by proddy