emsesp/EMS-ESP32 #921: WiFi disconnected. REASON code=auth expire

FredFeuerstein22, 2023-01-21:
The ems-esp regularly loses the WLAN connection and then also does not find it again.

The settings in the ESP-EMS should fit, since they do not change during use. Maybe one of the settings in my UNIFI system is wrong? DHCP, lease-Time, ...? But even with static IP ESÜ-EMS does not find back to the WLAN... Does anyone have an idea?

proddy, 2023-01-21:
A few questions
- which version of EMS-ESP are you using?
- what is the wifi strength of EMS-ESP, especially when it tries to reconnect
- is EMS-ESP roaming between access points?
- have you configured Unifi correctly for iot devices? do you mesh or roaming enabled?
- can you easily reproduce this so we can too?

proddy, 2023-01-21:
and I just realized from the title. the "auth expire" error is common bug on the ESP32 and Unifi networks when trying to join a new AP. There are a lot of posts about it, and no real good solutions.

FredFeuerstein22, 2023-01-21:
regarding your questions:
- latest version is used
- it´s about 65 db
- yes, roaming is allowed and it happend to a AP with worse wifi strength
- what do you mean with "correctly"? mesh and roaming are enabled
- after a certain time it happens

proddy, 2023-01-21:
and which circuit are you using? is it a bbqkees interface gateway?

proddy, 2023-01-22:
Some related issues to look at https://github.com/esphome/issues/issues/1371

FredFeuerstein22, 2023-01-22:
yes, it is a bbqkees interface gateway

FredFeuerstein22, 2023-02-06:
any additional ideas?

bbqkees, 2023-02-07:
What I have done in the past is set a SSID on a fixed channel on the closest AP and force the Gateway to only connect to that specific AP. Can all be done from the Unifi controller.

proddy, 2023-02-07:
thats one work around. Another way is to clean the nvm on the ESP32 chip by erasing the flash, and then you can connect to a different AP. But as soon as you roam the problem will come back. Maybe it's fixed in the latest 6.0.1 espressif core wifi code but I haven't tried it yet. Or in the Unifi beta firmware. It's not a bug in EMS-ESP but I'll try and find a fix

proddy, 2023-02-10:
worth trying with the latest 3.6.0 dev build if you want to download and compile yourself. it uses the latest espressif libs

closed 2023-04-23 by proddy