emsesp/EMS-ESP32 #922: custom telegram handler

proddy, 2023-01-21:
> Very long time ago I made a proposal to introduce "customizable" entities by telegram-type and offset. Maybe still an idea to support special installations and requirements.

I don't recall that, but its really good idea. We could add to the Customization page the ability to add a new Telegram to the EMS device in which you choose a name, the type (short, int), the divider (div10, div2 etc) and position. This would only be for read telegrams though.

_Originally posted by @proddy in https://github.com/emsesp/EMS-ESP32/issues/908#issuecomment-1399201035_

proddy, 2023-01-21:
What do you think @MichaelDvP ?

tp1de, 2023-01-21:
From my point of view you should decide / consider the following points:

- How to secure overall ems-esp stability and memory usage? What are the impacts of additional functions / languages on memory and cpu load? (b14 is stable b15 was not ...)
- Should this function (custom telegram handlers) be implemented in ems-esp code or in the home automation system (HA / ioBroker etc). Within ioBroker I implemented some months ago but made it hidden because it needs a lot of knowhow about telegram structure. (ioBroker adapter is used by approx. 550 people). It works for read / write on hex telegrams and even for telegram types not supported by ems-esp yet (holiday periods / switchtimes for my RC310)
- I could implement for HA too using node-red (to keep my JS code). I already implemented my km200 coding to HA including the power consumption recordings into the energy dashboard of HA.
- While implementing in home automation system I used an imbedded syslog server to analyse all telegrams from ems-esp. While using this way it would be good to add a filter within syslog function in ems-esp to reduce the number of telegrams send and therefore reducing the cpu load within home automation system.

tp1de, 2023-01-21:
That's the way the parameters are entered in ioBroker. I used it to test the functionality:
[image]

MichaelDvP, 2023-01-21:
Yes, i think it's possible.  I see two possible ways:
1. add it to customization device. Check in `EMSdevice::handle_telegram` for the custom entity, process and then look for normal telegrm processing.
2. add entity extra with parameters device_id, telegram, offset, type, divider and process in `EMSESP::process_telegram`. This allows to process parameters from unknown devices.

MichaelDvP, 2023-01-21:
But i'm not sure that this is really helpfull. I've tried something like this a while ago with some unknown settings/values from solar module (see https://github.com/emsesp/EMS-ESP32/blob/88404bcb56d6be0e79cb2dda2fc1931be0102e4d/src/devices/solar.cpp#L227-L237) without success. Changing the settings have no effect and the values are used for a internal calculation for dhw correction.

I'm sure we'll see a lot of F7/F9/BE/BF telegrams watched, but very rarly a usefull value. It's more for power-users/devolpers. Maybe use only a telnet command to add such a custom handler? Or limit it to 5 entites, usefull values should be reported here and added to the core.

tp1de, 2023-01-21:
I agree to Michael's comment. I added this function to my ioBroker adapter to store values of unknown telegrams and track changes. This was without success and I took it out. Better to observe the telegrams while changing values on thermostat or boiler.

MichaelDvP, 2023-01-21:
@tp1de
>  (b14 is stable b15 was not ...)

I think i found it (or one possible cause). Please set entity-language to en or try my new dev build.

tp1de, 2023-01-21:
@MichaelDvP testing b15 with entity language set to en is enough?

gwilford, 2023-01-21:
>  Better to observe the telegrams while changing values on thermostat or boiler.

Hey guys, the above is what I had in mind when I wrote [this](https://github.com/emsesp/EMS-ESP32/issues/908#issuecomment-1399230122)

proddy, 2023-01-21:
What triggered this enhancement was this issue from @gwilford (#918 - DHW flow rate found in UBAMonitorWWPlus(0xE9) telegram). Instead of getting a flood of requests to add more data elements, we could enable the end-user to add these manually and if there was enough interest bring it into the main code. Nothing to do with nodered, km200 or scaping the syslog file (which is nasty!). I'm just collecting ideas for the future.