emsesp/EMS-ESP32 #928: boiler_return_temperature vs boiler_heat_carrier_return_tc0

HansRemmerswaal, 2023-01-23:
**Question**

In HA I see the following two sensors having ALWAYS the same value.
sensor.boiler_return_temperature
sensor.boiler_heat_carrier_return_tc0

Are these two sensors for the same or are they actually different sensors?

**Screenshots**
[image]

**Device information**
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10477636/emsesp_info.txt)

**Additional context**

MichaelDvP, 2023-01-23:
> sensor.boiler_return_temperature

is from telegram E4 offset 7
> sensor.boiler_heat_carrier_return_tc0

is from telegram 48F offset 6

Maybe one sensor published twice by the bus.

HansRemmerswaal, 2023-01-23:
Looks like... and thanks for your quick reply. Will close it.

closed 2023-02-19 by proddy