emsesp/EMS-ESP32 #930: support for virtual room remote control

SpaceTeddy, 2023-01-24:
Dear Team,
i'm just thinking about the feature to use a "virtual remote control" like the small room remote controller devices RC20, RC20RF,.... 
Reason: a) RC20RF controller ist really expensive >200€.  b) clever guy's can develop an open source wifi or RF telegram based controller for people who wants to use a room temp. based controll mode. This open source controller could also read data from EMS-ESP or mqtt server to show boiler data, time, ...   
This virtual remote control could be activated or deacivated in the EMS-ESP web menu ect.

Do you think this could be a nice feature for the future?

kr,
Chris

MichaelDvP, 2023-01-24:
If you have RC35 as main thermostat, it's already implemented. For remote control the heatingcircuit hcx:
Set `thermostat/hcx/control` to 1 (RC20) 
Send the room temperature from any sensor (i use a aqara zigbee) to `thermostat/hcx/remotetemp` 
You will see a new device as RC20 in your device list.
To stop, set control to 2 (RC35) and set remotetemp to <0 or >100 (i normally use -1) (The recognized device stays in list)

proddy, 2023-01-24:
that's worth a write up in the wiki too. Those Aqara sensors & hub are really nice

SpaceTeddy, 2023-01-24:
@MichaelDvP 
sry to ask again for more detailed information...
Basically I have a RC20RF control, which is set as HC1 at my Sieger es73 controller.
[image]
This is working great and no special setup step was needed. To used an non EMS-Bus temp sensor, please explain more detailed how to use the HCX topic. 

- If no remote control (RC20/RC20RF) is installed to the RC35 controller, EMS-ESP shows an 'hcx/control device' topic, which I have to set to 1 in the web interface, or via telnet?
- if so, how do I send the temperatre value to hcx/remotetemp? via mqtt, telnet, or web interface? becasue with my es73 controller, there is no topic  hcx/remotetemp? 

thx.
kr,
Christian

MichaelDvP, 2023-01-24:
You can not send a temperature to a thermostat as roomtemp. There are some newer RC300 with rf sensor, using a wireless gateway at id 0x40 and the sensor sends from id50, but we have never seen the complete protocol how this works.
For RC35 as master it is possible to use one rmemote roomthermostat for each hc. If you already have a RC20RF you know how this works. In RC35 there are some settings to change: is the hc controlled by rc35 or the remote rc20RF (hcx/control), Is the hc controlled by outdoortemp or roomtemp, or outdoor with roominfluence, etc. For roominfluence you also have to set roominfluence > 0, and reset to 0 if you use rc35 only.
You can have one RC20 for each hc as remote control, if you have the Rc20RF already, you can not set a different temperature to this hc. You can only uninstall the RC20RF and start the RC20 emulation on ems-esp.
The ems-esp simply emulates a rc20 on the bus. For hc1 it registered with a rc20 product id and device_id 0x18 and sends the the temperature with telegram 0xAF to the bus. 

BTW: for junkers we have a log with FB10 as remote roomcontroller on hc1, we could add emulation to ems-esp, but we don't know how the protocol is for hc2, hc3, ..

SpaceTeddy, 2023-01-24:
alright, thanks a lot for your explaination and quick feedback.

yeah, basically I know how the RC20RF works in my environment. It is registered as device id 0x18 and sends  5 bytes telegram. I guess it is the temperature itself and RSSI or LQI value for sender & receiver (last 2 bytes).

```
020+16:18:23.748 N 0: [emsesp] Thermostat(0x18) -B-> All(0x00), RC20Remote(0xAF), data: 00 B1 00 04 06
``` 

Also some flags for WW on time is send via. 00 00 is OFF -  00 02 is wwontime ON
```
Thermostat(0x18) -W-> Boiler(0x08), UBAFlags(0x35), data: 00 00
```

If I give action to the RC20RF itself, it writes direct to the HC1 TypeId. Can not remember, but is is 0x3E or 0x3D.

But back to my previous question... how i can activate the RC20 emulation in EMS-ESP and how do I feed the EMS-ESP with temperature data in ```thermostat/hcx/remotetemp``` ? Or can I not see that in the Web interface or telnet, becasue it is hidden for my Sieger es73 controller? ;)

thanks again
Chris

MichaelDvP, 2023-01-24:
The ES73 should have the remottemp (it's mostly RC35 compatible), but if not set, it is not shown in dashboard, but via mqtt and api you can send a value.
Check Settings-Customizations, you should see:
[image]
Also for `http://ems-esp.local/api/thermostat/hc2/remotetemp` it shows:
```
{
  "name": "remotetemp",
  "fullname": "hc2 room temperature from remote",
  "circuit": "hc2",
  "type": "number",
  "min": -1,
  "max": 101,
  "uom": "°C",
  "readable": true,
  "writeable": true,
  "visible": true,
  "value": "not set"
}
```
You activate the emulation by sending via mqtt or api a valid value (0-100) to remotetemp, you deactivate the emulation by sending -1 or 101. **Do not activate on the heatingcircuit where you have the RC20RF!** This gives bus collision.

MichaelDvP, 2023-01-24:
> It is registered as device id 0x18 and sends 5 bytes telegram. I guess it is the temperature itself and RSSI or LQI value for sender & receiver (last 2 bytes).

Yes, the cable RC20 sends only 3 bytes, last always 0, first two the temperature.  
> Also some flags for WW on time is send via. 00 00 is OFF - 00 02 is wwontime ON

Yes, that is a boiler command, emsesp-boiler/wwonetime is the same command.

>  it writes direct to the HC1 TypeId. Can not remember, but is is 0x3E or 0x3D.

Its 0x3D for setpoint roomtemp of hc1, for hc2 it's 47, ... With emsesp you set the setpoint direct to the ES73 `thermostat/hc1/seltemp`

> 020+16:44:57.493 N 256: [emsesp] Thermostat(0x18) -R-> Controller(0x09), ?(0x29), length: 0x01
020+16:44:57.512 N 257: [emsesp] Controller(0x09) -W-> Thermostat(0x18), ?(0x29), data: 6B

Don't know, maybe the thermostat checks button status of wwonetime button on controller. My MC10 controller replys empty if i check `read 9 29`.

SpaceTeddy, 2023-01-24:
@MichaelDvP 
you're right, api command shows:
```
{
  "name": "remotetemp",
  "fullname": "hc1 room temperature from remote",
  "circuit": "hc1",
  "type": "number",
  "min": -3199,
  "max": 3199,
  "uom": "°C",
  "readable": true,
  "writeable": true,
  "visible": true,
  "value": "not set"
}
```

unfortunatly no mqtt topic:
```
{
  "lastcode": "A2(806) 20.01.2023 13:54 (0 min)",
  "datetime": "24.01.2023 16:05",
  "display": "external temperature",
  "language": "german",
  "clockoffset": 0,
  "intoffset": 0,
  "minexttemp": -10,
  "dampedoutdoortemp": 2,
  "building": "medium",
  "wwmode": "auto",
  "wwcircmode": "auto",
  "wwcircprog": "std prog",
  "wwdisinfecting": "off",
  "wwdisinfectday": "mo",
  "wwdisinfecthour": 4,
  "wwmaxtemp": 60,
  "wwswitchtime": "00 mo 04:30 on",
  "wwcircswitchtime": "00 mo 05:30 on",
  "wwholidays": "01.01.2000-01.01.2000",
  "wwvacations": "01.01.2000-01.01.2000",
  "hc1": {
    "seltemp": 19,
    "currtemp": 18.8,
    "mode": "auto",
    "modetype": "day",
    "daytemp": 19,
    "nighttemp": 17.5,
    "designtemp": 70,
    "offsettemp": 0,
    "holidaytemp": 17,
    "targetflowtemp": 42,
    "summertemp": 17,
    "summermode": "winter",
    "holidaymode": "off",
    "nofrosttemp": 0,
    "nofrostmode": "room",
    "roominfluence": 0,
    "minflowtemp": 5,
    "maxflowtemp": 45,
    "flowtempoffset": 5,
    "reducemode": "reduce",
    "control": "RC20",
    "holidays": "01.01.2000-01.01.2000",
    "vacations": "06.01.2000-01.01.2000",
    "program": "own 1",
    "pause": 0,
    "party": 0,
    "wwprio": "on",
    "switchtime1": "00 mo 05:10 on"
  }
}
```

But I will check the hc1 topics if the external RC20RF is deactivated.

BTW: read data the RC20RF gateway...
```
ems-esp:$ read 9 29
020+17:58:26.410 I 0: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x29), data: 6B
ems-esp:$ 
```

SpaceTeddy, 2023-01-24:
btw: scan of Device ID: 0x09

```
ems-esp:$ read 9 01
020+17:59:45.235 I 1: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x01), data: <empty>
ems-esp:$ read 9 02
020+17:59:48.402 I 2: [emsesp] Controller(0x09) -W-> Me(0x0B), Version(0x02), data: 44 02 03
ems-esp:$ read 9 03
020+17:59:56.657 I 3: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x03), data: <empty>
ems-esp:$ read 9 04
020+18:00:00.368 I 4: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x04), data: <empty>
ems-esp:$ read 9 05
020+18:00:02.052 I 5: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x05), data: <empty>
ems-esp:$ read 9 06
020+18:00:03.771 I 6: [emsesp] Controller(0x09) -W-> Me(0x0B), RCTime(0x06), data: <empty>
ems-esp:$ read 9 07
020+18:00:06.705 I 7: [emsesp] Controller(0x09) -W-> Me(0x0B), UBADevices(0x07), data: <empty>
ems-esp:$ read 9 08
020+18:00:08.718 I 8: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x08), data: <empty>
ems-esp:$ read 9 09
020+18:00:11.008 I 9: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x09), data: <empty>
ems-esp:$ read 9 0A
020+18:00:13.874 I 10: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0A), data: <empty>
ems-esp:$ read 9 0B
020+18:00:16.805 I 11: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0B), data: <empty>
ems-esp:$ read 9 0C
020+18:00:19.265 I 12: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0C), data: <empty>
ems-esp:$ read 9 0D
020+18:00:22.426 I 13: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0D), data: <empty>
ems-esp:$ read 9 0E
020+18:00:25.113 I 14: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0E), data: <empty>
ems-esp:$ read 9 0F
020+18:00:27.723 I 15: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x0F), data: <empty>
ems-esp:$ read 9 10
020+18:00:34.790 I 16: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAErrorMessage1(0x10), data: <empty>
ems-esp:$ read 9 11
020+18:00:37.302 I 17: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAErrorMessage2(0x11), data: <empty>
ems-esp:$ read 9 12
020+18:00:39.315 I 18: [emsesp] Controller(0x09) -W-> Me(0x0B), RCErrorMessage(0x12), data: <empty>
ems-esp:$ read 9 13
020+18:00:41.631 I 19: [emsesp] Controller(0x09) -W-> Me(0x0B), RCErrorMessage2(0x13), data: <empty>
ems-esp:$ read 9 14
020+18:00:43.047 I 20: [emsesp] Controller(0x09) -W-> Me(0x0B), UBATotalUptime(0x14), data: <empty>
ems-esp:$ read 9 15
020+18:00:44.243 I 21: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAMaintenanceData(0x15), data: 00
ems-esp:$ read 9 16
020+18:00:46.633 I 22: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAParameters(0x16), data: FF 5A
ems-esp:$ read 9 17
020+18:00:49.764 I 23: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x17), data: <empty>
ems-esp:$ read 9 18
020+18:00:52.080 I 24: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAMonitorFast(0x18), data: <empty>
ems-esp:$ read 9 19
020+18:00:55.586 I 25: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAMonitorSlow(0x19), data: <empty>
ems-esp:$ read 9 1A
020+18:00:59.961 I 26: [emsesp] Controller(0x09) -W-> Me(0x0B), UBASetPoints(0x1A), data: <empty>
ems-esp:$ read 9 1B
020+18:01:02.653 I 27: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x1B), data: <empty>
ems-esp:$ read 9 1C
020+18:01:05.139 I 28: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAMaintenanceStatus(0x1C), data: <empty>
ems-esp:$ read 9 1D
020+18:01:14.440 I 29: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x1D), data: <empty>
ems-esp:$ read 9 1E
020+18:01:17.225 I 30: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x1E), data: <empty>
ems-esp:$ read 9 1F
020+18:01:19.938 I 31: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x1F), data: <empty>
ems-esp:$ read 9 20
020+18:01:23.942 I 32: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x20), data: <empty>
ems-esp:$ read 9 21
020+18:01:27.798 I 33: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x21), data: <empty>
ems-esp:$ read 9 22
[... 162 lines cut ...]
020+18:06:53.914 I 115: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x74), data: <empty>
ems-esp:$ read 9 75
020+18:06:55.082 I 116: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x75), data: <empty>
ems-esp:$ read 9 76
020+18:06:56.554 I 117: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x76), data: <empty>
ems-esp:$ read 9 77
020+18:06:58.469 I 118: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x77), data: <empty>
ems-esp:$ read 9 78
020+18:07:00.482 I 119: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x78), data: <empty>
ems-esp:$ read 9 79
020+18:07:02.350 I 120: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x79), data: <empty>
ems-esp:$ read 9 7A
020+18:07:05.310 I 121: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7A), data: <empty>
ems-esp:$ read 9 7B
020+18:07:07.549 I 122: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7B), data: <empty>
ems-esp:$ read 9 7C
020+18:07:10.982 I 123: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7C), data: <empty>
ems-esp:$ read 9 7D
020+18:07:14.215 I 124: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7D), data: <empty>
ems-esp:$ read 9 7E
020+18:07:16.778 I 125: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7E), data: <empty>
ems-esp:$ read 9 7F
020+18:07:19.089 I 126: [emsesp] Controller(0x09) -W-> Me(0x0B), ?(0x7F), data: <empty>
ems-esp:$ 
```

MichaelDvP, 2023-01-24:
> unfortunatly no mqtt topic:

sure, no published value: It has no active value! It is published **after** you send a valid value.
The subscription works, you have to send the value to ems-esp! Use mqtt-explorer to test.
> btw: scan of Device ID: 0x09

0x02: version as shown in ems-esp-info, 0x16 heating on, 90°C max, 0x33 dhw on. 0x29 unknown.

SpaceTeddy, 2023-01-24:
Sry to ask a beginner question again. May you please explain how to write ems-esp/thermostat_hc1/remotetemp value as a single topic? 
I tried to publish with MQTT -explorer in the following way without success
Topic:
ems-esp/thermostat_data_hc1/remotetemp 
Raw: 20.0

Topic:
ems-esp/thermostat_data_hc1/remotetemp 
Raw: “20.0“

Topic:
ems-esp/thermostat_data_hc1
Raw: remotetemp 20.0

Topic:
ems-esp/thermostat_data_hc1
JSON: {“remotetemp”:20.0}

proddy, 2023-01-24:
there's some examples in https://emsesp.github.io/docs/Commands/#mqtt

it's `ems-esp/thermostat/hc1/remotetemp`

SpaceTeddy, 2023-01-25:
@proddy 
thanks a lot. works fine now.
-> closed

closed 2023-01-25 by SpaceTeddy