emsesp/EMS-ESP32 #931: Boolean format API/MQTT options 1/0 and true/false don't work as expected

pswid, 2023-01-26:
Latest v3.5.0-dev.16. 

If I choose the `1/0` option as the **Boolean Format API/MQTT**, the quotation marks are added anyway in MQTT, so the option works the same as the option `"1"/"0"`. The same problem with `true/false` option.

MichaelDvP, 2023-01-26:
Only if you are using HA. That's another stupid limitation of HA. Disable HA and you get the native format.

pswid, 2023-01-26:
Can you write something more about this HA limitation? Maybe there's a way to workaround it...

MichaelDvP, 2023-01-26:
No, i'm not using HA. Proddy has spend a lot of time and work to add code and comments for the HA extras. I'm only reading the comments (as quoted).

proddy, 2023-01-26:
It's a limitation of HA's MQTT Discovery (e.g. https://www.home-assistant.io/integrations/switch.mqtt/) which requires `payload_on` and `payload_off` as being Strings in the MQTT payload. I'll experiment a few workarounds.

..but if you're using Home Assistant you shouldn't really care what the API/MQTT format is? Do you have a use-case @pswid that I've overlooked? Like do you need numeric 1 and 0?

pswid, 2023-01-26:
in fact `payload_on` and `payload_off` doesn't need to be a string. In case of **"on/off"** option selected, just change in MQTT discovery config, from:
```
  "payload_on": "true",
  "payload_off": "false",
```

to:

```
  "payload_on": true,
  "payload_off": false,
```

remove the condition:
https://github.com/emsesp/EMS-ESP32/blob/ede8e7dfceb5012dcd13fb1589e51470373c5328/src/emsdevice.cpp#L1528-L1530
and boolean option **"on/off"** will start working (I've just tested it). Option **"0/1"** too! Howewer it would also be good to remove quotes from payload_on/off definitions (just for more "clarity").

... I'm using API/MQTT not only with HA.

proddy, 2023-01-26:
ok great, that's what I was hoping for. Home Assistant finally fixed it (I promise it didn't use to be like this!)

pswid, 2023-01-27:
@proddy do you want me to try to prepare a fix?

proddy, 2023-01-28:
If you have time that would be great. Do test all scenarios.

pswid, 2023-01-28:
ok I will. hopefully this weekend

pswid, 2023-01-29:
done. everything seems to work fine, though if you select option `true/false`, HA sends `True/False` (uppercase first letter) instead. This is not a problem, because `Helpers::value2bool` is case-insensitive (executes `toLower(value)` at the beginning).

proddy, 2023-01-29:
Nice! I'll try it out.

closed 2023-01-30 by proddy