emsesp/EMS-ESP32 #934: Junkers  boilers don't set any command

matteopedani, 2023-01-28:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/emsesp/EMS-ESP32/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

**Additional context**
*Add any other context about the problem here.*

matteopedani, 2023-01-28:
I have a junkers boiler but I can't send any command, but I can read all state of boilers. The boiler stay always ON. And there is no way to switch off. (I'm using Home Assistant integration) 

Can you give me the right command to switch off

bbqkees, 2023-02-09:
See also:  #951

closed 2023-02-19 by proddy