emsesp/EMS-ESP32 #938: Possibility to list all the boiler error codes

robertek, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
Sometimes the boiler may present multiple error codes. Sometimes even one error code, which is still valid, is masked with newer error code which happened after the running error but last only for a short time. This prevents investigating error codes from EMS-ESP at all and the possibility of reset them.

**Describe the solution you'd like**
The thermostat shows all the error codes present in system. It is a question if it has memory for all of them and snoop any error notification and then present them on the duration basis (or let say missing error ending notification). Or it has some EMSs to query all the present errors. I expect the first one is more probable. 
On this observation the final implementation will be based.

**Additional context**
I have a IVT AIR X HP which will need to have diagnosed the circuit between outdoor and indoor unit. Probably replace the weak circulation pump. Meanwhile this is done I'm experiencing regular defrost issues, in lower freezing temperatures it may be multiple errors. So I'm able to snoop that out and provide some information for correct implementation of diagnosis of multiple error issues. And also being able to reset them correctly.

bbqkees, 2023-01-30:
AFAIK only the last error code is sent over the bus, previous ones are overwritten.
EMS-ESP does not log data by itself, it passes it on to your home automation or smart home solution.
If you connected the Gateway to f.i. Home Assistant you can just log it there.

proddy, 2023-01-30:
that is the best solution. the ESP32 has limited memory and not enough to store historic data

elRadix, 2023-01-30:
can be done in home assistant using a template sensor and a log card
you need to consult the manual of your boiler  to be able to add the correct error code 

[image]

```
    ## ------------------------------------------------------------------
    ## service code number text for Bosch i7000iw 
    ## ------------------------------------------------------------------
    heating_service_code_message:
      friendly_name: "Heating message"
      attribute_templates:
         service-code: "{{states('sensor.boiler_servicecodenumber')|int(0)}}"
         timestamp: "{{now().strftime('%Y-%m-%d %H:%M:%S')}}"
         date: >
             {{ now().timestamp() | timestamp_custom("%Y-%m-%d", true) }}
         time: >
            {{ now().timestamp() | timestamp_custom("%H:%M:%S", true) }}
      value_template: >-
          {% set mapper = {
              '208':'208-The device is in chimney sweep mode. Chimney sweep mode is automatically deactivated after 30 minutes',
              '200':'200-The device is in heating mode', 
              '201':'201-Device in hot water mode',
              '202':'202-Cycle lock active: The time interval for switching the burner on again has not yet been reached',
              '305':'305-Holding time: The time interval for keeping the water warm has not yet been reached',
              '283':'283-The burner is starting',
              '265':'266-The heat requirement is lower than the minimum heat output of the device. The device works in on/off mode',
              '203':'203-The device is ready for operation, there is no heat requirement',
              '284':'284-The gas valve is opened, first safety time',
              '270':'270-The device electronics are starting up',
              '204':'204-The current flow temperature is higher than the target flow temperature. The boiler is switched off',
              '357':'357-Venting function active',
              '358':'358-Blocking protection for heating pump and 3-way valve active',
              '212':'212-Gradient monitoring heating flow has responded',
              '341':'341-Gradient limitation: Temperature rise too fast in heating mode',
              '342':'342-Gradient limitation: Temperature rise too fast in hot water mode',
              '282':'282-No speed feedback from the heating pump',
              '268':'268-The device is in test mode',
              '276':'276-The temperature at the flow temperature sensor is > 95 °C',
              '359':'359-The temperature at the hot water temperature sensor is too high',
              '341':'341-Gradient limitation: temperature rise too fast in heating mode',
              '281':'281-The heating pump does not generate any pressure',
              '264':'264-Fan failed.',
              '273':'273-The burner and blower have been in continuous operation for 24 hours and will be shut down for a short time for safety checks',
              '224':'224-Heating block temperature limiter',
              '350':'350-Flow temperature sensor defective (short circuit)',
              '351':'351-Flow temperature sensor defective (interruption)',
              '227':'227-Flame is not detected. After the 4th ignition attempt, the blocking fault becomes a locking fault',
              '229':'229-No ionization signal during burner operation',
              '232':'232-Temperature monitor AT90 has tripped',
              '290':'290-Internal error',
              '217':'217-Fan is not running',
              '214':'214-The fan will be switched off during the safety time',
              '216':'216-Fan too slow',
              '215':'215-Fan too fast',
              '224':'224-Heating block temperature limiter or flue gas temperature limiter has tripped',
              '222':'222-Flow temperature sensor defective (short circuit)',
              '223':'223-Flow temperature sensor defective (interruption)',
              '227':'227-Flame not detected',
              '228':'228-Flame is detected although burner is off',
              '306':'306-After gas shutdown: Flame is detected',
              '231':'231-Mains voltage was interrupted.',
              '261':'261-First safety time error',
              '280':'280-Restart attempt timed out',
              '235':'235-KIM not compatible with basic controller BC25',
              '360':'360-KIM not recognized.',
              '230':'230-Gas valve defective.',
              '234':'234-Connection cable of gas valve or gas valve or base controller BC25 defective',
              '238':'238-Connection cable of gas valve or gas valve or basic controller BC25 defective',
              '233':'233-KIM defective',
              '259':'259-KIM or Logamatic BC25 defective', 
              '323':'323-BUS-communicatie onderbroken.' 
            }
          %}
          {% set state = states('sensor.boiler_servicecodenumber') %}
          {% set message = mapper[state] if state in mapper %}
          {{ message }}

```

MichaelDvP, 2023-01-30:
Actual errors are send with telegram 0xBE from device and 0xBF from boiler, but only as long as the error exists. The errors logged in ems telegram 0x10 (need reset) and 0x11 (boiler tries again). 0x10 holds up to 8 last errors, 0x11 up t o 5, the newest first. EMS+ logs errors in 0xC2 with start of error and end of error, don't know how much errors really logged. It's 20byte per error, so the telegram could hold max 12 errors.

But this is a lot of redundant data, so we only read th newest error and publish it. If HA or  any other system log, you have all data in history. If somthing is missing, use teminal an type `read 8 C2` and you get all the boiler has logged. Needs a bit handwork to decode.

tefracky, 2023-02-05:
Just for information: Here is the sensor in the new Home-Assistant template format for all status numbers for the Buderus GB192i-19 in German:

```
## ------------------------------------------------------------------
## Service code number text for Buderus GB192i-19
## ------------------------------------------------------------------
- name: "Heizungsstatus"
  attributes:
     service-code: "{{states('sensor.heizung_ems_esp_boiler_statusmeldungsnummer')|int(0)}}"
     timestamp: "{{now().strftime('%Y-%m-%d %H:%M:%S')}}"
     date: >
         {{ now().timestamp() | timestamp_custom("%Y-%m-%d", true) }}
     time: >
        {{ now().timestamp() | timestamp_custom("%H:%M:%S", true) }}
  state: >-
      {% set mapper = {
          '200' : '200 - Das Gerät befindet sich im Heizbetrieb.',
          '201' : '201 - Das Gerät befindet sich im Warmwasserbetrieb.',
          '202' : '202 - Wartephase des Geräts. Wärmeanforderung durch RC-Regelgerät oder einen ON/OFF-Thermostat erfolgt in Intervallen von weniger als 10 Minuten.',
          '203' : '203 - Betriebsbereitschaft: keine Wärmeanforderung vorhanden',
          '204' : '204 - Wartephase des Geräts. Die gemessene Vorlauf- temperatur ist höher als die berechnete oder eingestellte Heizwassertemperatur. - Eingestellte Heizwassertemperatur am Gerät prüfen. Heizwassertemperatur ggf. erhöhen. Bei einer außentemperaturgeführten Regelung die eingestellte Heizkurve am Raumthermostat prüfen. Heizkurve ggf. ändern. Verkabelung und Funktion des Speichertemperaturfühlers prüfen. Bauteil ggf. austauschen.',
          '207' : '207 - Der Betriebsdruck ist zu niedrig (weniger als 0,2 bar). - Heizungsanlage bis zu 2 bar füllen. Ausdehnungsgefäß prüfen. Heizungsanlage auf undichte Stellen prüfen. Verkabelung und Funktion des Druckfühlers prüfen. Bauteil ggf. austauschen.',
          '208' : '208 - Das Gerät befindet sich im Schornsteinfegerbe-trieb oder Servicebetrieb.',
          '210' : '210 - Vom Abgastemperaturfühler gemessene Tem- peratur ist zu hoch und ist dadurch geöffnet. - Funktion des Abgastemperaturfühlers prüfen. Bauteil ggf. austauschen. Gerät auf Verschmutzung prüfen. Gerät ggf. warten.',
          '212' : '212 - Der Vorlauf- oder Sicherheitstemperaturfühler misst einen zu schnellen Temperaturanstieg. - Betriebsdruck prüfen. Heizungsanlage und Gerät entlüften. Heizungsanlage auf ausreichenden Wasserdurchfluss prüfen. Verkabelung und Funktion der Pumpe und des jeweiligen Temperaturfühlers prüfen. Bauteil ggf. austauschen.',
          '213' : '213 - Der Vorlauf- oder Rücklauftemperaturfühler misst einen zu schnellen Temperaturanstieg. - Betriebsdruck prüfen. Heizungsanlage und Gerät entlüften. Heizungsanlage auf ausreichenden Wasserdurchfluss prüfen. Wärmeleistung nach der Größe der Heizungsanlage einstellen. Verkabelung zur Pumpe oder zum jeweiligen Temperaturfühler prüfen. Bauteil ggf. austauschen.',
          '214' : '214 - Das Gebläse wird während der Sicherheitszeit abgeschaltet. - Verkabelung und Steckverbindungen des Gebläses prüfen. Betriebsverhalten des Geräts durch Austauschen des Gebläses prüfen. Steckverbindungen des Brennerautomaten prüfen. Brennerautomaten austauschen und Betriebsverhalten des Geräts prüfen.',
          '215' : '215 - Das Gebläse läuft zu schnell. - Abgasanlage prüfen, ggf. reinigen oder instandsetzen.',
          '216' : '216 - Das Gebläse läuft zu langsam. - Gebläsekabel mit Stecker prüfen, ggf. austauschen. Gebläse auf Verschmutzung und Blockierung prüfen, ggf. austauschen.',
          '217' : '217 - Das Gebläse läuft unregelmäßig während der Hochfahrphase. - Verkabelung und Steckverbindungen des Gebläses prüfen. Betriebsverhalten des Geräts durch Austauschen des Gebläses prüfen. Steckverbindungen des Brennerautomaten prüfen. Brennerautomaten austauschen und Betriebsverhalten des Geräts prüfen.',
          '218' : '218 - Die vom Vorlauftemperaturfühler gemessene Temperatur ist höher als 105 °C. - Betriebsdruck prüfen. Heizungsanlage und Gerät entlüften. Heizungsanlage auf ausreichenden Wasserdurchfluss prüfen. Funktion der Pumpe und des Vorlauftemperaturfühlers prüfen. Bauteil ggf. austauschen.',
          '219' : '219 - Der Sicherheitstemperaturfühler misst eine Temperatur über 105 °C. - Betriebsdruck prüfen. Heizungsanlage und Gerät entlüften. Heizungsanlage auf ausreichenden Wasserdurchfluss prüfen. Funktion der Pumpe und des Sicherheitstemperaturfühlers prüfen. Bauteil ggf. austauschen.',
          '220' : '220 - Kurzschluss des Sicherheitstemperaturfühlers oder gemessene Wassertemperatur ist höher als 130 °C.',
          '221' : '221 - Die Kontakte des Sicherheitstemperaturfühlers sind unterbrochen. - Stecker des Sicherheitstemperaturfühlers prüfen. Sicherheitstemperaturfühler austauschen und Betriebsverhalten des Geräts prüfen.',
          '222' : '222 - Die Kontakte des Vorlauftemperaturfühlers sind kurzgeschlossen. - Stecker des Vorlauftemperaturfühlers prüfen. Vorlauftemperaturfühler austauschen und Betriebsverhalten des Geräts prüfen.',
          '224' : '224 - Wärmeblock-Temperaturbegrenzer oder Abgas- temperaturbegrenzer hat ausgelöst. - Wenn die blockierende Störung längere Zeit bestehen bleibt, wird aus der blockie- renden Störung eine verriegelnde Störung. Wärmeblock-Temperaturbegrenzer und Anschlusskabel auf Unterbrechung prü- fen, ggf. austauschen. Bei Störung des Abgastemperaturbegrenzers erscheint eine Meldung max. nach 2 Stunden. Abgastemperaturbegrenzer und Anschlusskabel auf Unterbrechung prüfen, ggf [... cut ...]
          '2927' : '2927 - Flamme während Brennerbetrieb ausgefallen. - Hauptabsperreinrichtung öffnen. Geräteabsperrhahn öffnen. Gerät stilllegen und Gasleitung überprüfen. Signalauswertung auf Leiterplatte defekt. Überwachungselektrode austauschen. Schutzleiteranschluss (PE) im Steuergerät herstellen. Zündleitung austauschen. Anschlusskabel zur Überwachungselektrode austauschen. Gasarmatur austauschen.',
          '2928' : '2928 - Interner Fehler. - Feuerungsautomat austauschen.'
        }
      %}
      {% set state = states('sensor.heizung_ems_esp_boiler_statusmeldungsnummer') %}
      {% set message = mapper[state] if state in mapper %}
      {{ message }}
```

proddy, 2023-02-05:
> Just for information: Here is the sensor in the new Home-Assistant template format for all status numbers for the Buderus GB192i-19 in German:

thanks. I've added it to the wiki - https://emsesp.github.io/docs/tips-and-tricks/#home-assistant-sensor-template-for-all-status-numbers-on-a-gb192-boiler

MichaelDvP, 2023-02-05:
BTW: all codes (2019) are here: https://productslu.buderus.com/buderus/productslu.buderus.com/user/de/buderus_servicehandbuch_ems_2019.pdf

tefracky, 2023-02-05:
I was not aware of the maximum sensor length of 255 chars, so I shortened some messages. I updated my answer above, but maybe you should also update the Wiki.

pswid, 2023-02-10:
In my opinion, EMS_ESP should show/publish everything it can decode from the EMS bus. After all, each user can use the customization and block those entities that he does not need, right? However, I agree that decoding error numbers to text messages via EMS-ESP is not a good idea. As shown above, this can easily be done on the HomeAssiatant side.

BTW, in the new EMS+ systems (at least that's how it works in my Buderus WSW196i) you can read the entire log of the last 20 errors (the same as displayed by the RC300 controller). These errors can be read from Boiler with telegrams `0xC6` and `0xC7`. The structure is similar to `0xC2`. If there is room for Flash, I personally see no reason not to include it in future versions of EMS-ESP.

pswid, 2023-02-10:
BTW, I found an interesting file with many error codes, descriptions and potential causes:
https://github.com/bosch-thermostat/bosch-thermostat-client-python/blob/dev/bosch_thermostat_client/db/errorcodes_ivt.json

proddy, 2023-02-10:
> In my opinion, EMS_ESP should show/publish everything it can decode from the EMS bus. After all, each user can use the customization and block those entities that he does not need, right? 

It does that now, right? Everything is published that is decyphered. 

> BTW, in the new EMS+ systems (at least that's how it works in my Buderus WSW196i) you can read the entire log of the last 20 errors (the same as displayed by the RC300 controller). These errors can [...]

Flash is 99% full now. The ESP32 is a tiny thing. A lot of time has gone into making the software fit within the constraints of the ESP32. It's a tiny chip, the size of your pinky's nail and only has 0.5 MB RAM and 4 MB Flash. We've optimized the hell out of the code to make it all work and still facing memory issues. Compared to something like the RC300

ESP32
[image]

RC300
[image]

If we're storing error codes in the Flash, it may fit with some compression but there will be a lot of wear with all the writes and rotation, which is never a good thing.

pswid, 2023-02-22:
Yes, currently EMS-ESP decyphers all known parameters. What I meant was that it is no longer possible to add more (e.g. list of last 20 errors) due to out of memory.

The limitation here is not ESP32 itself, but FLASH and RAM. In order to free up some FLASH memory, it would be probably necessary to change the way of updating the firmware (currently ~2MB of the Flash is "wasted"!), e.g. as it is done in TASMOTA (they use "minimal" firmware during update) or drop the multilingual version and publish each language version separately (like TASMOTA). I don't like any of these ideas. Do you have any others?

I think a decision has to be made to abandon ESP32-4MB and switch to ESP32 with 16MB Flash and PSRAM. This will increase resources 4x! It would provide plenty of room for future project development. What do you think?

From what I can see most of the interfaces produced by BBQKees are based on the ESP32-WROOM module. Upgrading them would be relatively simple (for anyone who can solder and has the necessary equipment). Just replace the module with the latest WROOM-32E-N16R2 version and add a 10kohm pull-up resistor to GPIO16. This version has 16MB of flash and 2MB of PSRAM. Of course, future versions of the interface can use also e.g. (easily available) ESP32-WROVER-E-N16R8.

WROOM-32E-N16R2 is still hard to get (I saw it only in the official ESPRESSIF store on Aliexpress), so I'm currently testing using "ordinary" WROOM-32UE-N16 with soldered 8MB PSRAM chip. After enabling PSRAM (by adding `heap_caps_malloc_extmem_enable(0)`) I noticed no stability issues in EMS-ESP (unlike the version 4MB of Flash, no PSRAM). HEAP MAX Alloc is no longer =39kB.

[image]

[image]

proddy, 2023-02-22:
you're spot on. We realized this last year and already have started coding for the larger chips S2 and S3 with more Flash and PSRAM. For the 4MB variants, there is https://github.com/emsesp/EMS-ESP32/issues/608 which is done, and works but is still experimental and needs to be explained to end-users. We adapted the concept from Tasmota and Michael did all the coding so kudos to him.

I'm still worried about the wear on the Flash though if we're constantly updating the last error codes.

tefracky, 2023-02-22:
I think, it is not the question, if, but how the upgrade to 16 MB will be handled. In my case, I use a dedicated ESP-Board with an EMS-Bus converter from BBQKees. I didnt found 16 MB ESP32 Boards, which can be used directly, but only the chips. In my opinion, soldering the chip is no sufficient way.

MichaelDvP, 2023-02-22:
> I didnt found 16 MB ESP32 Boards, which can be used directly, but only the chips. 

https://www.lilygo.cc/products/t7-s3
Works as drop-in replacement in older gateway premium. You need to compile to S3. 
BBQkees uses a 16MB SOC in the newer S32 gateways. ETH-is a problem with S3, i think that's the reason why proddy plans W5500 support(see #1037).

bbqkees, 2023-02-23:
The S32 has had a 16MB flash chip for a long time now. The E32 has 4MB, which I cannot upgrade to 16MB in the current format. With the addition of all the languages the problem is more in the RAM of the ESP32 chip.

For the next gen S32 the choice is pretty simple. I am looking to use the S3 with 16MB flash and 8MB PSRAM. This is the biggest leap forward we can currently make on the ESP platform. 

However, for reasons unknown to me, they removed the convenient hardware Ethernet support from the ESP32-S3. That's why I asked Proddy to have a look at the W5500 Ethernet chip. But this will take some time.

So for the next gen E32 I may have to do an intermediate version with an ESP32 with additional PSRAM.

bbqkees, 2023-03-06:
You may be right about the E32 indeed. It's a shame they removed the Ethernet support in the S3. 
The ESP32+LAN8720 works well, with minimal additional coding effort.
For the S3 the W5500 is the only really viable Ethernet chip.
The other two supported MAC chips are very expensive and hardly used. The W5500 has been around for quite some years now. For our purpose the SPI speed should be more than enough.

So maybe it's better to remain a bit longer on the ESP32+PSRAM platform for the next E32 until the official Ethernet support in the IDF for the S3 etc has matured enough. (For the next-gen S32 the S3 is still the better choice).
On the other hand if I design a new E32 with the ESP32+PSRAM we have to stay on that platform for a year or two (to make the engineering effort worthwhile) but that won't be a problem as the ESP32 with PSRAM has enough resources.

The ETH01 was convenient, and that LilYGO looks nice but for the next E32 I am going for a complete board with all chips on it like as in the S32.
It's a bit of a pain in the ass to source add-on modules every time. Better have a full custom PCB design I can order in volume.

closed 2023-07-03 by proddy

misa1515, 2024-10-23:
@tefracky 

> Just for information: Here is the sensor in the new Home-Assistant template format for all status numbers for the Buderus GB192i-19 in German:
> 
> ```
> ## ------------------------------------------------------------------
> ## Service code number text for Buderus GB192i-19
> ## ------------------------------------------------------------------
> - name: "Heizungsstatus"
> ...
>       {{ message }}
> ```

I insert your code to template sensor, but facing issue. How to correct it?
[image]

tefracky, 2024-10-26:
@misa1515 You have to use the service code number instead of the last service code