emsesp/EMS-ESP32 #943: wording for silent mode on/off

harvey637, 2023-02-05:
hi @Proddy,

simple? enhancement:
the wording "hours" (Stunden in DE) for "silent mode from/to" is slightly misleading.
IMHO it means "from XX o`clock" "to XX o`clock" (german: "von XX Uhr" "bis XX Uhr" (possibly adding the minutes as
"from 22:00 o`clock" or "von 22:00 Uhr"
"to 06:00 o`clock" or "bis 06:00 Uhr" (is this correct or 06:59 ????) to make it even nicer.

everything with ems-esp32 is working great, only some entities do not reflect the state of my Bosch Compress 7000 like "el. heater step 0/1/2", which stay =0 but my auxheater is working in the morning .... just my observation. "aux heater mixing valve" is showing higher % when the auxheater is on, 0% with aux heater off.

cu
Harvey

MichaelDvP, 2023-02-05:
I think the `o'clock` unit is something to add in v3.6. For scheduling #701/#756 it is also usefull.

The `el .heater step` was introduced in #892. I have no heatpump and i'm not sure what it switches, but it's a setting, not a state. If you like to monitor the el. heater switching, you'll need to search the right telegram/offset.

MichScha, 2023-02-15:
In this case the silent mode from/to means always the calculcation from 0:00 + minutes that are send by the telegram. Only valid values are in steps of 15min.

closed 2023-07-03 by proddy