emsesp/EMS-ESP32 #944: Buderus GB172: Heating Pump operating Mode

rschiemann1, 2023-02-05:
Hi,
I usw EMS ESP together with my Gas boiler Buderus GB172. Besides the basic controller, it has a RC310, which allows to change the operating characteristics of the heating pump. Different modes can be set:
- proportional to burner power 
- DeltaP-1 (means constant pressure differential of 150mbar)
- DeltaP-2 (constant 200mbar)
- And two other DeltaP variants

In the EMS ESP, I don‘t see this setting. Is it supported? If not, I would strongly encourage to implement it. Can that be done? What info can I provide to support?

MichaelDvP, 2023-02-06:
I think there is a good chance that this information is readable/writeable. 
Please download the service info from help page of ems-esp and attach it. Then we can see if there is a special pump module.
Go to System/System log and select `loglevel all`, leave other options as is. On your heating control now change the pump settings to each of the options and back to first. Download the log and attach it here. Tell us the options you have set, we'll try to match the telegrams to the settings.

rschiemann1, 2023-02-06:
Here is the system info:
```
{
  "System Info": {
    "version": "3.5.0-dev.17",
    "platform": "ESP32",
    "uptime": "000+00:25:56.769",
    "uptime (seconds)": 1556,
    "free mem": 109,
    "max alloc": 39,
    "free app": 6155,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -69,
    "IPv4 address": "192.168.178.101/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 939,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
[... 582 lines cut ...]
000+00:30:07.530 TRACE 493: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19
000+00:30:07.641 DEBUG 494: [emsesp] Last Tx read successful
000+00:30:07.641 TRACE 495: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 45 00 01 00 00 00 00 00 00 02 F6 06 FF 00 00 00 00 00 00
000+00:30:10.058 DEBUG 496: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1044, retain=0, retry=1, size=502, pid=1)
000+00:30:10.159 DEBUG 497: [mqtt] Publishing topic ems-esp/boiler_data (#1045, retain=0, retry=1, size=788, pid=1)
000+00:30:10.260 DEBUG 498: [mqtt] Publishing topic ems-esp/thermostat_data (#1046, retain=0, retry=1, size=925, pid=1)
000+00:30:10.361 DEBUG 499: [mqtt] Publishing topic ems-esp/dallassensor_data (#1047, retain=0, retry=1, size=298, pid=1)
000+00:30:13.291 TRACE 500: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 39 02 1A 02 1A 21 00 00 03 00 03 73 88 00 4B 04 00 80 00
000+00:30:13.553 TRACE 501: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 23 01 28 64 00 01 01 20 40 80 00 02 1A 80 00 00 00 FF 30 59 00 CC 00 00 00
000+00:30:13.846 TRACE 502: [emsesp] boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
000+00:30:20.084 DEBUG 503: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1048, retain=0, retry=1, size=502, pid=1)
000+00:30:20.185 DEBUG 504: [mqtt] Publishing topic ems-esp/boiler_data (#1049, retain=0, retry=1, size=788, pid=1)
000+00:30:20.286 DEBUG 505: [mqtt] Publishing topic ems-esp/thermostat_data (#1050, retain=0, retry=1, size=925, pid=1)
000+00:30:20.387 DEBUG 506: [mqtt] Publishing topic ems-esp/dallassensor_data (#1051, retain=0, retry=1, size=298, pid=1)
000+00:30:23.188 TRACE 507: [emsesp] boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
000+00:30:23.500 TRACE 508: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 39 02 1A 02 1A 21 00 00 03 00 03 73 88 00 4B 04 00 80 00
000+00:30:23.748 TRACE 509: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
000+00:30:24.071 TRACE 510: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 23 01 28 64 00 01 01 20 40 80 00 02 1A 80 00 00 00 FF 30 59 00 CC 00 00 00
000+00:30:24.347 TRACE 511: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 00 1B 80 00 80 00 00 00 00 25 01 A1 86 0C C8 52 00 00 00 09 54 CA 01 56 82 80 00
000+00:30:24.585 TRACE 512: [emsesp] boiler(0x08) -B-> All(0x00), UBAMaintenanceStatus(0x1C), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:30:30.058 DEBUG 513: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1052, retain=0, retry=1, size=502, pid=1)
000+00:30:30.159 DEBUG 514: [mqtt] Publishing topic ems-esp/boiler_data (#1053, retain=0, retry=1, size=787, pid=1)
000+00:30:30.260 DEBUG 515: [mqtt] Publishing topic ems-esp/thermostat_data (#1054, retain=0, retry=1, size=925, pid=1)
000+00:30:30.361 DEBUG 516: [mqtt] Publishing topic ems-esp/dallassensor_data (#1055, retain=0, retry=1, size=298, pid=1)
```

MichaelDvP, 2023-02-06:
Ok, thanks, here is the setting in offset 11 of telegram 0x16:
`000+00:28:14.786 TRACE 230: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBAParameters(0x16), data: 02 (offset 11)`
Please udate with this build and check if all is working as expected:
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

BTW: attaching the file by drag/drop or click the footer in github editbox is easier and better searchable that copy/paste the whole content.

rschiemann1, 2023-02-06:
That works as intended. Wow, thank you for the quick and good work

MichaelDvP, 2023-02-07:
Is now in official release. Closing.

closed 2023-02-07 by MichaelDvP