emsesp/EMS-ESP32 #945: EMS-ESP occasionally losing entities on Buderus GB172 + RC310

rschiemann1, 2023-02-06:
Hi,

i have a Buderus GB172 and a RC310 controller. Generally, EMS-ESP works nice but every once in a while, it loses some entities of rc310. When that happens, i see in home Assistant:
[image]
The orange warnings say „Entity not found“. On ESP site, I also see that these entities are gone. Rescanning does not help. Eventually, they reappear by themselves. 

Seems to be a bug?

MichaelDvP, 2023-02-06:
Old software? Unstable network? Restarts? 
Please attach the `support info` from ems-esp Help page.

rschiemann1, 2023-02-06:
Sorry for missing Info. 

Here is the support info:
`System Status": {
    "version": "3.5.0b4",
    "uptime": "000+05:05:23.646",
    "freemem": 102,
    "free_app": 6230,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -57,
    "IPv4 address": "192.168.178.101/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 7928,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 4,
    "temperature sensor reads": 14656,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 2,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 16865,
    "bus reads (tx)": 3736,
    "bus writes (tx)": 2,
    "bus incomplete telegrams": 0,
    "bus reads failed": 213,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 95
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "06.08",
      "entities": 61,
      "handlers received": "0x10 0x11 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x15 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "18.03",
      "entities": 40,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x02AF 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x02CE 0x02D0 0x02D2 0x031E",
      "handlers ignored": "0x1A 0x23 0x35 0xBF"
    },
    {
      "type": "Controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.03",
      "entities": 0
    }
  `

I have no reasons to suspect bad network, I experience no connection issues with emsesp. Also Software seems to be very new (even newer than last official release).

MichaelDvP, 2023-02-06:
Version 3.5.0b4 have a issue with restarts when HA is enabled, this is solved in actual builds. Please update to latest dev.

proddy, 2023-02-06:
@rschiemann1 in general, if you're using a development build it will definitely have a lot of bugs. And once you're on that beta track you will need to keep updating to the latest dev builds. Or go back to using the stable official releases.

rschiemann1, 2023-02-06:
Hi,
I appreciate your quick replies. 
I agree, I as a newb should not use dev builds, but my esp was shipped with the shown build, I did not even knew that. Will update to latest build and see what changes. Is there any danger involved with simply clicking the update button? Should I take any measures before?

proddy, 2023-02-06:
ah ok. I would wait until the official 3.5.0 is released this week and then update that. You may run into problems with the partitions depending on when you purchased the bbqkees board, but there will be guidelines on the wiki's to help you migrate.

MichaelDvP, 2023-02-06:
@proddy afair BBQKees has updated his stock to this B4 after the partition issue was known, this is a 16M S32 Gateway with 6230k free app, there no issue with updating. I can't find the issue, but i've posted how to use the espressif updater with seperate files, and this example was also B4.

rschiemann1, 2023-02-06:
Thanks. Could not hold my horses and updated. Lost entities immediately reappeared. Looking good so far, thank you all!

bbqkees, 2023-02-06:
@MichaelDvP I have been loading B4 for a while back then and recently B7 (until 3.5.0 is released).
Get very little complaints with these versions and if there are, everyone can still update themselves to another or newer build.
All Gateways since October 2022 have the correct partition sizes for the larger 3.5.0.

closed 2023-02-19 by proddy