emsesp/EMS-ESP32 #946: hc1 design temperature can in HA only be changed with increments of 2

HansRemmerswaal, 2023-02-06:
**Question**
hc1 design temperature can with EMS-ESP web interface be changed with increments of 1.
hc1 design temperature can with Home Assistant only be changed with increments of 2.

Is this EMS-ESP telling Home Assistant that the increment is 2? Or is this a problem in Home Assistant itself?

**Screenshots**

**Device information**
[emsesp_info (1).txt](https://github.com/emsesp/EMS-ESP32/files/10662018/emsesp_info.1.txt)

**Additional context**

MichaelDvP, 2023-02-06:
>  Is this EMS-ESP telling Home Assistant that the increment is 2?

You can check in mqtt: 
```
homeassistant/number/ems/thermostat_hc2_designtemp/config = {"uniq_id":"ems_thermostat_hc2_designtemp","obj_id":"ems_thermostat_hc2_designtemp","cmd_t":"ems/thermostat/hc2/designtemp","mode":"box","step":1,"min":30,"max":90,"ic":"mdi:coolant-temperature","stat_t":"ems/thermostat_data_hc2","name":"HK2 Auslegungstemperatur","val_tpl":"{{value_json.designtemp}}","unit_of_meas":"°C","dev":{"ids":["ems-esp-thermostat"]}}
```
The step is set to the same value ems-esp uses internaly ("step": 1).

HansRemmerswaal, 2023-02-06:
In your mqtt message I see hc2 instead of hc1.

I need to find a way to capture the mqtt messages to see what is being send. I tried that before but without much luck.

proddy, 2023-02-06:
can you post a screen shot of HA showing the 2 point increment?

MichaelDvP, 2023-02-06:
> In your mqtt message I see hc2 instead of hc1.

I have only hc2 on a RC35 thermostat, but that does not matter. Also i don't use HA, so i can only check the mqtt output, not what HA do with it.
> I need to find a way to capture the mqtt messages to see what is being send. I tried that before but without much luck.

https://mqtt-explorer.com/

HansRemmerswaal, 2023-02-06:
My bad, the increment is 1.

I just managed to go from 40 to 42 to 43, but not 41. 
I guess that due to the scale, 0 to 254, it is becoming difficult to (always) change in steps of 1 as then the mouse resolution is coming into the picture.

Having said that, a max of 254 degrees is a bit high, the range for this value is on my device from 30 to 85.

[image]

[image]

MichaelDvP, 2023-02-06:
You can set min/max on the customization page, this will also affect HA. 
The limits vary for different boilers, it's hard to set a default. 254 is the limit of the byte value (255=not set).

MichaelDvP, 2023-02-06:
> Do you want me to checked the min / max for the temperatures and provide a list, or will you keep these anyhow on the 0 to 254 as default values?

If you like to do this, go on. 
When we started with min/max settings we had a lot of complains because boilers have different values (like 168% power). But now if a value is out of range, the min/max is automatically adapted to the value, also users can customize to the values their boiler have. If the default fits most boilers it is much better than 254 degrees for a water temperature at nearly atmospheric pressure.

closed 2023-02-19 by proddy