emsesp/EMS-ESP32 #950: upgrading to v3.5 for the first time from v3.4 on a BBQKees

JokerGermany, 2023-02-06:
> When upgrading to v3.5 for the first time from v3.4 on a BBQKees Gateway board you will need to use the EMS-EPS Flasher to correctly re-partition the flash. Make sure you backup the settings and customizations from the WebUI (System->Upload/Download) and restore after the upgrade.

How do i know if i am not correctly partioned? I already use 3.5 Beta version and i think a 3.5 beta version was installed when it was shipped to me.

MichaelDvP, 2023-02-07:
If update works, you have a correctly partitioned esp, If repeated trys to update always shows error, you need to update via usb or stay wth old software. A sign for the wrong partition scheme is if the system page shows a filesystem larger than 64k (sum of used/free).

proddy, 2023-02-07:
All of bbqkees' gateways are 3.5.0 ready as from since October 2022. They have the adjusted partition size and the latest firmware can be uploaded via the WebUI. For earlier versions you will need to follow the instructions (make a backup of the config, use the EMSESP Flasher, restore config).

closed 2023-02-19 by proddy