emsesp/EMS-ESP32 #951: Cannot write to Bosch Condens 2500W

hpeter1994, 2023-02-07:
**Bug description**
Hello.
I just connected my brand new E32 Gateway to my Bosch Condens 2500W boiler. It recognizes it as "Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3". 
I have no other devices connected to the bus (no termostat as i'm not using it).
I can read the status of every entity, but can't change anything, as it will immediately change back to the previous setting.

I have zero TX errors and the log says the command was sent successfully.
I'm using v3.5.0b7

**Steps to reproduce**
Default settings. I tried changing the "selected flow temperature", "heating activated", "dwh selected temperature", "dwh activated", "dwh one time charging" parameters but nothing sticks.

**Expected behavior**
I would expect to be able to switch on and off my boiler from the gateway. I read on the wiki that some Junkers boilers have this issue but i have found nothing to indicate that my boiler would be effected as well.

**Device information**

```
 {
  "System Status": {
    "version": "3.5.0b7",
    "platform": "ESP32",
    "uptime": "000+02:08:10.001",
    "freemem": 124,
    "free_app": 74,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "Ethernet",
    "hostname": "smart-kazan",
    "MAC": "C4:DE:E2:B1:D4:5F",
    "IPv4 address": "192.168.1.138/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "phy type": 1,
    "eth power": 16,
    "eth phy addr": 1,
    "eth clock mode": 0,
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Budapest"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 1906,
    "MQTT queued": 1,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 2954,
    "bus reads (tx)": 430,
    "bus writes (tx)": 33,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "43.10",
      "entities": 57,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "43.10",
      "entities": 0
    }
  ]
}
```

**Telnet console output**
```
smart-kazan:# call boiler heatingactivated off
smart-kazan:# 000+02:16:46.447 D 23: [command] Calling command 'boiler/heatingactivated' (heating activated) with value off
000+02:16:46.447 D 24: [telegram] [DEBUG] New Tx [#235] telegram, length 1
smart-kazan:# 000+02:16:46.471 D 25: [telegram] Sending write Tx [#235], telegram: 8B 08 16 00 00
000+02:16:46.487 D 26: [telegram] [DEBUG] New Rx telegram, message length 1
smart-kazan:# 000+02:16:46.487 D 27: [emsesp] Last Tx write successful
000+02:16:46.487 D 28: [telegram] [DEBUG] New Tx [#236] telegram, length 1
000+02:16:46.487 D 29: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
000+02:16:46.487 D 30: [emsesp] No telegram type handler found for ID 0x16 (src 0x0B)
smart-kazan:# 000+02:16:46.554 D 31: [telegram] [DEBUG] New Rx telegram, message length 12
smart-kazan:# 000+02:16:46.786 D 32: [telegram] Sending read Tx [#236], telegram: 8B 88 16 00 1B
smart-kazan:# 000+02:16:46.786 D 33: [telegram] [DEBUG] New Rx telegram, message length 1
smart-kazan:# 000+02:16:46.833 D 34: [emsesp] Last Tx read successful
000+02:16:46.833 D 35: [telegram] [DEBUG] New Rx telegram, message length 12
smart-kazan:# 000+02:16:46.833 I 36: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data_ww' payload={"wwsettemp":15,"wwseltemp":15,"wwtype":"buffer","wwcomfort":"hot","wwflowtempoffset":70,"wwcircpump":"off","wwchargetype":"3-way valve","wwhyston":0,"wwhystoff":0,"wwdisinfectiontemp":75,"
000+02:16:46.833 I 37: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data' payload={"heatingactive":"on","tapwateractive":"off","selflowtemp":40,"selburnpow":99,"heatingpumpmod":100,"curflowtemp":35.1,"burngas":"on","burngas2":"off","heatingpump":"on","fanwork":"on","ignwork
smart-kazan:# 000+02:16:46.931 D 38: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2024, retain=0, retry=1, size=462, pid=1)
smart-kazan:# 000+02:16:47.032 D 39: [mqtt] Publishing topic ems-esp/boiler_data (#2025, retain=0, retry=1, size=598, pid=1)
smart-kazan:# 000+02:16:48.348 D 40: [telegram] [DEBUG] New Rx telegram, message length 25
smart-kazan:# 000+02:16:50.001 I 41: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data_ww' payload={"wwsettemp":15,"wwseltemp":15,"wwtype":"buffer","wwcomfort":"hot","wwflowtempoffset":70,"wwcircpump":"off","wwchargetype":"3-way valve","wwhyston":0,"wwhystoff":0,"wwdisinfectiontemp":75,"
000+02:16:50.001 I 42: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data' payload={"heatingactive":"on","tapwateractive":"off","selflowtemp":40,"selburnpow":99,"heatingpumpmod":100,"curflowtemp":35.5,"burngas":"on","burngas2":"off","heatingpump":"on","fanwork":"on","ignwork
smart-kazan:# 000+02:16:50.078 D 43: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2026, retain=0, retry=1, size=462, pid=1)
smart-kazan:# 000+02:16:50.179 D 44: [mqtt] Publishing topic ems-esp/boiler_data (#2027, retain=0, retry=1, size=598, pid=1)
smart-kazan:# 000+02:16:55.531 D 45: [telegram] [DEBUG] New Rx telegram, message length 25
smart-kazan:# 000+02:16:55.740 D 46: [telegram] [DEBUG] New Rx telegram, message length 17
smart-kazan:# 000+02:16:58.341 D 47: [telegram] [DEBUG] New Rx telegram, message length 25
smart-kazan:# 000+02:17:00.001 I 48: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data_ww' payload={"wwsettemp":15,"wwseltemp":15,"wwtype":"buffer","wwcomfort":"hot","wwflowtempoffset":70,"wwcircpump":"off","wwchargetype":"3-way valve","wwhyston":0,"wwhystoff":0,"wwdisinfectiontemp":75,"
000+02:17:00.001 I 49: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data' payload={"heatingactive":"on","tapwateractive":"off","selflowtemp":40,"selburnpow":99,"heatingpumpmod":100,"curflowtemp":37.0,"burngas":"on","burngas2":"off","heatingpump":"on","fanwork":"on","ignwork
smart-kazan:# 000+02:17:00.133 D 50: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2028, retain=0, retry=1, size=462, pid=1)
smart-kazan:# 
```

MichaelDvP, 2023-02-08:
For some settings the controller (or thermostats) takes control and overwrites every setting from ems-esp. This is known for the controllers with mechanical knobs like Buderus MC10 and Junkers HT3.  Also the troubleshooting section of FAQ mention this.
Mostly you can control the selflowtemp if the knob is in high position and you send a lower value.

hpeter1994, 2023-02-08:
Yeah i saw the mention in the FAQ, but my boiler has no mechanical knobs, its all digital and i don't have a thermostat either. 
Since then i tested yes, i can set the flow temp lower than the set value on the display, but i have no way to turn on and off the heating and the dhw, or switching between them, and thats basically the only thing i want to controll :/
So it looks like i'm SOL. It just looks weird for me that the boiler won't accept the commands from a "service key"., then whats the service key good for?

MichaelDvP, 2023-02-08:
Your log does not show the telegrams, only that the write was successfull, so boiler has acknowledged. Try to check if it the contoller resets the command or the boiler itself, set `watch 16` before sending the command by `call boiler heatingactivated 0`
If you see the write from controller, try if the controller is writable `call system send "0B 09 16 00 00"` 
What can be set in the boilers differs often between boiler models and not always follows a clear logic. I think Bosch often roll dice to make reverse engineering more difficult.

hpeter1994, 2023-02-08:
If i understand the output correctly then, it looks like its the boiler.

```
smart-kazan:# call boiler heatingactivated 0
000+01:00:41.549 D 29: [command] Calling command 'boiler/heatingactivated' (heating activated) with value 0
000+01:00:41.549 D 30: [telegram] [DEBUG] New Tx [#201] telegram, length 1
000+01:00:41.606 D 31: [telegram] Sending write Tx [#201], telegram: 8B 08 16 00 00
000+01:00:41.606 D 32: [telegram] [DEBUG] New Rx telegram, message length 1
000+01:00:41.606 N 33: [emsesp] Me(0x0B) -W-> Boiler(0x08), UBAParameters(0x16), data: 00
000+01:00:41.606 D 34: [emsesp] No telegram type handler found for ID 0x16 (src 0x0B)
000+01:00:41.629 D 35: [emsesp] Last Tx write successful
000+01:00:41.629 D 36: [telegram] [DEBUG] New Tx [#202] telegram, length 1
000+01:00:41.629 D 37: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
000+01:00:41.719 D 38: [telegram] [DEBUG] New Rx telegram, message length 12
000+01:00:41.719 N 39: [emsesp] Boiler(0x08) -B-> All(0x00), UBAParameters(0x16), data: FF 24 63 00 00 FC 02 01 01 64 0A 02
000+01:00:41.950 D 40: [telegram] Sending read Tx [#202], telegram: 8B 88 16 00 1B
000+01:00:41.950 D 41: [telegram] [DEBUG] New Rx telegram, message length 1
000+01:00:41.950 N 42: [emsesp] Me(0x0B) -R-> Boiler(0x08), UBAParameters(0x16), length: 0x1B
000+01:00:41.999 D 43: [emsesp] Last Tx read successful
000+01:00:41.999 D 44: [telegram] [DEBUG] New Rx telegram, message length 12
000+01:00:41.999 N 45: [emsesp] Boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 24 63 00 00 FC 02 01 01 64 0A 02

```

 I was not seeing anything from the controller except my write command (i presume in this scenario the controller is the EMS-EPS, or there should be another controller inside the boiler?), but i tried the other write command anyway, with the same results it looks like.

```
smart-kazan:# call system send "0B 09 16 00 00"
000+01:01:48.544 D 114: [command] Calling command 'system/send' (send a telegram) with value 0B 09 16 00 00
000+01:01:48.544 D 115: [telegram] [DEBUG] New Tx [#206] telegram, length 1
000+01:01:48.598 D 116: [telegram] Sending write Tx [#206], telegram: 8B 09 16 00 00
000+01:01:48.598 D 117: [telegram] [DEBUG] New Rx telegram, message length 1
000+01:01:48.598 N 118: [emsesp] Me(0x0B) -W-> Controller(0x09), UBAParameters(0x16), data: 00
000+01:01:48.598 D 119: [emsesp] No telegram type handler found for ID 0x16 (src 0x0B)
000+01:01:48.620 D 120: [emsesp] Last Tx write successful
000+01:01:48.620 D 121: [telegram] [DEBUG] New Tx [#207] telegram, length 1
000+01:01:48.620 D 122: [telegram] Sending post validate read, type ID 0x16 to dest 0x09
000+01:01:48.679 D 123: [telegram] [DEBUG] New Rx telegram, message length 12
000+01:01:48.679 N 124: [emsesp] Boiler(0x08) -B-> All(0x00), UBAParameters(0x16), data: FF 24 63 00 00 FC 02 01 01 64 0A 02
000+01:01:48.942 D 125: [telegram] Sending read Tx [#207], telegram: 8B 89 16 00 1B
000+01:01:48.942 D 126: [telegram] [DEBUG] New Rx telegram, message length 1
000+01:01:48.942 N 127: [emsesp] Me(0x0B) -R-> Controller(0x09), UBAParameters(0x16), length: 0x1B
000+01:01:48.991 D 128: [emsesp] Last Tx read successful
000+01:01:48.991 D 129: [telegram] [DEBUG] New Rx telegram, message length 12
000+01:01:48.991 I 130: [emsesp] Controller(0x09) -W-> Me(0x0B), UBAParameters(0x16), data: FF 24 63 00 00 FC 02 01 01 64 0A 02
000+01:01:48.991 D 131: [emsesp] No telegram type handler found for ID 0x16 (src 0x09)
```

MichaelDvP, 2023-02-08:
The boiler do not store the new setting, the readback of telegram 0x16 gives heating on. There is no overwrite from controller. Bu the controller also stores the settings in telegram 0x16. This is different from my  system. My boiler accepts heatingactivated off, but controller overwirites and writing to controller does not work. 
Yu can check what's on the bus if you set heating to off by controller. Set in termnal `watch on` to see all telegrams and then change on the controller to heating off. Maybe there is a different setting for this.
( If i do this i see only the command from controller to boiler setting selflowtemp 30, and then heatingactivated to 0, that's all: 
```
2023-02-08 17:09:03.443 TRACE 72: [emsesp] controller(0x09) -W-> boiler(0x08), UBAParameters(0x16), data: 1E (offset 1)
2023-02-08 17:09:04.162 TRACE 75: [emsesp] controller(0x09) -W-> boiler(0x08), UBAParameters(0x16), data: 00
```

hpeter1994, 2023-02-09:
Okay, i tried it, and i'm seeing zero messages to and from controller, and no writes whatsoever. The only thing i see here is that the boiler is broadcasting the new states.

```
000+16:57:38.002 D 208: [telegram] [DEBUG] New Rx telegram, message length 15
000+16:57:38.002 N 209: [emsesp] Boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+16:57:38.232 D 210: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:38.232 N 211: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 32 01 63 63 18 09 02 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+16:57:38.441 D 212: [telegram] [DEBUG] New Rx telegram, message length 17
000+16:57:38.441 N 213: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C8 00 C8 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+16:57:38.671 D 214: [telegram] [DEBUG] New Rx telegram, message length 27
000+16:57:38.671 N 215: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 64 01 16 6A 0A C6 4F 00 00 00 0A 8F 14 01 0C C5 80 00
000+16:57:40.001 I 216: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data_ww' payload={"wwsettemp":15,"wwseltemp":15,"wwtype":"buffer","wwcomfort":"hot","wwflowtempoffset":70,"wwcircpump":"off","wwchargetype":"3-way valve","wwhyston":0,"wwhystoff":0,"wwdisinfectiontemp":75,"
000+16:57:40.001 I 217: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data' payload={"heatingactive":"on","tapwateractive":"off","selflowtemp":50,"selburnpow":99,"heatingpumpmod":100,"curflowtemp":35.5,"burngas":"on","burngas2":"off","heatingpump":"on","fanwork":"on","ignwork
000+16:57:40.078 D 218: [mqtt] Publishing topic ems-esp/boiler_data_ww (#13439, retain=0, retry=1, size=462, pid=1)
000+16:57:40.179 D 219: [mqtt] Publishing topic ems-esp/boiler_data (#13440, retain=0, retry=1, size=598, pid=1)
000+16:57:45.104 D 220: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:45.104 N 221: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 5F 63 00 08 00 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+16:57:45.104 I 222: [mqtt] [DEBUG] Adding to queue: (Publish) topic='heating_active' payload=off
000+16:57:45.145 D 223: [mqtt] Publishing topic ems-esp/heating_active (#13441, retain=0, retry=1, size=3, pid=1)
000+16:57:45.307 D 224: [telegram] [DEBUG] New Rx telegram, message length 12
000+16:57:45.307 N 225: [emsesp] Boiler(0x08) -B-> All(0x00), UBAParameters(0x16), data: 00 00 63 00 00 FC 02 01 01 64 0A 02
000+16:57:45.537 D 226: [telegram] [DEBUG] New Rx telegram, message length 20
000+16:57:45.537 N 227: [emsesp] Boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 20 00 01 0E 00 AE 80 00 00 80 00 80 00 80 00
000+16:57:46.103 D 228: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:46.103 N 229: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 5E 63 00 00 00 21 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+16:57:46.316 D 230: [telegram] [DEBUG] New Rx telegram, message length 20
000+16:57:46.316 N 231: [emsesp] Boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 20 00 01 0E 00 D2 80 00 00 80 00 80 00 80 00
000+16:57:46.787 D 232: [telegram] [DEBUG] New Rx telegram, message length 20
000+16:57:46.787 N 233: [emsesp] Boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 23 00 01 0E 00 E4 80 00 00 80 00 80 00 80 00
000+16:57:47.066 D 234: [telegram] [DEBUG] New Rx telegram, message length 20
000+16:57:47.066 N 235: [emsesp] Boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 02 00 01 0E 00 FC 80 00 00 80 00 80 00 80 00
000+16:57:47.292 D 236: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:47.292 N 237: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 5D 63 00 00 00 24 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+16:57:48.062 D 238: [telegram] [DEBUG] New Rx telegram, message length 17
000+16:57:48.062 N 239: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C5 00 C5 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+16:57:48.292 D 240: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:48.292 N 241: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 5D 63 00 00 00 24 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+16:57:50.001 I 242: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data_ww' payload={"wwsettemp":15,"wwseltemp":15,"wwtype":"buffer","wwcomfort":"hot","wwflowtempoffset":70,"wwcircpump":"off","wwchargetype":"3-way valve","wwhyston":0,"wwhystoff":0,"wwdisinfectiontemp":75,"
000+16:57:50.001 I 243: [mqtt] [DEBUG] Adding to queue: (Publish) topic='boiler_data' payload={"heatingactive":"off","tapwateractive":"off","selflowtemp":29,"selburnpow":99,"heatingpumpmod":100,"curflowtemp":34.9,"burngas":"off","burngas2":"off","heatingpump":"on","fanwork":"on","ignwo
000+16:57:50.078 D 244: [mqtt] Publishing topic ems-esp/boiler_data_ww (#13442, retain=0, retry=1, size=462, pid=1)
000+16:57:50.179 D 245: [mqtt] Publishing topic ems-esp/boiler_data (#13443, retain=0, retry=1, size=599, pid=1)
000+16:57:53.100 D 246: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:57:53.100 N 247: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 58 63 00 00 00 24 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
[... cut ...]
000+16:58:08.272 D 292: [telegram] [DEBUG] New Rx telegram, message length 17
000+16:58:08.272 N 293: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C7 00 C7 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+16:58:09.032 D 294: [telegram] [DEBUG] New Rx telegram, message length 25
000+16:58:09.032 N 295: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1D 01 49 63 00 00 00 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
```

I'll read up on the protocol today, so maybe i can be a better help figuring this out :)

MichaelDvP, 2023-02-09:
Yes, the boiler set first two bytess to zero (activated and selflowtemp), a little chance that writing both values stops the heating. Try: `call system send "0b 08 16 00 00 00"`

hpeter1994, 2023-02-09:
I set the loglevel to notice so it would be easier to monitor. (if its better for you in debug mode, i'll turn it back)
Sadly setting both values didn't work either. 
The display on the boiler has no separate settings for heating and dhw off, you have to lower the temp to the minimum and 1 setting below the minimum is the off state ( - - ). And it will store the setting after waiting for three seconds. So probably thats why it set both parameters.

```
000+19:01:04.536 N 1382: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2D 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:04.776 N 1383: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C2 00 C2 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+19:01:14.497 N 1384: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2C 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:14.706 N 1385: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C4 00 C4 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+19:01:18.307 N 1386: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2B 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
smart-kazan:# call system send "0b 08 16 00 00 00"
000+19:01:21.522 N 1387: [emsesp] Me(0x0B) -W-> Boiler(0x08), UBAParameters(0x16), data: 00 00
000+19:01:21.543 N 1388: [emsesp] Boiler(0x08) -B-> All(0x00), UBAParameters(0x16), data: FF 1E 63 00 00 FC 02 01 01 64 0A 02
000+19:01:21.850 N 1389: [emsesp] Me(0x0B) -R-> Boiler(0x08), UBAParameters(0x16), length: 0x1B
000+19:01:24.491 N 1390: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2B 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:24.701 N 1391: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C4 00 C4 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+19:01:34.508 N 1392: [emsesp] Boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+19:01:34.736 N 1393: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2B 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:34.947 N 1394: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C2 00 C2 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+19:01:35.175 N 1395: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 64 01 16 6B 0A C6 5E 00 00 00 0A 8F 23 01 0C C6 80 00
000+19:01:44.480 N 1396: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2A 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:44.690 N 1397: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C1 00 C1 A0 00 00 03 00 00 37 3B 00 09 A5 00
000+19:01:54.508 N 1398: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 1E 01 2A 63 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+19:01:54.749 N 1399: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 00 C2 00 C2 A0 00 00 03 00 00 37 3B 00 09 A5 00

```

ahhoj, 2023-02-21:
@hpeter1994 : is there any change, did you succedd to turn off boiler? I'm experiencing the very same issue in a very same setup...

matteopedani, 2023-02-21:
No, and sadly  I can't make any  test  jet. Beacuse the other boiler user
it's afraid to spent huge gas bill, and I had to remove the interface.

Matteo

ahhoj, 2023-02-21:
@matteopedani : if it helps in any way, I can execute some tests, just kindly instruct me what should be checked.

MichaelDvP, 2023-02-22:
I've logged what my thermostat sends to stop the heating. I've set to permanent summermode and there is only one message:
`thermostat(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00`, sent every minute. This is the same as sending in ems-esp `selflowtemp 0` and `selburnpow 0`. Check if this works for you. It does not set heating off, and my boiler reports 13°C setflowtemp, but effectivly the heating stays off. Maybe we need also the set the 3rd byte zero (heatingpump), then try with command `system/send` `"0B 1A 00 00 00 00 00"`, or from terminal `call system send "0B 1A 00 00 00 00 00"`.
If this does not work, maybe the boiler only takes commands from thermostat. You don't have one, so `call system send "10 1A 00 00 00 00 00"` will not give a conflict. (not working in v3.5.0b7, needs update to 3.5.0 release)

ahhoj, 2023-02-22:
Tried all, but really nothing happens when I try to send telegram to dest 1A. If I try to send to 08 with type 16, I do observe the same behaviour as described above.
When I turn off heating on the boiler itself, type 16 messages got broadcasted afterwards that heating activated is turned off. From then I cannot turn it on from console/web, only from the boiler itself.

MichaelDvP, 2023-02-23:
Ups, sorry, destination should be boiler 08 and type 1A: 
`call system send "0B 08 1A 00 00 00 00 00"`, `call system send "10 08 1A 00 00 00 00 00"`

ahhoj, 2023-02-23:
Looks like it didn't have any effect either, effective heating was still on after this. I mean not just the heating activated telegram remained ON, but I see that the boiler actually heating as well.

`
000+00:02:34.771 N 1: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 30 01 A3 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:02:38.463 N 2: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 30 01 A2 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:02:38.673 N 3: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:02:48.475 N 4: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 30 01 9E 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:02:48.685 N 5: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:02:58.509 N 6: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:02:58.737 N 7: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 30 01 9B 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:02:58.947 N 8: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:02:59.178 N 9: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 08 64 00 35 4C 02 0A F5 00 00 00 02 02 0D 00 33 72 80 00
000+00:03:00.034 N 10: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
000+00:03:00.057 N 11: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 05 A0 C4
000+00:03:00.159 N 12: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 0x1B
000+00:03:00.193 N 13: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 30 64 00 00 F6 0A 01 0A 64 0A 02
000+00:03:00.283 N 14: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:03:00.318 N 15: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWW(0x33), data: 08 00 0F 00 00 46 00 02 4B 00 FF FF
000+00:03:08.532 N 16: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 30 01 98 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:08.773 N 17: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
ems-esp:# call system send "0B 08 1A 00 00 00 00 00"
000+00:03:12.336 N 18: [emsesp] Me(0x0B) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
000+00:03:12.455 N 19: [emsesp] Me(0x0B) -R-> boiler(0x08), UBASetPoints(0x1A), length: 0x1B
000+00:03:12.475 I 20: [emsesp] boiler(0x08) -W-> Me(0x0B), UBASetPoints(0x1A), data: <empty>
000+00:03:12.566 N 21: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 97 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:18.511 N 22: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 96 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:18.721 N 23: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:03:27.616 N 24: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 93 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:28.514 N 25: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:03:28.742 N 26: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 93 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:38.568 N 27: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 91 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:38.809 N 28: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:03:48.547 N 29: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 8F 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:48.757 N 30: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 10 02 10 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:03:58.547 N 31: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:03:58.778 N 32: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 8B 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:03:58.988 N 33: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 11 02 11 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:03:59.250 N 34: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 64 00 35 4C 02 0A F5 00 00 00 02 02 0D 00 33 72 80 00
000+00:04:00.106 N 35: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
000+00:04:00.129 N 36: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 05 A0 C5
000+00:04:00.231 N 37: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 0x1B
000+00:04:00.265 N 38: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 30 64 00 00 F6 0A 01 0A 64 0A 02
000+00:04:00.327 N 39: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
[... cut ...]
000+00:05:08.853 N 66: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 10 02 10 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:05:18.689 N 67: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 77 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:05:18.930 N 68: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 10 02 10 A0 00 00 03 00 00 08 E8 00 01 DA 00
000+00:05:19.691 N 69: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 01 77 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00

`

MichaelDvP, 2023-02-23:
> Looks like it didn't have any effect either

Looks to me as both works. Before command you have:
`UBAMonitorFast(0x18), data: 30 01 A3 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00`
byte0: selflowtemp 48°C,byte3: burnerpower 100%, byte5: heating active 
after command it's:
`UBAMonitorFast(0x18), data: 00 01 88 00 00 00 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00`
seflowtemp 0, burnerpower 0, heating off, etc.

ahhoj, 2023-02-23:
Yes, probably I should descibed that originally. It just a coincidence that the flow temperature just exceeded he heating temperature, and burnerpower is automatically set to 0 by the boiler, and only the pump was running. But, after some minutes, the heating still turned on, consuming gas and pump was running at full speed again.
All in all, I couldn't turn off the heating actually.

MichaelDvP, 2023-02-23:
> But, after some minutes, 

Yes, i already wrote that this command have to be repeated every minute.

ahhoj, 2023-02-23:
Ohh, sorry, my bad, I do really missed that part. And yes, you are right. Sending this every minute caused that heating do really turned off after either pump delay or burn min period time (I'm not sure, as those are both the same for me). When I stopped sending this command, the heating started after around 4 minutes.

Is is possible to incorporate this behaviour into EMS-ESP (hope so :) )? Or do I have to make this work from my controlling device?

MichaelDvP, 2023-02-24:
> Is is possible to incorporate this behaviour into EMS-ESP (hope so :) )? Or do I have to make this work from my controlling device?

It is a bit of a special function (out of normal scope of ems-esp), but we already have the wwtapactivated for shower coldshot function, so i think it's ok to add. Please try the 3.5.1-dev from my repo, i named it `force heating off` Please also try what is set when ems-esp is restarted and what is set if the boiler is powered off/on. We don't want to set the heating off on a powercut. (I check the first publish of selflowtemp to set the initial state).
(https://github.com/MichaelDvP/EMS-ESP32/releases)

ahhoj, 2023-02-24:
I'm about to try it out. In the mean time, may I ask why make it dependent on the boiler's selflowtemp initial value? I think since this feature would be used to control the boiler via ems-esp, only this device should hold and retain the information if the boiler should be turned off or on.
While thinking about it, I'm curious if it is possible at all if the required state is to be turned off, and the boiler will be power cycled, will it start heating instantly? Or is it possible to detect that there was a power cycle, and send this telegram instantly?
One more thing to consider: and when we disable this force heating off, can we send out a telegram to start heating instantly (probably just set selflowtemp to heating temp). At least, for now there is a quite delay (copule minutes) when the boiler actually starts heating after the last "force heating off" telegram is sent.

MichaelDvP, 2023-02-24:
The esp-flash have very limited write cycles, if you use this for controlling the boiler and change setting many times a day, you will kill your esp in a few month. We can add in settings section a power-on-default state for boiler-forceOff, and initialize with this state. 
After poweron (esp or boiler) we always have to wait for the bus become active to detect the devices. As soon as the boiler sends out it's state we can send the telgram, but i don't know what the boiler do in this first minute. Try it.

ahhoj, 2023-02-24:
1, Setting forceheatingoff either to on or off does the expected thing: sending (or not) the required telegram periodically, and making the boiler stop heating
2, when forceheating is off, and emsesp is restarted, remains off
3, when forceheating is off, and boiler is restarted, remains off
4, when forceheating is on, and emsesp is restarted, remains on
5, when forceheating is on, and boiler is restarted, remains on
6, when forceheating is on, the boiler is powered off, emsesp restarted, and boiler turned on, forceheating is turned off
7, when forceheating is off, the boiler is powered off, emsesp restarted, and boiler turned on, forceheating remains off

I understand your concerns regarding the flash lifecycle, and yes, I agree, don't store the value. Your suggestion to have a default power-on-state sounds good. What I'd like to avoid that after a full power outage (ems-esp, boiler), when the power returns, I think it would be better not to turn on heating. At least I don't feel the neccessity for it in the summer ;). And if we are in the middle of winter: the controlling device will turn it on anyway in couple of minutes.
My only remainging concern/feature request would be (if possible): when turning off forceheatingoff (so from a user perspective: I'd like to turn on heating), it would be nice that heating would start right away, and don't have to wait for the timeout to automatically start. Not the end of the world, but would be nice.

MichaelDvP, 2023-02-24:
I've added the option in settings. Please test. The init checks this setting and the state of boiler (selflowtemp == 0).
When disabeling forceheatingoff  i send now the heatingtemp as selflowtemp and burnermax to 100%, not sure if we shoud use burnermaxpow instead? Also not clear if restart needs a value for pump (3rd byte). I don't have a heatingpump at the boiler, my mixer controls it's own pump.

ahhoj, 2023-02-24:
Updated to latest. After that, I don't see anymore the force heating off entity at the boiler. It is still visible at customization, but have no value.
Turning on the "Start boiler with forced heating off" setting and restarting ems-esp didn't change anything regarding this.

MichaelDvP, 2023-02-24:
Yes, there was a typo in first upload, this is fixed now. Please download again. Also after disable i set now heatingTemp, burnMaxPower and pumpModMax.

ahhoj, 2023-02-24:
Tried the setting for both states with both preceding value with forceheatingoff. Also, tried how heating activates when forceheatingoff is turned off.
I think all work well, thank you very much for your support.

Maybe one more thing to consider until final release: it is somewhat "strange" to turn something ON so the boiler will turn off, and vica versa. Not a big problem, but maybe another name and inversed logic for on/off would be more straightforward.

MichaelDvP, 2023-02-24:
> it is somewhat "strange" to turn something ON so the boiler will turn off, and vica versa

Yes, but this is not a native boiler setting and it's more common to have a default "off" and explicide activate special functiions. 
The users need to notice that they are overriding a boiler setting, if you can suggest a wording that makes this clear, the logic can be changed. In #954 there is a similar problem, but there the boiler shuts down and needs periodic commands to stay on. Bosch is always very creative in changing settings and logic for different brands/models.

ahhoj, 2023-02-26:
>Yes, but this is not a native boiler setting and it's more common to have a default "off" and explicide activate special functiions.
The users need to notice that they are overriding a boiler setting, if you can suggest a wording that makes this clear, the logic can be changed

Good point, I didn't thought about it. And looking at the other issue... Well, nothing better wording comes to my mind. Maybe just let it as it is.

closed 2023-07-03 by proddy

martini1992, 2023-08-13:
v3.6.0 lost the "force heating off" option. Downgrading to v3.5.1 fixes the issue.

proddy, 2023-08-13:
Thanks for reporting. I'll create a new issue and I'm sure Michael will take a look when he's back.

MichaelDvP, 2023-08-13:
In 3.6 you have to create a schedule with timer option to use the heating off. It's not an extra option anymore.

martini1992, 2023-08-13:
I don't use the scheduling bits of ems-esp though, I use it as an interface to control my boiler through Home Assistant.