emsesp/EMS-ESP32 #953: Compiler ESM-ESP

isolhara, 2023-02-10:
I have a Bosch fresh water station FF20. I decrypted the telegrams. Would like to integrate this into EMS-ESP. Compiling with Visual Code brings error. How can I compile the EMS-ESP to get a BIN file.

proddy, 2023-02-10:
what fails? please post a screenshot

https://emsesp.github.io/docs/Coding/

isolhara, 2023-02-11:
Unfortunately I don't know English and translated with Google. Would German also be possible? Attached is the screenshot
[image]

proddy, 2023-02-11:
have you installed npm (nodejs)?

isolhara, 2023-02-11:
No, that was not installed. Now it worked. Thanks very much.
[image]

proddy, 2023-02-11:
Ok great. I've also updated the documentation to mention the pre-requisite for nodejs. Thanks for pointing that out!

closed 2023-02-19 by proddy