emsesp/EMS-ESP32 #1034: Missing Status of  VS1 for Buderus SM200

briegra, 2023-02-15:
I miss the status of the valve VS1. This is the valve which switches  from Boiler to Solar.
Can someone tell me where to find?

MichaelDvP, 2023-02-15:
The SM200 have a lot of possible configurations and only a few users have checked the telegrams in their configuration. 
I think this valve status could be in telegram 0x36A. Try in terminal `read 30 36A`, wait for a change of the valve and read again. I guess it is status 03 for off and 04 for on.
If you find it, please also suggest a shortname for mqtt and a fullname for the web dashboard.

briegra, 2023-02-15:
Great! Thank you for the quick response!
I can confirm that I get 03 for a switched off valve. Now I must wait for sunshine to confirm the status 04. Hope it will not last too long anymore.
My suggestions are:

  *   MQTT would be: “Valve_VS1”
  *   Fullname for dashboard: “Solar Valve (VS1) open”

BTW: I also miss the flame status. Is there also something available?

BR

Ralf

MichaelDvP, 2023-02-16:
>  BTW: I also miss the flame status. Is there also something available?

A flame in a SM200 is no good, smells like molten plastic.

briegra, 2023-02-16:
LOL

briegra, 2023-02-17:
I found something but I don’t know how to put it to a read command. This could be the flame status I’m looking for.
Can u help?

***@***.***

briegra, 2023-02-20:
Hi,

I can confirm both. 03 for closed and 04 for open valve.
Will you put this information to next version?

MichaelDvP, 2023-02-21:
> I can confirm both. 03 for closed and 04 for open valve.

Which byte? The telegram 0x36A looks like this: `data: 03 03 03 03 01 03 03 03 03 03 01 03`. We know, second byte is pump2, 5th byte VS2, 11th byte pump1, I can add if i know the right position. 

For the flame we need motre info from your system and what you've found.

briegra, 2023-02-21:
Let me Double Check it again. I had the change in the first one. When I remember right there was another change in the queue.

briegra, 2023-02-25:
This is what I got in the different state:
[emsesp] solar(0x30) -W-> Me(0x0B), SM100Status2(0x036A), data: 04 03 03 03 03 03 03 03 03 03 03 03 --> Valve open
[emsesp] solar(0x30) -W-> Me(0x0B), SM100Status2(0x036A), data: 03 03 03 03 03 03 03 03 03 03 03 03 --> Valve closed

MichaelDvP, 2023-02-25:
Ok, lucky me, i have already implemented the right offset. Try 3.5.1-dev from https://github.com/MichaelDvP/EMS-ESP32/releases or compile the latest tech-upgrade branch from here.

briegra, 2023-02-25:
Great – Thank you very much!

proddy, 2023-04-23:
assuming this is resolved now. Otherwise please re-open.

closed 2023-04-23 by proddy