emsesp/EMS-ESP32 #1035: MQTT HomeAssistant relay control

aarturoo, 2023-02-16:
I bought an EMS module from you and connected a WemosD1 S32 and a relay module to it. I added all this to my smart home "HomeAssistant". Unfortunately, I don't see any possibility for mqtt to add control of this relay directly from "HomeAssistant".

configuration.yaml I add this:

mqtt:
  switch:
    - name: "Piec1"
      state_topic: "ems-esp/analogsensor_22"
      command_topic: "ems-esp/analogsensor_22/set"
      payload_on: "1"
      payload_off: "0"

But isn't work.

Is it possible for me to be able to turn the relay on and off from within the smart home?

MichaelDvP, 2023-02-16:
For connecting a relais, especially the dangerous D1-shields, read: https://bbqkees-electronics.nl/wiki/gateway/using-gateway-gpio.html

Switching a gpio depends on ems-esp-version. You have not added the support-info, so i guess actual:
- add analog output, gpio 22, name piec1
- use command `ems-esp/analogsensor/piec1`
- or use `ems-esp/analogsensor/setvalue` with payload id:22 and value:on/off (or 1/0, true/false)

With HA integration i can't help.

aarturoo, 2023-02-16:
>

aarturoo, 2023-02-16:
> commented

> For connecting a relais, especially the dangerous D1-shields, read: https://bbqkees-electronics.nl/wiki/gateway/using-gateway-gpio.html
> 

Yes, I know. That's why I made my own shield with an optoisolator. In addition, the relay is to switch the voltage-free (signal) contact.

 
> Switching a gpio depends on ems-esp-version. You have not added the support-info, so i guess actual:
>
> * add analog output, gpio 22, name piec1
> * use command `ems-esp/analogsensor/piec1`
> * or use [...]

I add analog output GPIO22, name relay.

I add configuration.yaml:
mqtt:
  switch:
    - name: "Piec1"
      state_topic: "ems-esp/analogsensor/relay"
      command_topic: "ems-esp/analogsensor/relay/command"
      payload_on: "ON"
      payload_off: "OFF"

But it doesn't work.

The relay works for me only as I directly control it from the EMS side. Unfortunately with the home assistant after adding the above code and trying to control it, nothing happens.

MichaelDvP, 2023-02-17:
Thx for the support info, you have ha-discovery enabled, so we need to fix this discovery to set the command topic right.
The command topic should be `ems-esp/analogsensor/relais`
The state-topic `ems-esp/analogsensor`
and the val_tpl `{{value_json['22'].value}}` for the nested mqtt (your setting) or for single: `{{value_json['relais']}}`

MichaelDvP, 2023-02-17:
Please check if discovery in https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works. (I can not test, i don't use HA).

closed 2023-07-03 by proddy