emsesp/EMS-ESP32 #1037: Add support for W5500 ethernet

proddy, 2023-02-18:
From what I've read the Wiznet 5500 is not supported by the ESP-IDF natively as it doesn't use the SPI interface and the PHY module. It also has it's own hardware TCP/IP engine so not using LWiP which the HTTP and mqtt clients use. It'll take some time to find how to get this working alongside the other modules.

bbqkees, 2023-02-23:
The ESP32 has a hardware Ethernet MAC (EMAC). To add Ethernet to an ESP32 board you need to add a simple Ethernet PHY chip like the LAN8720.
[image]

The PHY chip is connected to the ESP32 via a dedicated RMII interface.
You can find lots of development boards with the LAN8720 (mainly because it is a very cheap chip).

For reasons unknown they removed the EMAC from the ESP32-S3.

However, the ESP IDF supports both internet EMAC and external Ethernet MAC+PHY ones like in the W5500 module.
External Ethernet modules communicate via SPI.
https://github.com/espressif/esp-idf/blob/master/examples/ethernet/README.md#using-spi-ethernet-modules

Currently there are no ESP32-S3 dev boards with Ethernet afaik. So apparently it is not trivial to get it working properly.
Also probably lots of library support will be missing and the implementation if lots of these features were only added Q4 last year.

Some further reading here:
https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic

proddy, 2023-03-04:
with this change I may look into moving from ESPAsyncWebServer to the IDF native https server and TCP stack

rjjrbatarao, 2023-07-17:
Hi @proddy  async is enough I have a working w5500 on esp32 both supported in arduino ide and idf, it uses SPI but with additional pin for interrupt, it will not work without this pin :)
```
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_netif.h"
#include "esp_eth.h"
#include "esp_event.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#define CONFIG_EXAMPLE_USE_W5500 1
#define CONFIG_ETH_USE_SPI_ETHERNET 1 
#if CONFIG_ETH_USE_SPI_ETHERNET
#include "driver/spi_master.h"
#endif // CONFIG_ETH_USE_SPI_ETHERNET

static const char *TAG = "eth_example";

#if CONFIG_EXAMPLE_USE_SPI_ETHERNET
#define INIT_SPI_ETH_MODULE_CONFIG(eth_module_config, num)                                      \
    do {                                                                                        \
        eth_module_config[num].spi_cs_gpio = CONFIG_EXAMPLE_ETH_SPI_CS ##num## _GPIO;           \
        eth_module_config[num].int_gpio = CONFIG_EXAMPLE_ETH_SPI_INT ##num## _GPIO;             \
        eth_module_config[num].phy_reset_gpio = CONFIG_EXAMPLE_ETH_SPI_PHY_RST ##num## _GPIO;   \
        eth_module_config[num].phy_addr = CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR ##num;                \
    } while(0)

typedef struct {
    uint8_t spi_cs_gpio;
    uint8_t int_gpio;
    int8_t phy_reset_gpio;
    uint8_t phy_addr;
}spi_eth_module_config_t;
#endif

/** Event handler for Ethernet events */
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(TAG, "Ethernet Link Up");
        ESP_LOGI(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        break;
    }
}

/** Event handler for IP_EVENT_ETH_GOT_IP */
static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
                                 int32_t event_id, void *event_data)
{
[... 191 lines cut ...]
#endif // CONFIG_ETH_USE_SPI_ETHERNET

    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    /* start Ethernet driver state machine */
#if CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
#endif // CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET
#if CONFIG_EXAMPLE_USE_SPI_ETHERNET
    for (int i = 0; i < CONFIG_EXAMPLE_SPI_ETHERNETS_NUM; i++) {
        ESP_ERROR_CHECK(esp_eth_start(eth_handle_spi[i]));
    }
#endif // CONFIG_EXAMPLE_USE_SPI_ETHERNET
}

void setup(){
  app_main();
}

void loop(){
  
}
```

proddy, 2023-07-24:
thanks @rjjrbatarao ! This is something we do want to add so your code will really help.

bbqkees, 2023-07-24:
This was also discussed in another issue about board compatibility.
Getting the W5500 to work would be nice, but not a huge priority at the moment.

The W5500 is needed in combination with the ESP32-S3.
I tried to look for the actual speed improvements of the ESP32-S3 over the ESP32 but there is no good comparison available. 
Also depends on the use case and firmware of course.
Otherwise for EMS-ESP the ESP32 with some PSRAM and bigger flash should work fine for the next E32 for another few years at least.

Jason2866, 2023-10-01:
There is support for SPI Ethernet in the WiP Arduino 3.0. And there is a S3 board from Lilygo with a W5500. SPI Ethernet with Arduino Core 2.0.x will not work. It clashes with the way how SPI is done here.

bbqkees, 2023-10-02:
There is also basic support for the W5500 in the Espressif IDF.
But actually nothing beats the stability of the ESP32 with the LAN8720.
That combo works mostly in hardware and is pretty rock solid.
From what I read the W5500 is not always performing that well.
My E32 successor will have the ESP32 with bigger flash, some PSRAM and the LAN8720.

proddy, 2023-11-19:
Closing this. We'll stay with the LAN8720 for now.

closed 2023-11-19 by proddy

mercenaruss, 2023-12-05:
> That combo works mostly in hardware and is pretty rock solid.

Can you share with me the rock solid schematic?

karol27, 2023-12-18:
Will WT32-ETH01 work? It has LAN8720. Or code still need some modifications to work with WT32-ETH01?

proddy, 2023-12-19:
> WT32-ETH01

yes, supported. See https://github.com/emsesp/EMS-ESP32/issues/938#issuecomment-1455875576

bbqkees, 2023-12-19:
> Will WT32-ETH01 work? It has LAN8720. Or code still need some modifications to work with WT32-ETH01?

The WT32-ETH01 is supported. Just use the E32 board profile in the EMS-ESP settings.
Add an EMS interface board to the following GPIO pins of the ETH module:
[image]

However the WT32-ETH01 is not recommended anymore for EMS-ESP 3.6 or later.
3.5.1 Seems to be the last properly working firmware for it. 3.4.x are also very solid on this module.
With Ethernet and other services enabled in the firmware the RAM of the chip is not sufficient for larger EMS systems.
With a simple boiler and with all non-essential firmware features turned off it may still work reliably.
However if you connect this board loaded with 3.6.x to a heat pump or a larger EMS system with additional mixer modules or a RC310 thermostat with dozens of entities the WT32-ETH01 will struggle to process all the entities in the tiny ESP32 memory, especially with MQTT Discovery enabled.

ranX, 2025-04-15:
Hi to all,

hopefully it is OK, when I revive this this old topic and ask, if it makes sense, to reopen it?
At the time, it was closed, there was only the Lilygo ESP32-S3 with onboard W5500 ethernet.
By now waveshare with the [ESP32-S3-ETH](https://www.waveshare.com/wiki/ESP32-S3-ETH) added at least one more.

To me it looks pretty tempting as it not only has ethernet but also the option to supply it remotely by PoE.
As the platform including the PoE module is available for about 20-25€ from China, this might be a nice thing for people like me, that prefer a wired solution. 

Thanx for reading and commenting!
Best regards
ranX

Jason2866, 2025-04-15:
@ranX Using actual Arduino core 3.1.x or 3.2.x and the W5500 is fully supported, only the code needs some little changes.

ranX, 2025-04-15:
@Jason2866 Well this sounds promising!
So I understand there's still some work to do but the implementation of the W5500 interface is in progress.
I'd love to contribute but I'm more handcrafter than developer - thus I appreciate your efforts even more!

proddy, 2025-04-20:
EMS-ESP is still on Arduino Core 2.0.17 / IDF 4.4.7. I was working on trying to switch to Arduino Core 3.1.3 (based on IDF 5.3.2) but the default lib uses 20% more heap and causes EMS-ESP to crash, so now looking at Jason's hybrid mode to include `mbedtsl` which is needed for MQTT TLS (via `wifi_secure.h`) but it's super hard to get right.

xqoperator, 2026-01-07:
Does anyone have a fork that has W5500 support?