emsesp/EMS-ESP32 #1071: external sensor turns into unknown device when rebooting the ems

breuerm710, 2023-02-20:
### PROBLEM DESCRIPTION
When I turn the burner off and on again the sensor list shown an unknown device in addition to the DS18B20 which is connected as external sensor. With every cycle the list of unknown devices grows by one.
Rebooting the EMS clears the list of unknown devices.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :
```lua
  System information output here:
{
  "System Info": {
    "version": "3.5.0",
    "platform": "ESP32",
    "uptime": "000+08:35:20.109",
    "uptime (seconds)": 30920,
    "free mem": 99,
    "max alloc": 39,
    "free app": 7,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -70,
    "IPv4 address": "192.168.5.120/255.255.255.0",
    "IPv4 gateway": "192.168.5.1",
    "IPv4 nameserver": "192.168.5.1",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 3942,
    "MQTT queued": 1,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 1,
    "temperature sensor reads": 6183,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 2132,
    "bus reads (tx)": 62,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 478075,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 1,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 5
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "03.05",
      "entities": 66,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA",
      "handlers ignored": "0x04"
    },
    {
      "type": "thermostat",
      "name": "RC35",
      "device id": "0x10",
      "product id": 86,
      "version": "01.06",
      "entities": 59,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x13 0x42 0x48 0x47 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers ignored": "0x01 0xA4 0x1A"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.03",
      "entities": 0
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x73",
      "product id": 50,
      "version": "146.3",
      "entities": 0
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x3C",
      "product id": 146,
      "version": "04.39",
      "entities": 0
    }
  ]
}

### TO REPRODUCE
_Steps to reproduce the behavior:_
power off burner, power on again

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no unkown devices when power cycling the burner

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2023-02-20:
how do you "turn the burner off and on again"?

breuerm710, 2023-02-20:
There is a switch on the panel.
[image]

MichaelDvP, 2023-02-21:
I think there is a issue with bus connection and ems-esp is mostly reading garbage.: 
```
   "bus telegrams received (rx)": 2132,
    "bus incomplete telegrams": 478075,
    "bus rx line quality": 1,
```
Check cable and connections.

(BTW: if connecting the E32 gateway via wifi it is better to use custom board profile and set the Eth-Phy-type to disabled. This saves a lot of memory. But i don't think it is related to this issue)

breuerm710, 2023-02-21:
The incomplete are coming from the fact that I turned the heater off during the day (not needed as it is warm enough) and power it on again at the evening. Every evening it adds one unknown device. I will check it today again and expect a 3rd one. 

I changed to custom board profile and due to the needed reboot the unknown devices are gone. So expect only one new this evening.

proddy, 2023-02-21:
turn off, Telnet to the console, `watch raw` to see all traffic, turn switch on and paste back here what telegrams come in during the first 10 seconds.

MichaelDvP, 2023-02-21:
Could you check before turning the heating on, if there is this new device from garbage? 
And before starting the heating: set loglevel to all in web-system-log, leave page open. Now start the heating, wait some minutes and download the log. Attach the log here (drag&drop).

MichaelDvP, 2023-02-21:
Ups, proddy was faster, telnet log is also ok.

MichaelDvP, 2023-02-21:
Tested this with my boiler and a E32 gateway connected direct to the bus and powered by bbqkees powersupply (one dallas connected). I have a different boiler (GB125) but identical thermostat and controller. I set the rocker-switch on MC10 to off, wait 30 sec and put it on again. I did not get any garbage, incomplete telegram this time. Also after the restart the devices are unchanged.
[log-heating_off-on.txt](https://github.com/emsesp/EMS-ESP32/files/10795020/log-heating_off-on.txt)
There is an interesting communication between controller and boiler with telegram 0x04.

breuerm710, 2023-02-21:
I just want to power it on and checking the web interface there is already an unknown device listed.
[image]
[image]
I stored the log but haven't found something most likely as is was already overwritten.
Will reboot the device and power on the burner now.

breuerm710, 2023-02-21:
Restarting the burner did not generate an unknown device. Nevertheless stored the log if it is from any interest. Repeated it with 1 minutes burner off but again no unknown device.
So looks like my fault. I obviously didn't check the web interface very often during the burner was turned off but checked it after I started it. So the observation of the new device and the restart seem to be only coincidental.
The only change was that I switched to a custom board profile but as an unknown device was nevertheless somewhen generated this should not be a game changer. I'm not aware of a way to trigger the issue.

Do you want me to close the ticket or is there anything else I could do for you?

breuerm710, 2023-02-22:
During todays burner power off time two new device occured.
[image]

breuerm710, 2023-02-24:
[image]
Is there anything I can do to help you find the problem or is the behavior expected?

proddy, 2023-07-03:
Not sure what to do with this issue. If it persists with the latest 3.6.x we're about to release then please re-open.

closed 2023-07-03 by proddy