emsesp/EMS-ESP32 #1076: Turkish language support

proddy, 2023-02-21:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/868

<div type='discussions-op-text'>

<sup>Originally posted by **dreixin** December 31, 2022</sup>
I am trying to add Turkish language support to EMS-ESP. Anyone willing to help is welcome.</div>

MichaelDvP, 2023-07-02:
When merging the it-translations i see that there are a lot of tr-translations missing. 
Reopening, maybe someone can help and write into this excel sheet.
[locale_translations_w_it.xlsx](https://github.com/emsesp/EMS-ESP32/files/11929185/locale_translations_w_it.xlsx)

proddy, 2023-07-03:
@dreixin could you please take a look

dreixin, 2023-07-03:
Hi everyone, 
As you know we had a big earthquake and personally I had so many other stuff to deal with during this time and I had to leave this project on a lover priority for a long time. attached is a quick touch with all the translations done until line 270.
I will spare more time on the project and try to do my best to finish as soon as possible.
[locale_translations_w_it.xlsx](https://github.com/emsesp/EMS-ESP32/files/11936715/locale_translations_w_it.xlsx)

proddy, 2023-07-15:
We'll include the ones up to line 270 for now, as I want to release the next version of EMS-ESP soon.

MichaelDvP, 2023-07-15:
The translations up to line 270 are included in dev since 27.feb. See: https://github.com/emsesp/EMS-ESP32/commit/94749e46154b7cf62129fa170e6b6d3df013375b

proddy, 2023-07-15:
oh, I thought more were added

dreixin, 2023-07-29:
[locale_translations_w_it.xlsx](https://github.com/emsesp/EMS-ESP32/files/12206538/locale_translations_w_it.xlsx)
on going process, I hope it will be completed in a week or so.

MichaelDvP, 2023-07-29:
I've added the new translations to my dev2, you can test here https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
I'll make a PR when complete.

dreixin, 2023-07-31:
[locale_translations_w_it_v2.xlsx](https://github.com/emsesp/EMS-ESP32/files/12211665/locale_translations_w_it_v2.xlsx)
last 30-40 translations left

proddy, 2023-08-11:
We'll add the remaining translations later in a minor release of 3.6.x

dreixin, 2023-09-10:
Finished all the translations on Turkish. 
Attached is the final file. 
[locale_translations_w_it_v2.xlsx](https://github.com/emsesp/EMS-ESP32/files/12567519/locale_translations_w_it_v2.xlsx)

MichaelDvP, 2023-09-10:
Thank you, i'll add it to the dev version.

In the dev there are also some new entities and web texts, can you please complete the translation too.
```
MAKE_WORD_TRANSLATION(commands_response, "get response","Hole Antwort","Verzoek om antwoord","","","","","") // TODO translate
MAKE_WORD_TRANSLATION(coldshot_cmd, "send a cold shot of water", "", "", "", "", "", "", "", "") // TODO translate

// burner energy
MAKE_TRANSLATION(nrgHeat, "nrgheat", "energy heating", "Energie Heizen", "", "", "", "", "", "", "") // TODO translate
MAKE_TRANSLATION(nrgWw, "nrgww", "energy dhw", "Energie Warmwasser", "", "", "", "", "", "", "") // TODO translate
MAKE_TRANSLATION(nomPower, "nompower", "nominal Power", "Brennerleistung", "", "", "", "", "", "", "") // TODO translate

// remote thermostat
MAKE_TRANSLATION(remotehum, "remotehum", "room humidity from remote", "Raumfeuchte Remote", "", "", "", "", "", "", "") // TODO translate
```
And from webpage index.tsl
```
  RESET: 'Reset', // TODO translate
  APPLY_CHANGES: 'Apply Changes ({0})', // TODO translate
  UPDATE: 'Update', // TODO translate
  EXECUTE: 'Execute', // TODO translate
  VERSION_ON: 'You are currently on', // TODO translate
  RELEASE_IS: 'release is', // TODO translate
  DOWNLOAD_SCHEDULE_TEXT: 'Download Scheduler Events', // TODO translate

  BLOCK_NAVIGATE_1: 'You have unsaved changes', // TODO translate
  BLOCK_NAVIGATE_2: 'If you navigate to a different page, your unsaved changes will be lost. Are you sure you want to leave this page?', // TODO translate
  STAY: 'Stay', // TODO translate
  LEAVE: 'Leave', // TODO translate
  SCHEDULER: 'Scheduler', // TODO translate
  SCHEDULER_HELP_1: 'Automate commands by adding scheduled events below. Set a unique Name to enable/disable activation via API/MQTT.', // TODO translate
  SCHEDULER_HELP_2: 'Use 00:00 to trigger once on start-up', // TODO translate
  SCHEDULE: 'Schedule', // TODO translate
  TIME: 'Time', // TODO translate
  TIMER: 'Timer', // TODO translate
  SCHEDULE_UPDATED: 'Schedule updated', // TODO translate
  SCHEDULE_TIMER_1: 'on startup', // TODO translate
  SCHEDULE_TIMER_2: 'every minute', // TODO translate
  SCHEDULE_TIMER_3: 'every hour', // TODO translate
  CUSTOM_ENTITIES: 'Custom Entities', // TODO translate
  ENTITIES_HELP_1: 'Fetch custom entities from the EMS bus', // TODO translate
  ENTITIES_UPDATED: 'Entities Updated', // TODO translate
  WRITEABLE: 'Writeable', // TODO translate
  SHOWING: 'Showing', // TODO translate
  SEARCH: 'Search', // TODO translate
  CERT: 'TSL root certificate (leave blank to disable TSL)' // TODO translate
```

dreixin, 2023-09-14:
[translate_new .xlsx](https://github.com/emsesp/EMS-ESP32/files/12609253/translate_new.xlsx)

proddy, 2023-10-01:
In 3.6.2

MichaelDvP, 2023-10-01:
The entities are complete, on the webpage still some translation are missing.

proddy, 2023-11-19:
closing. Nothing we can do - the translations need to come from the community

closed 2023-11-19 by proddy