emsesp/EMS-ESP32 #1078: MQTT settings randomly dissapeat

proddy, 2023-02-23:
I've been doing lots of re-work on MQTT and noticed a few times the whole MQTT settings would be reset. Haven't looked into the reason why.  Have you noticed this too @MichaelDvP ?

MichaelDvP, 2023-02-23:
Could be this: https://github.com/emsesp/EMS-ESP32/blob/3e9f63a913403c010720b5c98a9e0627f24e08cb/lib/framework/StatefulService.h#L17 
We have this 8k for customizations, but most configs does not need it. If alloc fails the config safe clears the config (i guess). 
I've tested to go back to 1k and set the buffer for each config-file independent. I could make a PR for this.

proddy, 2023-02-23:
I don't believe FS_BUFFER_SIZE is used at all.

The MQTT isn't the largest file either. Pressing the button on the dev board shows me

[image]

proddy, 2023-03-01:
closing this. I think it happens when USB flashing and may be driver related. Will re-open if I can reproduce it.

proddy, 2023-03-03:
re-opening. happened again upgrading from 3.5.0 to 3.6.0. time to debug....

MichaelDvP, 2023-03-03:
Was this only mqtt? Or also other settings affected? We had this very few times with settings (board-profile), but it's long ago.

proddy, 2023-03-03:
yeah, only the MQTT is missing. I think it's something in `System::check_upgrade()`

proddy, 2023-03-03:
I tested 3.5.0 -> 3.6.0 and back and can't reproduce.

proddy, 2023-03-04:
I noticed also some of the settings (like shower time, led etc) were also reset. It happens between version uploading but I can't reproduce it. I'll keep this issue open in case it happens again

closed 2023-04-23 by proddy