emsesp/EMS-ESP32 #1080: List of problems after updating to Version 3.5.0

mtbsteve, 2023-02-25:
### PROBLEM DESCRIPTION
I just upgraded my EMS-ESP32 module from 3.4.4 to version 3.5.0. Therefore I used the EMS-ESP flasher since the upload on the web page of the device failed with a "Server Error".

I then rebooted, and connected to the ems access point. Read back in the 3.4.4 settings and configuration files and rebooted again.
So far so good.

After connecting to the EMS bus of my Buderus WLW196i heatpump, I am getting now random errors when I am open the dashboad.

1. Randomly, not all of my connected devices are listed
2. Random number of parameters are retrieved 
3. My defined favorites in the parameter list are gone
4. Randomly, I am getting the error message "Problem loading (Offset is outside the bounds of the DataView)"
   When this error occurs, only a reboot of the EMS-ESP helps. 

All was working 100% ok with firmware 3.4.4 before. No changes the the bus setup nor network.

All other device settings, like Network, NTP, MQTT, System, look good.

My environment: 
EMS-ESP Version v3.5.0
Device (Platform / SDK) ESP32 / v4.4.2
EMS-ESP connected by Ethernet

mtbsteve, 2023-02-25:
To add: I just downgraded back to 3.4.4. Everything is working again.
So there seems to be a fundamental problem with the 3.5 code running on my ESP32.
Any suggestions?

proddy, 2023-02-25:
what ESP32 are you using?

mtbsteve, 2023-02-26:
> what ESP32 are you using?

Thanks for getting back. I am using the E32 Gateway from BBQkees https://bbqkees-electronics.nl/product/gateway-e32-ethernet-ausgabe/?lang=de purchased in September 2022. The device is connected via Ethernet and powered by the BBQkees 12V power supply.

MichaelDvP, 2023-02-26:
After repartitioning the data partition is unformated, so mount failed. EMS-ESP formats the partitin and remounts than.
With empty data partition EMS-ESP uses default settings. For board-profile it tries to init the ETH chip and sets E32-board if chip answers, otherwise S32-board. Because at this point the network isn't started, you get this message. Once the board profile is set, you don't get thi message.
I think you have left brower open when updating, so browser have cached the v3.4.4 webpage. When fetching data it now gets v3.5.0 data formats, resulting in "problem fetching data" or scrambled webpage. You have to reload the page to brower to update the scripts. With web-upload the page reload is triggered automatically, but always better to reload page after update or close/reopen browser.

mtbsteve, 2023-02-26:
@MichaelDvP I just reflashed 3.5 with the same results. I cleared the cache, I accessed the EMS-ESP from a different browser and also tried on a different computer. No difference, Same errors as originally reported.

The problem seems to be in the way how the 3.5 version accesses the EMS bus compared to 3.4.4. In the "EMS Bus and Activity Status" overview, I see a high error rate of EMS Tx reads, also, the  EMS bus status shows permanent connects/disconnects every few seconds. With 3.4.4, the error rate is zero.
My EMS bus installation is default Buderus, I have a Buderus RC310 control unit, a Buderus MM100 Mixer and a Buderus RC100 Thermostat connected to the Logatherm 196i.
I just reinstalled 3.4.4, and again, everything is working seamlessly.

MichaelDvP, 2023-02-26:
There is no change in bus handling. A lot of people with similar sytems have v3.5 running. 
Try download the bin file again, check MD5, set ems-esp to factory defaults and configure all from scratch.

mtbsteve, 2023-02-26:
No success. I have to admit that I dont know what you mean with "check MD5" Could not find it in the documentation.
Here is a debug logfile and my system settings summary. 
1. The info file:
At the time I downloaded the info file only two devices were listed as connected, the Boiler and one Thermostat. The other 2 devices are missing. Tx quality is at 73%. Both varies over time, sometimes you get all 4 devices with all parameters listed, and a few seconds later they disappear, sometimes none listed as connected. Tx quality jumps randomly between 33 and 80 percent. 
[emsesp_info (2).txt](https://github.com/emsesp/EMS-ESP32/files/10834295/emsesp_info.2.txt)

2. The Logfile with ALL on:
[log.txt](https://github.com/emsesp/EMS-ESP32/files/10834302/log.txt)

I have no insights into the EMS protocol, but I found entries like:
```TRACE 926: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A7), data: <empty>
 DEBUG 927: [emsesp] This telegram (RC300Monitor) is not recognized by the EMS bus
```
That also occurs randomly, and seems to correlate with the sudden disappearance of devices or parameters in the dashboard.

Maybe that helps to trace down the issue.

proddy, 2023-02-26:
I can't see anything odd in the settings. From the log it looks like API command calls coming in - do you have any other processes/jobs/systems running that work with EMS-ESP? A log file from boot-up is useful so we can see the handshaking. Also switch off MQTT for now until you get your system stable as it just clutters up the log

MichaelDvP, 2023-02-27:
The lot of commands on startup are strange, comming in in 100ms cycles. Have you changed the mqtt enable right before logging?
But i think it's memory related. The info is only 30 sec old, very low heap and max_alloc, 180 mqtt messages queued and at this point only 23 boiler entities and no entities for thermostat. Maybe it will restart after a minute uptime when queueing the other HA-entity-configs. Please check the uptime in system if it ever gets to more than 5 minutes.

If this is the cause, the mqtt-changes form @pswid maybe helps. Coud you try the v3.5.1-dev from here: https://github.com/MichaelDvP/EMS-ESP32/releases.

mtbsteve, 2023-02-27:
Thanks @proddy @MichaelDvP for getting back.

> do you have any other processes/jobs/systems running that work with EMS-ESP?

Yes, an ioBroker instance is connected via the EMS-ESP adapter to EMS-ESP

> Have you changed the mqtt enable right before logging?

No. MQTT is enabled right from startup.

> Please check the uptime in system if it ever gets to more than 5 minutes.

I have not stopped the uptime, but I would guess that it never exceeds maybe 30-60 seconds before it disconnects from the EMS-bus. Typically, it takes just a few seconds until the entities or devices in the list change or disappear.

I will test 3.5.1dev in a couple of days and report.