emsesp/EMS-ESP32 #1098: hc don't change to night.

proddy, 2023-03-04:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1096

<div type='discussions-op-text'>

<sup>Originally posted by **JokerGermany** March  4, 2023</sup>
[image]
[image]

[image]
[image]

Locks like the option isn't transfered to ems-esp?</div>

proddy, 2023-03-04:
Michael fixed with https://github.com/emsesp/EMS-ESP32/pull/1097

Please test

MichaelDvP, 2023-03-04:
BTW: the title is wrong, changing to night works, but the modetype wasn't updated after this change.

JokerGermany, 2023-03-16:
Thanks
Didn't notice, that this happened again.

closed 2023-03-16 by proddy