emsesp/EMS-ESP32 #1102: v3.6.0-dev.2 - DallasSensors problem

sq4bja, 2023-03-04:
Hi,
v3.6.0-dev.2 - with Buderus GB072 and Domoicz (2023.1 (build 15109)) with Raspberry.
After upgrade form v3.6.0-dev.0 i have lost all dallas sensors.
In MQTT i can see new option - Domoticz Discovery, but i can`t enable it.

Tom

proddy, 2023-03-05:
what do you mean by 'lost'? Do you see them in the WebUI and Console? Is the data being published in MQTT? Or are the MQTT discovery topics gone? Is it just the Dallas sensors? Did you check your MQTT settings as we've seen issues there too.

sq4bja, 2023-03-05:
The problem is with domoticz integration. After the EMS-ESP update, the temperature sensors stop being updated in Domoticz. There are also no other new temperature sensors. They exist in MQTT, but apparently the structure of the frame is illegible for Domoticz. Temperature values are also updated in the EMS-ESP web interface.

MichaelDvP, 2023-03-05:
Does it work if you switch to single mqtt format?

proddy, 2023-03-05:
could you compare the MQTT topics between 3.6.0-dev-0 and 3.6.0-dev-2 with MQTTExplorer to see if you can spot the difference? that would help us.

sq4bja, 2023-03-05:
Ok, i compared topics and data dallas sensors. Dev 0 works with domoticz, but dev 2 does not. 

```
3.6.0.dev.2
homeassistant/sensor/ems-esp
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-B07E-AE13-1901'].temp if value_json['28-B07E-AE13-1901'] is defined}}","obj_id":"dallassensor_28-B07E-AE13-1901","uniq_id":"dallassensor_28-B07E-AE13-1901","name":"28-B07E-AE13-1901","dev":{"ids":["ems-esp"]},"avty":[{"t":"ems-esp/dallassensor_data","val_tpl":"{{'online' if value_json['28-B07E-AE13-1901'] is defined else 'offline'}}"}],"avty_mode":"all"}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-F043-2800-0080'].temp if value_json['28-F043-2800-0080'] is defined}}","obj_id":"dallassensor_28-F043-2800-0080","uniq_id":"dallassensor_28-F043-2800-0080","name":"28-F043-2800-0080","dev":{"ids":["ems-esp"]},"avty":[{"t":"ems-esp/dallassensor_data","val_tpl":"{{'online' if value_json['28-F043-2800-0080'] is defined else 'offline'}}"}],"avty_mode":"all"}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-28DA-B413-1901'].temp if value_json['28-28DA-B413-1901'] is defined}}","obj_id":"dallassensor_28-28DA-B413-1901","uniq_id":"dallassensor_28-28DA-B413-1901","name":"28-28DA-B413-1901","dev":{"ids":["ems-esp"]},"avty":[{"t":"ems-esp/dallassensor_data","val_tpl":"{{'online' if value_json['28-28DA-B413-1901'] is defined else 'offline'}}"}],"avty_mode":"all"}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-0E2C-2800-0080'].temp if value_json['28-0E2C-2800-0080'] is defined}}","obj_id":"dallassensor_28-0E2C-2800-0080","uniq_id":"dallassensor_28-0E2C-2800-0080","name":"28-0E2C-2800-0080","dev":{"ids":["ems-esp"]},"avty":[{"t":"ems-esp/dallassensor_data","val_tpl":"{{'online' if value_json['28-0E2C-2800-0080'] is defined else 'offline'}}"}],"avty_mode":"all"}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-6E76-BA13-1901'].temp if value_json['28-6E76-BA13-1901'] is defined}}","obj_id":"dallassensor_28-6E76-BA13-1901","uniq_id":"dallassensor_28-6E76-BA13-1901","name":"28-6E76-BA13-1901","dev":{"ids":["ems-esp"]},"avty":[{"t":"ems-esp/dallassensor_data","val_tpl":"{{'online' if value_json['28-6E76-BA13-1901'] is defined else 'offline'}}"}],"avty_mode":"all"}

3.6.0.dev.0
homeassistant/sensor/ems-esp
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-B07E-AE13-1901'].temp}}","object_id":"dallassensor_28-B07E-AE13-1901","uniq_id":"dallassensor_28-B07E-AE13-1901","name":"28-B07E-AE13-1901","dev":{"ids":["ems-esp"]}}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-F043-2800-0080'].temp}}","object_id":"dallassensor_28-F043-2800-0080","uniq_id":"dallassensor_28-F043-2800-0080","name":"28-F043-2800-0080","dev":{"ids":["ems-esp"]}}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-28DA-B413-1901'].temp}}","object_id":"dallassensor_28-28DA-B413-1901","uniq_id":"dallassensor_28-28DA-B413-1901","name":"28-28DA-B413-1901","dev":{"ids":["ems-esp"]}}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-0E2C-2800-0080'].temp}}","object_id":"dallassensor_28-0E2C-2800-0080","uniq_id":"dallassensor_28-0E2C-2800-0080","name":"28-0E2C-2800-0080","dev":{"ids":["ems-esp"]}}
{"dev_cla":"temperature","stat_t":"ems-esp/dallassensor_data","unit_of_meas":"°C","val_tpl":"{{value_json['28-6E76-BA13-1901'].temp}}","object_id":"dallassensor_28-6E76-BA13-1901","uniq_id":"dallassensor_28-6E76-BA13-1901","name":"28-6E76-BA13-1901","dev":{"ids":["ems-esp"]}}

3.6.0.dev.2
ems-esp/dallassensor_data
{"28-B07E-AE13-1901":{"name":"28-B07E-AE13-1901","temp":32.2},"28-F043-2800-0080":{"name":"28-F043-2800-0080","temp":45.1},"28-28DA-B413-1901":{"name":"28-28DA-B413-1901","temp":29.0},"28-0E2C-2800-0080":{"name":"28-0E2C-2800-0080","temp":31.4},"28-6E76-BA13-1901":{"name":"28-6E76-BA13-1901","temp":25.0}}

3.6.0.dev.0
ems-esp/dallassensor_data
{"28-B07E-AE13-1901":{"name":"28-B07E-AE13-1901","temp":32.1},"28-F043-2800-0080":{"name":"28-F043-2800-0080","temp":45.1},"28-28DA-B413-1901":{"name":"28-28DA-B413-1901","temp":29.0},"28-0E2C-2800-0080":{"name":"28-0E2C-2800-0080","temp":31.3},"28-6E76-BA13-1901":{"name":"28-6E76-BA13-1901","temp":24.4}}
```

MichaelDvP, 2023-03-05:
Are the other devices (boler, etc.) working? They have also the avty setting in dev-2.

sq4bja, 2023-03-05:
Yes, some EMS-ESP devices are not updated, but it is difficult to determine which ones. Some devices containing the temperature taken from the boiler using the EMS protocol appeared correctly, e.g. dhw temperature.

MichaelDvP, 2023-03-05:
I think domoticz can't read the number devices, that's what the domoticz-setting replace by sensor, but if others work, try https://github.com/MichaelDvP/EMS-ESP32/releases I've modified the avty setting of dallas to `if defined else`  as in the ems topics. Also the domotic-setting should work now.

sq4bja, 2023-03-05:
Unfortunately, it didn't help. Still sensors are not updated. Only the MQTT communication selection feature has improved - I can now select Domoticz.

MichaelDvP, 2023-03-05:
Sorry, there was a typo, try again.

sq4bja, 2023-03-05:
Still the same...

MichaelDvP, 2023-03-05:
Then i'm out of ideas. The dallas discovery topic now looks the same as the topics from ems-entities. You have to find out what domoticz don't like in this topic and how to fix it. Then we can use the domoticz switch for a different topic.

MichaelDvP, 2023-03-05:
Just for testing: The dallas uses `value_json['28-B07E-AE13-1901'].temp` because HA dont like numbers (or the minus?). Other entities use the name direct, e.g. `value_json.hc2.seltemp` . Let's try if domoticz likes it if i set (only if domoticz is activated in mqtt)  `value_json.28-B07E-AE13-1901.temp`. Updated in my repo.

proddy, 2023-03-05:
I don't have Domoticz but I suggest the easiest way to test this @sq4bja  is by going into MQTTExplorer and modifying the config topics directly there - and then seeing how Domoticz responds. Easier than coding and doing builds.

sq4bja, 2023-03-07:
Ok, i`ll try this, but i found some errors in Domoticz:
```
 2023-03-07 23:40:41.480 Error: EMS-ESP: Exception (GetValueFromTemplate): in Json::Value::resolveReference(key, end): requires objectValue! (Template: value_json[28-0E2C-2800-0080] if value_json[28-0E2C-2800-0080] is defined else -55)
2023-03-07 23:40:41.480 Error: EMS-ESP: Exception (GetValueFromTemplate): in Json::Value::resolveReference(key, end): requires objectValue! (Template: value_json[28-28DA-B413-1901] if value_json[28-28DA-B413-1901] is defined else -55)
2023-03-07 23:40:41.480 Error: EMS-ESP: Exception (GetValueFromTemplate): in Json::Value::resolveReference(key, end): requires objectValue! (Template: value_json[28-6E76-BA13-1901] if value_json[28-6E76-BA13-1901] is defined else -55)
2023-03-07 23:40:41.480 Error: EMS-ESP: Exception (GetValueFromTemplate): in Json::Value::resolveReference(key, end): requires objectValue! (Template: value_json[28-B07E-AE13-1901] if value_json[28-B07E-AE13-1901] is defined else -55)
2023-03-07 23:40:41.481 Error: EMS-ESP: Exception (GetValueFromTemplate): in Json::Value::resolveReference(key, end): requires objectValue! (Template: value_json[28-F043-2800-0080] if value_json[28-F043-2800-0080] is defined else -55) 
```
Option: sending with individual topics

sq4bja, 2023-03-07:
The only problem is value in dallassensor config:
` if value_json['28-B07E-AE13-1901'] is defined else -55`
If I remove it, the temperature sensor starts to update.

Here is previous topic:
```
 "val_tpl": "{{value_json['28-B07E-AE13-1901'] if value_json['28-B07E-AE13-1901'] is defined else -55}}",
  "obj_id": "dallassensor_28-B07E-AE13-1901",
```
And now the new working:
```
"val_tpl": "{{value_json['28-B07E-AE13-1901']}}",
  "object_id": "dallassensor_28-B07E-AE13-1901",
```

MichaelDvP, 2023-03-08:
Is it working if you set:
`"val_tpl": "{{value_json.28-B07E-AE13-1901 if value_json.28-B07E-AE13-1901 is defined else -55}}",`

The "if .. is defined" is used for all entitiies, Strange that it only makes issues for dallas.

proddy, 2023-03-08:
> The "if .. is defined" is used for all entitiies, Strange that it only makes issues for dallas.

only for Domoticz as they don't fully support the Discovery specification. We would need to remove the if condition for Domoticz then I'm guessing.

MichaelDvP, 2023-03-08:
> only for Domoticz as they don't fully support the Discovery specification.

Yes, but why it works with some entities and other not. I've ask here https://github.com/emsesp/EMS-ESP32/issues/1102#issuecomment-1455089742 but did not get an answer that allows analysis. I don't have domoticz, i don't have homeassistant and i don't have a crystall ball. But removing the availibility we have to add the remove again and run in mqtt-queue overflow and low ram, or get the next complains about strange errormessages from discovery service.

proddy, 2023-03-08:
yes, we'll leave it in. I'm not going to back to how it was before! What @sq4bja needs to do is report this in the Domoticz forum to get an explanation of what's going on, and use MQTT Explore to change the config and see what works/doesn't work. Next feature is a crystal ball for Michael ;-)

sq4bja, 2023-03-08:
> 
Currently, other built-in sensors are working properly. The problem is only with the Dallas temperature sensors.
I appreciate your work and I don't expect the entire project to be adapted to the errors of one of the systems, but if a differentiation of protocols for HA and Domoticz has already been introduced. Maybe correcting these topics only for Domoticz would be possible?

sq4bja, 2023-03-08:
> 
`"val_tpl": "{{value_json.28-B07E-AE13-1901 if value_json.28-B07E-AE13-1901 is defined else -55}}",`
Nothing has changed. Still no update.

MichaelDvP, 2023-03-09:
If you set individual topics and name the sensor without numbers (eg.: dallasname), you can try 
`val_tpl": "{{value_json.dallasname if value_json.dallasname is defined else -55}}`
which is exactly the same format as all other ems-entities. Otherwise we have to skip the availability for domotics, but that needs to manual remove the sensors in case of disconnect.

sq4bja, 2023-03-09:
I renamed the sensor to Dallas1. This resulted in changing the val_tpl variable as below.

```
{
  "dev_cla": "temperature",
  "stat_t": "ems-esp/dallassensor_data",
  "unit_of_meas": "°C",
  "val_tpl": "{{value_json['DALLAS1'] if value_json['DALLAS1'] is defined else -55}}",
  "obj_id": "dallassensor_28-B07E-AE13-1901",
  "uniq_id": "dallassensor_28-B07E-AE13-1901",
  "name": "DALLAS1",
  "dev": {
    "ids": [
      "ems-esp"
    ]
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/dallassensor_data",
      "val_tpl": "{{'online' if value_json['DALLAS1'] is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}
```
Unfortunately, the state of the sensor in Domoticz still does not update.

MichaelDvP, 2023-03-10:
I know that this won't work, that's why a suggusted something different.
Last try:
Check: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
use nested mqtt!
For nested dallas i prefix the nest with "ds" and for analog with "io". This should work with HA and domoticz without changing the entity. Only individual topic will work only with a alphabetic name in domoticz. 
If this does not work, i give up.

sq4bja, 2023-03-10:
Sorry, but when i try to update, i get:
`Upload failed (500: Internat Server Error)
`
Newer version dev 5 works fine (without any temperaure sensors (dallas and internal boiler temperature sensors).

MichaelDvP, 2023-03-11:
Ups, a few byes to large for partition. Try again, now updating works for me.

sq4bja, 2023-03-11:
In the dev4 version, Dallas sensors still do not appear, and additionally the boiler temperature sensors do not refresh.
I have set topic format Nested.
Well, I will finish testing EMS-ESP on dev0 - the last working version, and I will patiently wait for the Domoticz or EMS-ESP community to solve the problem.
Thank you again for your help and I keep my fingers crossed for the development of this project.

closed 2023-05-25 by proddy