emsesp/EMS-ESP32 #1109: error 500 updating 3.0.0-dev2 to 3.6.0-dev4 not possible

harvey637, 2023-03-09:
Hi,
updating from 3.6.0-dev.2 to 3.6.0-dev.4 results in 
internal error 500

I did:
reinstalled esp32 with new partitioning from 3.5.0-B2
factory reset
added old config manually (and compared downloaded configs)
added my custimization (mostly disabling warmwater ....)

back to 3.6.0-dev.2: working, but
some problems with customizations, like NOT saving some WW related stuff, json looks broken:
```
         "03/wwchargeduration|>0<3810",
          "03/wwcharge",
          "03/wwextra1",
          "03/wwextra2",
          "03/wwdisinfecting",
          "03/wwdisinfectday",
          "03/wwdisinfecttime|>0<1431",
          "03hybridstrategy",
          "03switchovertemp",
```
and 
```
          "03elheatstep2",
          "03elheatstep3",
          "03ww/wwalternatingop",
          "03ww/wwaltopprioheat|>20<120",
          "03ww/wwaltopprioww|>30<120",
          "03ww/wwcomfoff|>15<65",
          "03ww/wwecooff|>15<65",
```
I am hopiong that this might be fixed, but cannot test due to update problem :-(
cu
Harvey

MichaelDvP, 2023-03-09:
Yes, the actual dev4 is to large for 4MB flash (with symetrical ota partitions). My dev3 is mostly up to date with dev4 but fits in partiton. See #608 for a possible solution.

3.5.0b2 have not all customization-features, upload the customization file again after update to 3.6.0-dev.x

harvey637, 2023-03-09:
Hi @ MichaelDvP,
thank you for your answer.
So I will wait for a near future solution. I dont want to burn my productive ems-esp, this gets me into family trouble!

If a solution might be to strip down languages, what about a small subset? How needs to switch from swedisch to vietnamese on the fly? Or load a language package as addon with only the english basic?
If the software grows (always goes into this direction :-) an "english only" version with the upload (and deletion!) of a choosen language? As bare minimum I can live with an english_only version, that is uptodate, better as no updates possible or buy new hardware just to switch between 10 languages.
 
Just my idea, might be easier as to play around with partiition schemes and multilevel OTA updating.
cu
Harvey

proddy, 2023-03-10:
dev-4 should fit now.

closed 2023-03-10 by proddy

harvey637, 2023-03-10:
Hi,
thank you - but .....
Do I have to wait for a new build, as a freshly downloaded 3.6.0-dev.4 still breaks the upload, and it is 2 days old :-)

I will check again later today ...
cu+
Harvey

proddy, 2023-03-10:
> Hi, thank you - but ..... Do I have to wait for a new build, as a freshly downloaded 3.6.0-dev.4 still breaks the upload, and it is 2 days old :-)
> 
> I will check again later today ... cu+ Harvey

oops, forgot to upload...!