emsesp/EMS-ESP32 #1120: Dynamically upload Device Entity translations via WebUI

proddy, 2023-03-13:
up to 3.6.0 all the device entity names for the ~8 languages are hardcoded into Flash. And we're running short on available Flash on a 4MB ESP32 board (like a BBQKees gateway S32 or E32).

What @tp1de suggested is to have all the language packs as separate files and you can choose which one to load via the Customizations page. The file would be persisted in the filesystem and added to DROM when the entity is registered (if possible!). This would save 70K in Flash and enables us to switch to the latest espressif 6.1.0/IDF 5.x and add more web features.

proddy, 2023-03-15:
@MichaelDvP what do you think?

MichaelDvP, 2023-03-15:
As far as i understand this means to provide customization files to upload ,where the custom name is the translated entitiy. 
This will work and save flash, but we'll run out of ram. Also the enums, commands, etc are not translated this way.
Difficult.

tp1de, 2023-03-15:
I am using now since some days the different partioning by Michael. 
This gives more than 700k additional memory. 
Isn't this the better and easier solution?

proddy, 2023-03-15:
> As far as i understand this means to provide customization files to upload ,where the custom name is the translated entitiy. This will work and save flash, but we'll run out of ram. Also the enums, commands, etc are not translated this way. Difficult.

that's exactly the challenge. I'm not sure it's possible to write to Flash memory dynamically from the running code. I was hoping there was a work-around but if not there's no point as it'll blow up the RAM.

MichaelDvP, 2023-03-16:
You can read/write to partitons https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/partition.html or direct to flash https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_flash/index.html
before writing the sector (4k, alligned) has to be erased.
The difficulti is to access the strings. The compiler stores them in a row and puts the starting addresses in the code. For changing language you can:
- reserve a fixed length for each string to always access on the same address, but that's to short for some pl texts and wast of space for short words like on/off
- store the strings in a row and extra store an index of the starting addresses. The index can be generated on upload by esp or generated extern and uploaded with the list.

In both cases adding a new string have to be done at end of list, not thematical sorted, difficult to handle. Otherwise updating software and using an old language file ends in chaos. If you prepare the index external you can keep the list sorted and only keep the index compatible with new entries at the end.