emsesp/EMS-ESP32 #1130: EMS-ESP32 stops working

SchusterPeter, 2023-03-19:
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/11011610/emsesp_info.txt)

### PROBLEM DESCRIPTION

I'm using MQTT, everything works fine, but after 3 or probably 5 t0 7 days the EMS-ESP32 stops sending MQTT information to the server and it also does not response to ping also the web page does not open. Only after power reset it starts to work again.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

Sould run without any stops.

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

How can I analyse the reasons for this problem? What logs shall I trace?

**(Please, remember to close the issue when the problem has been addressed)**

SchusterPeter, 2023-03-19:
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/11011646/emsesp_info.txt)

proddy, 2023-03-19:
Could you upgrade to 3.5.1 and see if the problem is still there?

MichaelDvP, 2023-03-19:
Do you really need qos2? If a message is lost in qos1/2 sometimes ems-esp waits for response until queue is overflowed and the messages is removed. Systems with lot of entitiies run in low heap and are not reponsible until mqtt queue is cleared again. 
I've seen this sometimes with qos1, but have fewer entities, so it works again.
For now i can only recommend to use qos0.

proddy, 2023-07-03:
I agree, I haven't seen any issue with 3.6.x or any reports of more mqtt disconnects. I'll close this for now, please re-open if it happens again

closed 2023-07-03 by proddy