emsesp/EMS-ESP32 #1131: Heartbeat setting

SchusterPeter, 2023-03-19:
### PROBLEM DESCRIPTION

I'm setting the heartbeat to 60 sec, but if I subscripe to the MQTT server to see the heartbeat, I see the following:

{"bus_status":"connected","uptime":"000+00:03:14.330","uptime_sec":194,"ntp_status":"0","rxreceived":259,"rxfails":0,"txreads":85,"txwrites":0,"txfails":1,"mqttcount":268,"mqttfails":0,"mqttconnects":33,"apicalls":0,"apifails":0,"sensorreads":0,"sensorfails":0,"freemem":142}
{"bus_status":"connected","uptime":"000+00:03:20.328","uptime_sec":200,"ntp_status":"0","rxreceived":267,"rxfails":0,"txreads":86,"txwrites":0,"txfails":1,"mqttcount":276,"mqttfails":0,"mqttconnects":34,"apicalls":0,"apifails":0,"sensorreads":0,"sensorfails":0,"freemem":142}
{"bus_status":"connected","uptime":"000+00:03:26.326","uptime_sec":206,"ntp_status":"0","rxreceived":277,"rxfails":0,"txreads":90,"txwrites":0,"txfails":1,"mqttcount":284,"mqttfails":0,"mqttconnects":35,"apicalls":0,"apifails":0,"sensorreads":0,"sensorfails":0,"freemem":143}

It looks like there is an update every 6 seconds.

What is the reason for this?

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ x
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/11011670/emsesp_info.txt)
] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2023-03-19:
Thanks, heartbeat is additional called in health check where the LED blinking is triggered. I think just removing this call should fix it.

proddy, 2023-03-23:
> Thanks, heartbeat is additional called in health check where the LED blinking is triggered. I think just removing this call should fix it.

I think so too. Is it in your latest dev build?

MichaelDvP, 2023-03-23:
Yes. I'll make a PR if the two actual PRs are closed.

closed 2023-04-23 by proddy