emsesp/EMS-ESP32 #1132: 3.5.1 does not turn off heating pump modulation

DonMiguel69, 2023-03-20:
### PROBLEM DESCRIPTION

Hi, I discovered a potential issue with 3.5.1: 

I am using HA automation to turn off heating on my Buderus GB152-16 when there is no heating demand from my tado thermostats for 5 minutes. In order to turn off heating, I am setting thermostat_hc1_mode to night and thermostat_hc1_reducemode to frost protection on the RC35 via MQTT push. 

With 3.5.0, this worked like charm and set both burner power (boiler_curburnpow) and heating pump modulation (boiler_heatingpumpmod) to 0%. 

With 3.5.1, this is setting burner power to 0%, but heating pump modulation to 100%. 

This happened for 2 nights, then I rolled back to 3.5.0, and now the pump modulates back to 0% when the burner power is at 0%. 

Is this a bug in 3.5.1? I know there is a new switch in 3.5.1 that can be used to disable heating, but the night mode / frost protection has the advantage that I can just press the "auto" or "day" button on the RC35 in case anything goes wrong with the automation...

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

### TO REPRODUCE

System reverted back to original behaviour after downgrade to 3.5.0

### EXPECTED BEHAVIOUR

Heating pump modulation to switch to 0% when burner power is at 0% - as it was with 3.5.0.

### SCREENSHOTS
3.5.0 Behaviour:
[image]

3.5.1 Behaviour:
[image]

See "Heizungspumpe 1 Modulation" (lower graph) in the attached screenshots.
[system.txt](https://github.com/emsesp/EMS-ESP32/files/11017124/system.txt)

MichaelDvP, 2023-03-20:
Seems that nofrost mde works like described in the manual, if outdoor temperature falls below set setpoint, pump stays on to avoid freezing of the pipes. These two nights were a bit colder than the others. 
Ems-esp only do what you have set: switch to night mode and switch reducemode. The pumps are controlled by the thermostat depending on mode settings.

But there is a bug in emsesp, the nofrostmode shows outdoor when in thermostat set to room, and shows room if in thermostat set to outdoor. I've fixed this in dev build.

DonMiguel69, 2023-03-20:
Hi Michael,

wow - looks like the firmware upgrade at the time of lower temperatures was a big coincidence. 

Exactly the 2 nights when I was on 3.5.1 had lower temperatures, and the damped night temperature was below 5 degrees:

[image]

So 100% pump modulation happens when the heating switched to frost protection? Then I'll try to switch to 3.5.1. again and see what happens ;-)

BTW: My tado thermostats would have switched on my radiators if the indoor temperature was below 16 degrees. This would have led to switching the boiler to reduced night mode. So no actual risk of pipes freezing ;-)

So if I change the nofrost mode to "room" this will then only enable the pump to switch to frost protection once the RC35 thermostat falls below 5 degrees?

Thanks a lot for your great help,

Michael

DonMiguel69, 2023-03-21:
Hi, yesterday I upgraded to 3.5.1. again, and during the night the pump modulation went down to 0% (along with the burner modulation) once there was no more heating demand from the thermostats.
So pump modulation works as designed and the 100% pump modulation was in fact triggered by frost protection.

Regarding frost protection: I can confirm the mixing-up of "outdoor" and "room" in frost protection mode. Thanks for fixing this in dev build!
On the other hand, the RC35 does not seem to accept the frost protection (trigger) temperature. I can change it to 10 degrees via EMS-ESP and it seems to get acknowledged, but in the RC35 UI it still shows 5 degrees:

[image]
[image]

Isn't this supposed to be the same temperature?

Thanks a lot!

MichaelDvP, 2023-03-21:
RC35 Manual says:
```
6.4.5 Frostschutz
Die Frostschutzfunktion umfasst folgende Möglichkeiten:
•
Kein Frostschutz (Frostschutz ist ausgeschaltet).
•
Außentemperatur (Außentemperaturfühler erforderlich) Wenn die Außentemperatur die
Schwelle der einstellbaren Frostgrenztemperatur unterschreitet, wird die Pumpe des Heizkreises
automatisch eingeschaltet.
•
Raumtemperatur (Raumtemperaturfühler des RC35 oder RC2x) Wenn die Raumtemperatur
unter den fest eingestellten Wert von 5 °C sinkt, wird die Pumpe des Heizkreises automatisch
eingeschaltet. Wenn die Raumtemperatur über 7 °C steigt, wird die Pumpe des Heizkreises
automatisch ausgeschaltet.
```
for nofrostmode room it's fixed 5°C, for mode outdoor you can change the value.

DonMiguel69, 2023-03-23:
Hi, I assumed the RC35 is using its internal temperature sensor for frost protection via "room" temperature, so one of those 2 temperature sensors:

[image]

But maybe "room" setting requires an additional sensor, or that heating regulation is configured to use the RC35 temperature and not an external temperature sensor? Sorry, just guessing... And if this is not a bug, then I can also take the discussion to discord....

Cheers,
DonMiguel

DonMiguel69, 2023-03-23:
So the sensor is there and displaying a temperature (see screenshot above, "Temperatursensor 1" and "Temperatursensor 2"), but of course not used to control the heating temperature, as it is mounted to the boiler and located in the basement.
The idea was to still switch frost protection to "room" in order to avoid the pump running every night when it is below 5 degrees outside, as the house is fully insulated. I cannot imagine a situation where the temperature in the rooms or within the walls would drop significantly below 0 degrees and heating pipes would freeze. In that case, there are still my tado Thermostats which would detect temperature falling below 16 degrees and requesting heat for a room. And witjhout electrical power there is also no frost protection ;-)  So maybe frost protection off is the preferred choice for me....

DonMiguel69, 2023-04-17:
Hi Michael, 
thanks a lot for all your input! Heating control is all stable, no more issues ;-)

By the way: Due to "Wife Acceptance Factor", I am preferring to temporarily switch off heating via night mode. 
In case automation with EMS/ESP + Home Assistant does not work as expected, heating can easily be re-enabled by pressing the "day" or "auto" button on the RC35 without having to go into the RC35 menu structure ;-)

Greetings, DonMiguel

closed 2023-07-03 by proddy