emsesp/EMS-ESP32 #1134: Wrong mapping for "hc1 nofrost mode"

Marcwa19197, 2023-03-21:
### PROBLEM DESCRIPTION

Wrong mapping for "hc1 nofrost mode". It shows "room" in the WebUI but on the device (Buderus) its "outside".
Im using EMS-ESP Version v3.5.1.

Or is this fixed in the latest DEV Version?

### TO REPRODUCE

Set "room" for "hc1 nofrost mode" via WebUI and check the setting on the device (RC35). It shows "Outside" and not "room". (and if you choose "outside" in webUI it shows "room" on the device.) so there must be a wrong mapping.

### EXPECTED BEHAVIOUR

Mapping should be correct.

### SCREENSHOTS
[image]

[image]

MichaelDvP, 2023-03-21:
I've already mentioned it here: https://github.com/emsesp/EMS-ESP32/issues/1132#issuecomment-1476172282
It is fixed in the 3.6.dev. (also mypatch-branch, but not merged to official yet).

Marcwa19197, 2023-03-21:
Yeah, sorry for creating a new issue! This took a time to figure out. I was wondering why my pump war working in during the night, so i cam across this bug... ;-)

You can close it! Thanks!

closed 2023-03-21 by Marcwa19197