emsesp/EMS-ESP32 #1144: Dev EMS-ESP Version v3.6.0-dev.8 Dashboard Status EMS Read Quality

Vikingn, 2023-03-24:
### PROBLEM DESCRIPTION

The Raad was about 50% after upload of the newest version

After some minutes its 91% now
[image]

Then i use this instead: EMS-ESP-3_6_0-dev_7-ESP32

The read etc are at 100% Quality from the start !

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Vikingn, 2023-03-25:
I will try the dev-8 again today and power off and on again then to test

Vikingn, 2023-03-25:
Tried now after about 30-60 s it look like this 
[image]
after 2 min like this

[image]
So i go back to EMS-ESP-3_6_0-dev_7-ESP32.bin
After 1 min look like this:
[image]

Vikingn, 2023-03-25:
[image]

Vikingn, 2023-03-25:
We can do a web meeting and look at it together if you want.

proddy, 2023-03-25:
are you using the bins from the GH repo or compiling & building yourself? The only thing I can think of is the increased uart stack but it shouldn't make a difference.

Vikingn, 2023-03-25:
I downloaded it from your GH repo

proddy, 2023-04-07:
still seeing this issue?

nhmariend, 2023-04-07:
I csn try and check out but i tried det 8 2-3 times very slow to increase the read percentage. On det 7 its 100%. After short time. So something does norsk fine here. We can do s web meeting on it if you want that.

closed 2023-07-03 by proddy