emsesp/EMS-ESP32 #1146: Unrecognized EMS device

FirstCoderNil, 2023-03-25:
Hello.
I compiled by my self to get it working on an AZ-Delivery ESP32 DevKitC V2 and it works so far.
I can access the WebInterface and it find a my Boiler but not my room thermostate which is a Bosch CR 100.

I get this via Telnet:

┌──────────────────────────────────────┐
│ EMS-ESP version 3.6.0-dev.8          │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

000+00:04:58.879 I 0: [shell] User session opened on console pty0
000+00:05:01.574 E 1: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 02 00 1B
000+00:05:01.574 W 2: [emsesp] Unrecognized EMS device (device ID 0x18, no product ID). Please report on GitHub.
000+00:05:08.332 E 3: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 02 00 1B
000+00:05:08.332 W 4: [emsesp] Unrecognized EMS device (device ID 0x18, no product ID). Please report on GitHub.
000+00:05:14.276 E 5: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 02 00 1B
000+00:05:14.276 W 6: [emsesp] Unrecognized EMS device (device ID 0x18, no product ID). Please report on GitHub.

I'm not sure if I have maybe a problem with my DIY EMS adapter which is new.

This is maybe helpful too:

000+00:19:15.700 W 215: [emsesp] Unrecognized EMS device (device ID 0x18, no product ID). Please report on GitHub.
ems-esp:$ show ems
EMS Bus is connected, but Tx is not stable.

EMS Bus info:
  Tx mode: 3
  Bus protocol: HT3
  #recognized EMS devices: 2
  #telegrams received: 567
  #read requests sent: 0
  #write requests sent: 0
  #incomplete telegrams: 0
  #read fails (after 3 retries): 3
  #write fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 0%

Rx Queue is empty

Tx Queue (2 telegrams):
 [99*] READ  Me(0x0B) -W-> 18(0x18), Version(0x02), data: 1B
 [94 ] READ  Me(0x0B) -W-> 18(0x18), Version(0x02), data: 1B

MichaelDvP, 2023-03-26:
Please attach the system information from help page.
Seems the TX is not working. Wrong gpio? broken cable? More nformation about your system is needed to help.

bbqkees, 2023-03-27:
Indeed no TX it seems.
If you made your own circuit make sure you also implemented the TX part with the correct resistor values and proper circuit ground.

FirstCoderNil, 2023-03-28:
Hello,
yes it is my own circuit and there could be still a problem with it. I had not exactly all parts but I did try to get close as possible by choosing and combining compareable.. maybe I have to do some fine tuning..

But receiving seems to work? Or could it be that it is not working correct too?

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/11088997/emsesp_info.txt)

and some more log.. maybe useful too:

https://pastebin.com/n5xVysJT

MichaelDvP, 2023-03-28:
> But receiving seems to work? 

Yes, works good. But ems-esp needs a working tx to request the version info from the devices. The boiler have a fallback to "generic", but for thermostat this is difficult. You should check your tx gpio, connection and circuit.

FirstCoderNil, 2023-03-30:
Thank you. This infos are very helpful. Now I understand much better how it works. I'll check my circuit later and let you know if I get it working.

FirstCoderNil, 2023-04-01:
I got it working now. I have connected my circuit with some bad and long temp test wires which seems to cause the problem.
I found this: https://github.com/emsesp/EMS-ESP/issues/364
Then I did try without the inductors too and it solved the problem.

closed 2023-04-01 by FirstCoderNil