emsesp/EMS-ESP32 #1147: Exhaust temperature always zero on GB125/MC110/RC310

minusdreidb, 2023-03-26:
Hi,

I noticed that the exhaust gas temperature reported by EMS-ESP32 is always zero on my system while the RC310 shows the actual value. Th3M3 notes (https://github.com/Th3M3/buderus_ems-wiki/blob/master/Quelle_08.md) align with the data I found on my system and I implemented a now tested fix which can be found here: https://github.com/minusdreidb/EMS-ESP32/commit/982a43b8d438e3d2926bd038f4c971139702b66b.

The location at offset 6 in telegram 0xE5 is always zero on my system causing the initial problem, therefore I commented it out.
Telegram 0x19 which also seems to carry the exhaust temp. according to boiler.cpp seems not to be sent on my system.

What is the right strategy to handle this situation (assuming that offset 6 in 0xE5 is a known good location for the exhaust temperature and not a bug)?

My system setup including SW-Versions is documented in the above linked commit's message.

proddy, 2023-03-26:
What do you think @MichaelDvP ? Just remove the ExhaustTemp from 0xE5/UBAMonitorSlowPlus for everyone?

MichaelDvP, 2023-03-26:
The e5 offset 6 was just a guess from https://github.com/emsesp/EMS-ESP/issues/620#issuecomment-731527364 becauseof fast changing temperaturerise after burner start. 
@minusdreidb  If you can verify that e4 offset 31 is the exhaust temp, make a PR.
But in the old thread e4 offset 31 was always zero. Maybe read both and take the nonzero value?

proddy, 2023-03-26:
also note its a short 2-byte so you make sure the value at offset 6 and 7 are both zero

minusdreidb, 2023-03-27:
>@minusdreidb If you can verify that e4 offset 31 is the exhaust temp, make a PR. But in the old thread e4 offset 31 was always zero. Maybe read both and take the nonzero value?

@MichaelDvP I verified the location against the diagnostics reading for the exhaust temp. on my RC310.

>But in the old thread e4 offset 31 was always zero. Maybe read both and take the nonzero value?

This is something that should be handled in a consistent manner by has_update() in my opinion.

> also note its a short 2-byte so you make sure the value at offset 6 and 7 are both zero

@proddy Sorry, I don't understand your comment. Does has_update() not read only 2 bytes for exhaustTemp_?

proddy, 2023-03-27:
> minusdreidb@982a43b

I think exhaustTemp_ is a uint16_t which means it's two bytes long, starting at offset 6. So zero is 0x0000

minusdreidb, 2023-03-27:
Ok, got it. You meant for testing the "raw" bytes in 0xE5 and conditionally calling has_update().

commit d993b356 (2023-04-02): Merge pull request #1150 from minusdreidb/dev

Fix #1147 - Exhaust temperature always zero on GB125/MC110/RC310
commit e2fec03a (2023-04-02): Use exhaust temperature from 0xe4, remove from 0xe5 until verified, closes #1147.
closed 2023-07-03 by proddy